raspiblitz/raspiblitz #2599: SSD test for health with smartmontools

openoms, 2021-10-05:
A quick test if SSD failure is suspected:

```
# install
sudo apt install smartmontools

# quick health info
sudo smartctl -d sat -H /dev/sda
 
# full test (with the X825 USB-SATA board - note `-d sat`)
sudo smartctl -a -d sat /dev/sda
# SMART overall-health self-assessment test result: PASSED

# Count should be 0 on the following:
sudo smartctl -a -d sat /dev/sda | grep "Reallocated"
#  5 Reallocated_Sector_Ct   0x0032   100   100   010    Old_age   Always       -       0
# 196 Reallocated_Event_Count 0x0032   100   100   000    Old_age   Always       -       0

sudo smartctl -a -d sat /dev/sda | grep "Reported"
# 187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       0
```

openoms, 2021-10-05:
Short self test:
```
sudo smartctl -t -d sat short /dev/sda
```
Results:
```
sudo smartctl -d sat -l selftest /dev/sda
```

openoms, 2021-10-05:
There are limitations to this as it doesn't clearly catch my known failed disk after long usage and recurrent bad sectors (was reformatted since):
```
$ sudo smartctl -a -d sat /dev/sda
smartctl 6.6 2017-11-05 r4594 [x86_64-linux-5.10.0-0.bpo.8-amd64] (local build)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Device Model:     WDC WDS500G2B0B-00YS70
Serial Number:    183016801236
LU WWN Device Id: 5 001b44 8b63866cd
Firmware Version: X61190WD
User Capacity:    500,107,862,016 bytes [500 GB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Form Factor:      M.2
Device is:        Not in smartctl database [for details use: -P showall]
ATA Version is:   ACS-4 T13/BSR INCITS 529 revision 5
SATA Version is:  SATA 3.3, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Tue Oct  5 08:56:35 2021 BST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x00)	Offline data collection activity
					was never started.
					Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0)	The previous self-test routine completed
					without error or no self-test has ever 
					been run.
Total time to complete Offline 
data collection: 		(    0) seconds.
Offline data collection
capabilities: 			 (0x11) SMART execute Offline immediate.
					No Auto Offline data collection support.
					Suspend Offline collection upon new
					command.
					No Offline surface scan supported.
					Self-test supported.
					No Conveyance Self-test supported.
					No Selective Self-test supported.
SMART capabilities:            (0x0003)	Saves SMART data before entering
					power-saving mode.
					Supports SMART auto save timer.
Error logging capability:        (0x01)	Error logging supported.
					General Purpose Logging supported.
Short self-test routine 
recommended polling time: 	 (   2) minutes.
Extended self-test routine
recommended polling time: 	 (  10) minutes.

SMART Attributes Data Structure revision number: 4
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  5 Reallocated_Sector_Ct   0x0032   100   100   ---    Old_age   Always       -       0
  9 Power_On_Hours          0x0032   100   100   ---    Old_age   Always       -       12872
 12 Power_Cycle_Count       0x0032   100   100   ---    Old_age   Always       -       45176
165 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       1877159775855
166 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       29
167 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       62
168 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       139
169 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       464
170 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       0
171 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       0
172 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       0
173 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       60
174 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       72
184 End-to-End_Error        0x0032   100   100   ---    Old_age   Always       -       0
187 Reported_Uncorrect      0x0032   100   100   ---    Old_age   Always       -       0
188 Command_Timeout         0x0032   100   100   ---    Old_age   Always       -       0
194 Temperature_Celsius     0x0022   074   061   ---    Old_age   Always       -       26 (Min/Max 7/61)
199 UDMA_CRC_Error_Count    0x0032   100   100   ---    Old_age   Always       -       1
230 Unknown_SSD_Attribute   0x0032   100   100   ---    Old_age   Always       -       7864185784103
232 Available_Reservd_Space 0x0033   100   100   004    Pre-fail  Always       -       100
233 Media_Wearout_Indicator 0x0032   100   100   ---    Old_age   Always       -       30177
234 Unknown_Attribute       0x0032   100   100   ---    Old_age   Always       -       44792
241 Total_LBAs_Written      0x0030   253   253   ---    Old_age   Offline      -       23740
242 Total_LBAs_Read         0x0030   253   253   ---    Old_age   Offline      -       41709
244 Unknown_Attribute       0x0032   000   100   ---    Old_age   Always       -       0

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
No self-tests have been logged.  [To run self-tests, use: smartctl -t]

Selective Self-tests/Logging not supported
```

openoms, 2021-10-05:
An other erroneous one:
```
$ sudo smartctl -d sat -l error /dev/sda
smartctl 6.6 2017-11-05 r4594 [aarch64-linux-5.10.60-v8+] (local build)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF READ SMART DATA SECTION ===
SMART Error Log Version: 1
Warning: ATA error count 0 inconsistent with error log pointer 1

ATA Error Count: 0
	CR = Command Register [HEX]
	FR = Features Register [HEX]
	SC = Sector Count Register [HEX]
	SN = Sector Number Register [HEX]
	CL = Cylinder Low Register [HEX]
	CH = Cylinder High Register [HEX]
	DH = Device/Head Register [HEX]
	DC = Device Command Register [HEX]
	ER = Error register [HEX]
	ST = Status register [HEX]
Powered_Up_Time is measured from power on, and printed as
DDd+hh:mm:SS.sss where DD=days, hh=hours, mm=minutes,
SS=sec, and sss=millisec. It "wraps" after 49.710 days.

Error -4 occurred at disk power-on lifetime: 0 hours (0 days + 0 hours)
  When the command that caused the error occurred, the device was active or idle.

  After command completion occurred, registers were:
  ER ST SC SN CL CH DH
  -- -- -- -- -- -- --
  00 00 00 00 00 00 00   at LBA = 0x00000000 = 0

  Commands leading to the command that caused the error were:
  CR FR SC SN CL CH DH DC   Powered_Up_Time  Command/Feature_Name
  -- -- -- -- -- -- -- --  ----------------  --------------------
  61 18 f8 17 dc 00 00 00      00:00:00.000  WRITE FPDMA QUEUED
  61 ff f8 ff ff ff 00 00      00:00:00.000  WRITE FPDMA QUEUED
  60 80 00 c8 8c fc 00 00      00:00:00.000  READ FPDMA QUEUED
  60 80 00 c8 8c fc 00 00      00:00:00.000  READ FPDMA QUEUED
  60 08 00 40 b1 57 00 00      00:00:00.000  READ FPDMA QUEUED
```

Had a run of Crucial BX500 disks with like 50% failure rate (https://github.com/rootzoll/raspiblitz/issues/1805)

david-bakin-sl, 2024-02-13:
The article mentioned in issue main comment "need to use smartctl -d sat with all the commands"
is now at https://linuxopsys.com/topics/test-ssd-hdd-health-in-linux

rootzoll, 2024-03-13:
Consider a short stress test when an NVMe is connected .. writing 16GB and check temp.

openoms, 2024-05-18:
NVMe check on a new disk:

```
sudo smartctl -a /dev/nvme0n1 -d nvme
```
For NVMe:
We should warn about the `Percentage Used: ` 
above 75%  - orange warning
above 90% - red warning

And compare the:
`Data Units Written: ` to the given specs (eg 600 TBW in this case)

```
smartctl 7.3 2022-02-28 r5338 [aarch64-linux-6.6.20+rpt-rpi-v8] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       CT1000P3PSSD8
Serial Number:                      2342445568EF
Firmware Version:                   P9CR411
PCI Vendor/Subsystem ID:            0x1344
IEEE OUI Identifier:                0x00a075
Controller ID:                      0
NVMe Version:                       1.4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          1,000,204,886,016 [1.00 TB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            00a075 01445568ef
Local Time is:                      Sat May 18 11:39:36 2024 BST
Firmware Updates (0x14):            2 Slots, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL Self_Test
Optional NVM Commands (0x00d7):     Comp Wr_Unc DS_Mngmt Sav/Sel_Feat Timestmp Verify
Log Page Attributes (0x1e):         Cmd_Eff_Lg Ext_Get_Lg Telmtry_Lg Pers_Ev_Lg
Maximum Data Transfer Size:         64 Pages
Warning  Comp. Temp. Threshold:     83 Celsius
Critical Comp. Temp. Threshold:     85 Celsius
Namespace 1 Features (0x08):        No_ID_Reuse

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     5.50W       -        -    0  0  0  0        0       0
 1 +     3.00W       -        -    1  1  1  1        0       0
 2 +     1.50W       -        -    2  2  2  2        0       0
 3 -   0.0300W       -        -    3  3  3  3     5000    2500
 4 -   0.0025W       -        -    4  4  4  4     8000   40000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         1
 1 -    4096       0         0

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        47 Celsius
Available Spare:                    100%
Available Spare Threshold:          5%
Percentage Used:                    0%
Data Units Read:                    16,874,928 [8.63 TB]
Data Units Written:                 8,649,652 [4.42 TB]
Host Read Commands:                 329,030,135
Host Write Commands:                89,876,003
Controller Busy Time:               463
Power Cycles:                       40
Power On Hours:                     961
Unsafe Shutdowns:                   9
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               47 Celsius

Error Information (NVMe Log 0x01, 16 of 255 entries)
No Errors Logged
```

For SSD:
```
177 Wear_Leveling_Count     0x0013   093   093   000    Pre-fail  Always       -       158
```
 this means 93 % life is left with 158 block reallocated (representing 7% of the spare blocks on the device)

and:
```
241 Total_LBAs_Written      0x0032   099   099   000    Old_age   Always       -       122793459646
```
note: look at nodl dev branch for a prioritization of different smartctl codes