raspiblitz/raspiblitz #2603: [RFC] Systemd Shutdown script

dipunm, 2021-10-06:
On my pi, I have installed a UPS (sugarpi) and it has a safe shutdown feature. Unfortunately, it does not allow me to configure a custom shutdown script.

I did a little reading, and digging, and found this article: https://trstringer.com/simple-vs-oneshot-systemd-service/#run-a-service-at-shutdown

My thought is that we can inject the official shutdown script in a systemd service that will run on shutdown to make sure bitcoind, lnd etc. all have the appropriate time to clean up and exit before the system shuts down.

I have experienced that running shutdown normally doesn't do this at the moment (in v 1.7.0).

This is an RFC, because I want to first find out whether I am missing something obvious here, is this a good idea to implement or would it be a bad idea? If it is a good idea, I am happy to implement this and try to submit a PR :)

nyxnor, 2021-10-07:
Calling this script does that https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/XXshutdown.sh
Off is in your path, you can call `off` to shutdown https://github.com/rootzoll/raspiblitz/blob/2651abeaa3632792da85308a5ad7292492ad5613/home.admin/_commands.sh#L64

dipunm, 2021-10-07:
I think you misunderstood, the tool I am using to `safe shutdown` (working with a UPS) **doesn't** have the ability to call a custom script.

That's why I am suggesting a `systemd service`. It's like an OS level shutdown hook.

dipunm, 2021-10-07:
That was interesting, thanks! Had a quick look. This is unfortunately not appropriate, for me.
I am using this bad boy: https://github.com/PiSugar/PiSugar/wiki/PiSugar2-Pro

It does not support the apc ups standard. It instead uses it's own (opensource) custom UPS functionality.
Fortunately, my idea isn't specific to this device, it is an attempt to create a more general solution.

dipunm, 2021-10-08:
I think I can close this issue. Thanks for bouncing this idea with me @nyxnor. I will provide a proof of concept PR later :).

closed 2021-10-08 by dipunm