raspiblitz/raspiblitz #2611: Tor Source for debian

mrsieb, 2021-10-11:
https://github.com/rootzoll/raspiblitz/blob/4ead6f00d0d53beaeab5bb7209cb854dd5195f34/build_sdcard.sh#L206-L220

Any special reason why  debian is not covered?

dipunm, 2021-10-13:
As identified here: https://github.com/rootzoll/raspiblitz/issues/1752. I think we should add `[arch=arm64]` to this when appropriate (64bit OS)

rootzoll, 2021-12-11:
OK for v1.7.2 the `build_sdcard.sh` is using the new `tor.install.sh` script by @nyxnor to install Tor for all baseimages from the debian repos. To be final tested if also is working will all images.

rootzoll, 2021-12-22:
@nyxnor got this error during sd card build - do we still need to adapt the install? How best to check if Re-Install was successfull? And manybe add a automatic check & error exit if unsucessfull?

```
Err:5 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
```

Full log in context:

```
*** Adding Tor Sources ***

deb [arch=arm64] tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye main
deb-src [arch=arm64] tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org  bullseye main

- OK sources added

*** Reinstall ***
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye-updates InRelease                                                                                
Hit:3 http://security.debian.org/debian-security bullseye-security InRelease                                                                 
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease                                                                            
Get:5 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease [3,526 B]
Err:5 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
Reading package lists... Done
W: http://deb.debian.org/debian/dists/bullseye/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/torproject.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://deb.debian.org/debian/dists/bullseye-updates/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/torproject.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://security.debian.org/debian-security/dists/bullseye-security/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/torproject.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://archive.raspberrypi.org/debian/dists/bullseye/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/torproject.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org/dists/bullseye/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/torproject.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: GPG error: tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
E: The repository 'tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
tor is already the newest version (0.4.5.10-1~deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
apt-transport-tor is already the newest version (0.5).
curl is already the newest version (7.74.0-1.3+deb11u1).
gpg is already the newest version (2.2.27-2).
nyx is already the newest version (2.1.0-2.1).
obfs4proxy is already the newest version (0.0.8-1+b6).
python3-stem is already the newest version (1.8.0-3).
torsocks is already the newest version (2.3.0-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

*** PREPARING BITCOIN ***

```

nyxnor, 2021-12-22:
The gpg needs to be dearmored.
```
--dearmor
              Pack or unpack an arbitrary input into/from an OpenPGP ASCII armor.
```
It is currently like this https://github.com/rootzoll/raspiblitz/blame/dev/home.admin/config.scripts/tor.install.sh#L142
```sh
torsocks curl -s https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | sudo gpg --no-default-keyring --keyring gnupg-ring:/etc/apt/trusted.gpg.d/torproject.gpg --import
```

See the original method: https://github.com/rootzoll/raspiblitz/blame/e7f04fd5d5ada068c316d5d54983a0c988e817f7/home.admin/config.scripts/tor.install.sh#L47
```sh
torsocks curl --tlsv1.3 --proto =https --connect-timeout 10 https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --dearmor | sudo tee /usr/share/keyrings/tor-archive-keyring.gpg >/dev/null
```

I would change this now to:
```sh
tor_deb_repo_clean="${tor_deb_repo#*tor+}"
curl -s -x socks5h://127.0.0.1:9050 --tlsv1.3 --proto =https --connect-timeout 10 "${tor_deb_repo_clean}"/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/torproject.gpg >/dev/null
```
This will avoid the need to use torsocks if it failed to install before and save the key dearmored inside the apt/trusted.gpg.d folder. This will also use the same url from the `tor_deb_repo_clean` but without the preprending `tor+` needed for apt, as now this is the curl process.

nyxnor, 2021-12-22:
@rootzoll which branch are you on so I can commit?

rootzoll, 2021-12-23:
That sd card build was on `dev` branch

rootzoll, 2022-01-04:
PR #2845 merged ... TODO: restest sd card build

rootzoll, 2022-02-17:
sd catd builded fine for latest RCs ... closing issue.

closed 2022-02-17 by rootzoll