raspiblitz/raspiblitz #2622: Password C is asked two times after startup

cstenglein, 2021-10-14:
At startup, Password C is asked to unlock the lightning wallet. (Working as intended).

But after that, it takes some minutes to re-sync everything and Password C is asked a second time to unlock.

Dehumanizer77, 2021-11-24:
I have installed my raspiblitz yesterday and I am also experiencing this. However, after initial setup and a few reboots after that, it was only asked once. Only after I have enabled some more apps, it started happening, so it might be connected with some apps beeing enabled...

Tracachang, 2021-12-13:
I have also experimented and it is because bitcoind is still loading (Loading block index...).

To avoid this instead of entering the password when is prompted, I wait until all blocks have been loaded and bitcoind starts running and then entering just once is enough.

rootzoll, 2021-12-13:
@Tracachang thx thats worth a try

rootzoll, 2021-12-16:
merged the two PRs .. putting on final testing for v1.7.2 release candidates

rootzoll, 2022-02-16:
OK havent seen this problem anymore since the release candidates. Considered fixed. Closing issue.

closed 2022-02-16 by rootzoll