raspiblitz/raspiblitz #2624: Telegraf: dashboard missing information

dipunm, 2021-10-15:
There are a couple of issues, one that I have a fix for, the other not:
The file `/home.admin/assets/telegraf/raspiblitz-grafana-dashboard.json` has the grafana dashboard, however it is trying to read ip address information from a `measurement` (table) called `raspiblitz_ip_info`. This information is however stored in a measurement called `ipinfo`. It also tries to give IPv6 information, but at least on my setup, there is no v6 IP address, so I modified it to show IPv4 information.

I exported the JSON of the modified dashboard and pasted it at the bottom.

The part that is harder, is that mempool information pulled from electrs no longer works since electrs v0.9.0. I have created a relevant issue there, but it looks like the source information has been completely refactored and electrs may not be a good source for this information going forward. https://github.com/romanz/electrs/issues/551

```
{
  "annotations": {
    "list": [
      {
        "builtIn": 1,
        "datasource": "-- Grafana --",
        "enable": true,
        "hide": true,
        "iconColor": "rgba(0, 211, 255, 1)",
        "limit": 100,
        "name": "Annotations & Alerts",
        "showIn": 0,
        "target": {
          "limit": 100,
          "matchAny": false,
          "tags": [],
          "type": "dashboard"
        },
        "type": "dashboard"
      },
      {
        "datasource": "-- Grafana --",
        "enable": true,
        "hide": true,
        "iconColor": "rgba(255, 96, 96, 1)",
        "limit": 100,
        "matchAny": true,
        "name": "Show reboots",
        "showIn": 0,
        "tags": [
          "reboot"
        ],
        "target": {
          "limit": 100,
          "matchAny": true,
          "tags": [
            "reboot"
          ],
          "type": "tags"
        },
        "type": "tags"
      },
      {
        "datasource": "-- Grafana --",
        "enable": false,
        "hide": true,
        "iconColor": "#5794F2",
        "limit": 100,
        "name": "Global Annotations",
        "showIn": 0,
        "tags": [
          "global"
        ],
        "type": "tags"
      }
    ]
  },
  "editable": true,
  "fiscalYearStartMonth": 0,
  "gnetId": null,
  "graphTooltip": 0,
  "id": 2,
[... 6933 lines cut ...]
  },
  "time": {
    "from": "now-4d",
    "to": "now"
  },
  "timepicker": {
    "refresh_intervals": [
      "5s",
      "10s",
      "30s",
      "1m",
      "5m",
      "15m",
      "30m",
      "1h",
      "2h",
      "1d"
    ]
  },
  "timezone": "",
  "title": "Bitcoin/LN/electrs - Node",
  "uid": "ImRpu1igk",
  "version": 3
}
```

dipunm, 2021-10-16:
Good news, the author added back some of the metrics, unfortunately, the mempool_vsize statistics are not grouped by fee_rate so it is not quite the same, I have asked if they will make another change. Overall though, progress is being made :)

rootzoll, 2021-11-30:
@dipunm do you like to contribute your Telegraf config back to the main project for the v1.7.2 release?

rootzoll, 2021-12-09:
pushing to v1.8.0 to check again back then

dipunm, 2022-09-29:
Apologies, this issue was no longer on my radar. I can provide by telegraf config, I was hoping it would be a small change where the diffs could be compared, but unfortunately I think it is not easy to create. I will upload my config as soon as I can :)

dipunm, 2022-09-29:
Here is the latest json:
```json
{
  "annotations": {
    "list": [
      {
        "builtIn": 1,
        "datasource": {
          "type": "datasource",
          "uid": "grafana"
        },
        "enable": true,
        "hide": true,
        "iconColor": "rgba(0, 211, 255, 1)",
        "limit": 100,
        "name": "Annotations & Alerts",
        "showIn": 0,
        "target": {
          "limit": 100,
          "matchAny": false,
          "tags": [],
          "type": "dashboard"
        },
        "type": "dashboard"
      },
      {
        "datasource": {
          "type": "datasource",
          "uid": "grafana"
        },
        "enable": true,
        "hide": true,
        "iconColor": "rgba(255, 96, 96, 1)",
        "limit": 100,
        "matchAny": true,
        "name": "Show reboots",
        "showIn": 0,
        "tags": [
          "reboot"
        ],
        "target": {
          "limit": 100,
          "matchAny": true,
          "tags": [
            "reboot"
          ],
          "type": "tags"
        },
        "type": "tags"
      },
      {
        "datasource": {
          "type": "datasource",
          "uid": "grafana"
        },
        "enable": false,
        "hide": true,
        "iconColor": "#5794F2",
        "limit": 100,
        "name": "Global Annotations",
        "showIn": 0,
        "tags": [
          "global"
        ],
        "type": "tags"
      }
    ]
  },
  "editable": true,
  "fiscalYearStartMonth": 0,
[... 7373 lines cut ...]
  "time": {
    "from": "now-4d",
    "to": "now"
  },
  "timepicker": {
    "refresh_intervals": [
      "5s",
      "10s",
      "30s",
      "1m",
      "5m",
      "15m",
      "30m",
      "1h",
      "2h",
      "1d"
    ]
  },
  "timezone": "",
  "title": "Bitcoin/LN/electrs - Node",
  "uid": "ImRpu1igk",
  "version": 12,
  "weekStart": ""
}
```

rootzoll, 2024-10-06:
OK added Telegraf service to SSH menus fpr RaspiBlitz v1.11.3rc1 with PR #4788

PR #4788 (2024-10-06): #2624 Add Telegraf to SSH menus
fixes #2624
closed 2024-10-06 by rootzoll