raspiblitz/raspiblitz #2634: c-lightning announcing 127.0.0.1:9736

rootzoll, 2021-10-16:
@openoms is there a special reason why c-lightning is doing an annoucement of the local-network IP to the global peer network with `announce-addr=127.0.0.1:${portprefix}9736` see:

https://github.com/rootzoll/raspiblitz/blob/aa962c105a589ee8b711e32718542787095a690f/home.admin/config.scripts/cl.install.sh#L186

openoms, 2021-10-16:
That line was used initially to work around this (now fixed) issue: https://github.com/ElementsProject/lightning/issues/4597

Since https://github.com/ElementsProject/lightning/pull/4603 is merged it should be ok to remove.

Alternatively can be changed to use a public IP or IP2Tor address.

This is our use case: https://lightning.readthedocs.io/TOR.html#case-7-c-lightning-has-no-public-ip-address-a-fixed-tor-v3-service-address

openoms, 2021-10-17:
the fix fix changes the config retroactively as well with the cl.check.sh on restart. Will run with it until the 127.0.0.1 announcement disappears from the gossip and will cherry-pick to `v1.7`
 too

openoms, 2021-10-17:
done, part of `v1.7` and `dev` also

openoms, 2021-11-08:
was part of the updated v1.7.1 image

commit 4c0e0d1b (2021-10-17): CL: do not announce 127.0.0.1 and check in CLCONF
fixes #2634
commit 2993ad76 (2021-12-11): CL: do not announce 127.0.0.1 and check in CLCONF
fixes #2634
closed 2021-11-08 by openoms