raspiblitz/raspiblitz #2636: Add squeaknode app

yzernik, 2021-10-17:
This app has dependency on Bitcoin and LND.
https://github.com/yzernik/squeaknode

yzernik, 2021-10-17:
Yes, I'll try to get a pull request up in the next few days. Thanks

openoms, 2021-11-08:
What's a squeak?

yzernik, 2021-11-08:
@openoms It's the name for a post in the squeaknode app, similar to a tweet. But it's open source and immutable, and timestamped with the latest bitcoin block hash, and signed with a private key.

Looks like this:
[image]

openoms, 2021-11-08:
Thanks for the explanation! Would be nice to have this clear straight in the project readme.

yzernik, 2021-11-08:
@openoms Good idea. Updated the readme: https://github.com/yzernik/squeaknode/pull/1788

yzernik, 2021-11-26:
Opened PR: https://github.com/rootzoll/raspiblitz/pull/2731

rootzoll, 2022-05-17:
@yzernik  on preparing the PR for merge I am not sure anymore if the zmq entry are still correct - its merged now into dev but please check if you see anything missing

rootzoll, 2022-06-01:
@yzernik on my testing machine after installing Squeaknode it does not startup. Can you help? See error:

```
Jun 01 17:58:13 raspberrypi systemd[1]: Started squeaknode.
Jun 01 17:58:16 raspberrypi sh[362698]: Traceback (most recent call last):
Jun 01 17:58:16 raspberrypi sh[362698]:   File "/home/squeaknode/squeaknode/./venv/bin/squeaknode", line 5, in <module>
Jun 01 17:58:17 raspberrypi sh[362698]:     from squeaknode.main import main
Jun 01 17:58:17 raspberrypi sh[362698]:   File "/home/squeaknode/squeaknode/venv/lib/python3.9/site-packages/squeaknode/main.py", line 31, in >
Jun 01 17:58:17 raspberrypi sh[362698]:     from squeaknode.node.squeak_node import SqueakNode
Jun 01 17:58:17 raspberrypi sh[362698]:   File "/home/squeaknode/squeaknode/venv/lib/python3.9/site-packages/squeaknode/node/squeak_node.py", >
Jun 01 17:58:17 raspberrypi sh[362698]:     from squeaknode.admin.webapp.app import SqueakAdminWebServer
Jun 01 17:58:17 raspberrypi sh[362698]:   File "/home/squeaknode/squeaknode/venv/lib/python3.9/site-packages/squeaknode/admin/webapp/app.py", >
Jun 01 17:58:17 raspberrypi sh[362698]:     from flask_login import current_user
Jun 01 17:58:17 raspberrypi sh[362698]:   File "/home/squeaknode/squeaknode/venv/lib/python3.9/site-packages/flask_login/__init__.py", line 16>
Jun 01 17:58:17 raspberrypi sh[362698]:     from .login_manager import LoginManager
Jun 01 17:58:17 raspberrypi sh[362698]:   File "/home/squeaknode/squeaknode/venv/lib/python3.9/site-packages/flask_login/login_manager.py", li>
Jun 01 17:58:17 raspberrypi sh[362698]:     from .utils import (login_url as make_login_url, _create_identifier,
Jun 01 17:58:17 raspberrypi sh[362698]:   File "/home/squeaknode/squeaknode/venv/lib/python3.9/site-packages/flask_login/utils.py", line 13, i>
Jun 01 17:58:17 raspberrypi sh[362698]:     from werkzeug.security import safe_str_cmp
Jun 01 17:58:17 raspberrypi sh[362698]: ImportError: cannot import name 'safe_str_cmp' from 'werkzeug.security' (/home/squeaknode/squeaknode/v>
Jun 01 17:58:17 raspberrypi systemd[1]: squeaknode.service: Main process exited, code=exited, status=1/FAILURE
```

rootzoll, 2022-06-02:
Removing squeaknode visibly from v1.8.0 release (deactivating entry in SERVICES menu) - pushing to v1.8.1

rootzoll, 2022-11-26:
@yzernik whats the status - can you help pn the error posted above. Otherwise I will push up to a later release.

closed 2022-12-15 by rootzoll