raspiblitz/raspiblitz #2639: Restoring from backup file (_provision.setup.sh fail)

The34Goon, 2021-10-17:
Trying to recover from backup_file to fresh SD card with existing blockchain data:

- **Preparing SSD**

\*************************************
\* JUST MAKING BACKUP TO SD CARD
\*************************************
please wait ..
\# *** LND.RESCUE --> BACKUP
mv: cannot move '/home/admin/lnd.version.info' to '/mnt/hdd/lnd/version.info': No such file or directory
chown: cannot access '/mnt/hdd/lnd/version.info': No such file or directory
\# Stopping lnd...
\# OK

tar: Removing leading `/' from member names
tar: /mnt/hdd/lnd: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
error='backup is empty'

!!!! This will DELETE your personal data & POSSIBLE FUNDS from the HDD !!!!
--> It will keep Blockchain data - so you dont have to download/copy again.
Press ENTER to really continue - CTRL+c to CANCEL (last chance)

stopping services ... (please wait)
- swap
- background
- lnd
- blockchain
selective delete ... (please wait)
\*************************
OK - the HDD is now clean
\*************************
reboot -> sudo shutdown -r now
power off -> sudo shutdown now
Connection to 192.168.0.126 closed by remote host.
Connection to 192.168.0.126 closed.

- **INSERT FRESH SD 1.7.1**
SSH in to the new node and continue

_**NOTE:**_
_When promted to your fresh Rapsiblitz there are three options but none point to 
\*From Resucue File*
which is found two steps later when choosing FRESHSETUP_

- **Setup New Blitz**

<USE BLOCKCHAIN>

<DELETE DATA>

Choose a new name
→Entering existing old name.

Choose Client:
→ LND

Choose NEW/OLD
→ OLD

Choose Recover 
→ LNDRESCUE

UPLOAD RESCUE FILE with instructions

In new terminal, inside the folder that contains the rescue-file:
scp -r ./lnd-rescue-*.tar.gz bitcoin@192.168.0.126:/home/bitcoin/temp/upload/

PW: raspiblitz 
→ 100% switching back to other terminal

→ file found

Get hint: 
The Raspiblitz uses 3 Passwords and that setting A + B need to be set now.
_**No further hints to when PW C will be set.**_

→ PW A accepted
→ Pw B accepted

**Next step fails:**

\###########################################################
\# /home/admin/raspiblitz.log

\***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Sun 17 Oct 15:12:35 BST 2021
\***********************************************
\# *** blitz.ssh.sh checkrepair
\# countKeyFiles(8)
\# OK: SSHD RUNNING
No HDMI switch found. 
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No SSHRESET switch found. 
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 604
drwxr-xr-x   2 root root   4096 Oct 12 00:12 .
drwxr-xr-x 118 root root   4096 Oct 12 00:12 ..
-rw-r--r--   1 root root 565189 Jan 31  2020 moduli
-rw-r--r--   1 root root   1580 Jan 31  2020 ssh_config
-rw-r--r--   1 root root   3250 Jan 31  2020 sshd_config
-rw-------   1 root root   1381 Oct 12 00:12 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Oct 12 00:12 ssh_host_dsa_key.pub
-rw-------   1 root root    513 Oct 12 00:12 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Oct 12 00:12 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Oct 12 00:12 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Oct 12 00:12 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Oct 12 00:12 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Oct 12 00:12 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Mar  5  2021 ssh_import_id
wifiIsSet=0
hddBackupLocationAvailable=0
hddRestoreConfigAvailable=0
memRestoreConfigAvailable=0
wifiRestore=0
wifiBackup=0
No UASP FIX needed (2nd-try).
isMounted: 0
Checking for UASP deactivation ...
HDD already part of system: 0
Temp mounting (1) data drive (sda)
Temp mounting (1) result: 1
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
\# df 
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root       14987608   4385864   9945396  31% /
devtmpfs         1933976         0   1933976   0% /dev
tmpfs            1968600         0   1968600   0% /dev/shm
tmpfs            1968600      8632   1959968   1% /run
tmpfs               5120         0      5120   0% /run/lock
tmpfs            1968600         0   1968600   0% /sys/fs/cgroup
tmpfs              32768         4     32764   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095     30357    227738  12% /boot
tmpfs             393720         0    393720   0% /run/user/1000
/dev/sda1      960379920 415068516 496456980  46% /mnt/hdd
\# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda                
└─sda1      ext4   BLOCKCHAIN
mmcblk0            
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
\# RASPIBLITZ DATA DRIVE Status

\# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=0
hddRaspiVersion=''
isSSD=0
datadisk='sda'
datapartition='sda1'
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddBytes=1000203091968
hddGigaBytes=931
hddUsedInfo='396G (46%)'
hddDataFreeKB=496456980
hddAdapterUSB='174c:1153'
hddAdapterUSAP=0

\# RAID
isRaid=0
raidCandidates=0

\# SWAP
isSwapExternal=0

Displaying Info Message: Please Login for Setup
\## WAIT LOOP: USER SETUP/UPDATE/MIGRATION
\# PROVISION PROCESS with setupPhase(setup)
Temp mounting (2) data drive (sda)
Temp mounting (2) result: 1
Refreshing links between directories/drives ..
\# Sourcing /mnt/hdd/raspiblitz.conf 
\# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.7.1'
lcdrotate=1
lightning=lnd
network=bitcoin
chain=main
hostname='ZERF'
runBehindTor=on
\# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup 
\# RASPIBLITZ SETUP STATE
network=bitcoin
hostname=ZERF
lightning=lnd
network=bitcoin
chain=main
setPasswordA=1
setPasswordB=1
lndrescue='/home/bitcoin/temp/upload/lnd-rescue-7d9e128a83992e1746d6975840f92935.tar.gz'
\# df 
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root       14987608   4285924  10045336  30% /
devtmpfs         1933976         0   1933976   0% /dev
tmpfs            1968600         0   1968600   0% /dev/shm
tmpfs            1968600     25032   1943568   2% /run
tmpfs               5120         0      5120   0% /run/lock
tmpfs            1968600         0   1968600   0% /sys/fs/cgroup
tmpfs              32768        20     32748   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095     30357    227738  12% /boot
tmpfs             393720         0    393720   0% /run/user/1000
tmpfs             393720         0    393720   0% /run/user/1001
/dev/sda1      960379920 415068536 496456960  46% /mnt/hdd
\# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda                
└─sda1      ext4   BLOCKCHAIN
mmcblk0            
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
\# Update /home/admin/raspiblitz.info 
state=provision
message='Starting Provision'
baseimage=raspios_arm64
cpu=aarch64
board=rp4
ramMB=3844
ramGB=4
network=bitcoin
chain=main
localip='192.168.0.126'
online='1'
dnsworking=1
fsexpanded=1
displayClass=lcd
displayType=
setupStep=0
setupPhase='setup'
fundRecovery=0
hostname=
hddCandidate='sda'
hddGigaBytes=931
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddGotMigrationData=
\# migrationFile()
\# setting PASSWORD A

OK - password A changed for user pi, root, admin & bitcoin
...
Calling _provision.setup.sh for basic setup tasks ..
EXIT _provision.setup.sh BECAUSE OF ERROR STATE (5)
This can also happen if _provision.setup.sh has syntax errors
\# /home/admin/raspiblitz.provision-setup.log

\###################################
\# _provision.setup.sh
\###################################
\*** Prepare bitcoin ***
mkdir: cannot create directory ‘/mnt/hdd/bitcoin’: File exists
mkdir: cannot create directory ‘/mnt/hdd/bitcoin/blocks’: File exists
mkdir: cannot create directory ‘/mnt/hdd/bitcoin/chainstate’: File exists
File Exists: /mnt/hdd/bitcoin/bitcoin.conf --> 1
\# setting PASSWORD B
\# OK -> RPC Password B changed
\# Reboot is needed
...
\*** Start bitcoin (SETUP) ***
- This can take a while ..
\# (0/50) checking if bitcoind is running ... 
\# (1/50) checking if bitcoind is running ... 
\# (2/50) checking if bitcoind is running ... 
\# (3/50) checking if bitcoind is running ... 
OK bitcoin startup successful 
Prepare Lightning (lnd)
\############## Setup LND
FAIL: missing passwordC in (/var/cache/raspiblitz/temp/raspiblitz.setup)!
\***********************************************************
ERROR - please report to development team
\***********************************************************
state(error) message(_provision.setup.sh fail)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
admin@raspberrypi:~ $ 

→ off

Blitz shuts down.
***********
Unplug power to reboot.

SSH back in to the raspiblitz, now with new PW: A
Now prompted to

PLEASE START WITH A FRESH SD CARD IMAGE
(leads to same result)

rootzoll, 2021-10-17:
OK later in the logs we see:
```
Calling _provision.setup.sh for basic setup tasks ..
EXIT _provision.setup.sh BECAUSE OF ERROR STATE (5)
```

The exit code 5 of that script is: `FAIL: missing passwordC in setupfile`
https://github.com/rootzoll/raspiblitz/blob/02c4ee5323ec17eaadd95bbe8890708a6cbd3188/home.admin/_provision.setup.sh#L158

But normally there will be no Password C to be set - because that is already part of the LND backup.
It needs to be checked if this is just a strange side effect or a general problem.

stakoholic, 2021-11-02:
hello, 

i have the same problem with restoring my lnd wallet/channels, with the image from 12.10 and the patched 28.10.

it doesn´t work with the lnd recue..tar.gz file, same error with "missing password C" like described above, i tried it via the script /home/admin/config.scripts/lnd.backup.sh lnd-import-gui" i get error='mising parameter' and i tried just now with seed + channel.backup file with error "FAIL: setup: lnd import SCB failed
unknown parameter" 

thats the lnd.log

`admin@raspberrypi:~ $ sudo tail -f -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
2021-11-02 17:17:10.410 [INF] LTND: Version: 0.13.3-beta commit=v0.13.3-beta, build=production, logging=default, debuglevel=debug
2021-11-02 17:17:10.410 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2021-11-02 17:17:10.410 [INF] RPCS: Generating TLS certificates...
2021-11-02 17:17:10.448 [INF] RPCS: Done generating TLS certificates
2021-11-02 17:17:10.450 [DBG] ARPC: Autopilot RPC server successfully register with root gRPC server
2021-11-02 17:17:10.450 [DBG] NTFR: ChainNotifier RPC server successfully register with root gRPC server
2021-11-02 17:17:10.450 [DBG] VRPC: Versioner RPC server successfully registered with root gRPC server
2021-11-02 17:17:10.450 [DBG] WLKT: WalletKit RPC server successfully registered with root gRPC server
2021-11-02 17:17:10.450 [DBG] SGNR: Signer RPC server successfully register with root gRPC server
2021-11-02 17:17:10.450 [DBG] IRPC: Invoices RPC server successfully registered with root gRPC server
2021-11-02 17:17:10.450 [DBG] RRPC: Router RPC server successfully register with root gRPC server
2021-11-02 17:17:10.451 [INF] RPCS: RPC server listening on 0.0.0.0:10009
2021-11-02 17:17:10.461 [DBG] ARPC: Autopilot REST server successfully registered with root REST server
2021-11-02 17:17:10.462 [DBG] NTFR: ChainNotifier REST server successfully registered with root REST server
2021-11-02 17:17:10.462 [DBG] VRPC: Versioner REST server successfully registered with root REST server
2021-11-02 17:17:10.462 [DBG] WLKT: WalletKit REST server successfully registered with root REST server
2021-11-02 17:17:10.462 [DBG] SGNR: Signer REST server successfully registered with root REST server
2021-11-02 17:17:10.462 [DBG] IRPC: Invoices REST server successfully registered with root REST server
2021-11-02 17:17:10.466 [DBG] RRPC: Router REST server successfully registered with root REST server
2021-11-02 17:17:10.466 [INF] RPCS: gRPC proxy started at 0.0.0.0:8080
2021-11-02 17:17:10.466 [INF] LTND: Opening the main database, this might take a few minutes...
2021-11-02 17:17:10.466 [INF] LTND: Opening bbolt database, sync_freelist=true, auto_compact=false
2021-11-02 17:17:10.505 [INF] CHDB: Checking for schema update: latest_version=22, db_version=22
2021-11-02 17:17:10.506 [INF] LTND: Database now open (time_to_open=40.162555ms)!
2021-11-02 17:17:10.507 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.`

and thats the log from restore setup with seed + channel.backup file

`###########################################################
# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Thu 28 Oct 11:00:51 BST 2021
***********************************************
# *** blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
No HDMI switch found. 
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No SSHRESET switch found. 
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 604
drwxr-xr-x   2 root root   4096 Oct 28 11:00 .
drwxr-xr-x 118 root root   4096 Oct 28 11:00 ..
-rw-r--r--   1 root root 565189 Jan 31  2020 moduli
-rw-r--r--   1 root root   1580 Jan 31  2020 ssh_config
-rw-r--r--   1 root root   3250 Jan 31  2020 sshd_config
-rw-------   1 root root   1381 Oct 28 11:00 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Oct 28 11:00 ssh_host_dsa_key.pub
-rw-------   1 root root    513 Oct 28 11:00 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Oct 28 11:00 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Oct 28 11:00 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Oct 28 11:00 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Oct 28 11:00 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Oct 28 11:00 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Mar  5  2021 ssh_import_id
wifiIsSet=0
hddBackupLocationAvailable=0
[... 155 lines cut ...]
*** Start bitcoin (SETUP) ***
- This can take a while ..
# (0/50) checking if bitcoind is running ... 
# (1/50) checking if bitcoind is running ... 
# (2/50) checking if bitcoind is running ... 
# (3/50) checking if bitcoind is running ... 
# (4/50) checking if bitcoind is running ... 
# (5/50) checking if bitcoind is running ... 
# (6/50) checking if bitcoind is running ... 
# (7/50) checking if bitcoind is running ... 
OK bitcoin startup successful 
Prepare Lightning (lnd)
############## Setup LND
Creating new LND config ...
*** Init LND Service & Start ***
Starting LND Service ...
Preparing static channel backup file /home/bitcoin/temp/upload/channel.backup ...
FAIL: setup: lnd import SCB failed
unknown parameter
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(_provision.setup.sh fail)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off`

openoms, 2021-11-02:
Until we can reproduce this issue and find a reliable solution there is also the previous image still available at the `v1.7` tag:
https://github.com/rootzoll/raspiblitz/tree/v1.7.0#downloading-the-software

The only thing stops you from downgrading afaik is LND not running with a previous version after updating the database. So use the command line to enter:
`menu`
then `UPDATE` -> `LND` to the latest release.

openoms, 2021-11-03:
Edit you LND conf to (remove the other entries from the [Bitcoin] part):
```
[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1
```
according to: https://github.com/rootzoll/raspiblitz/blob/v1.7.0/home.admin/assets/lnd.bitcoin.conf

stakoholic, 2021-11-03:
my lnd.conf looks exactly like that and no other entries

> Edit your lnd.conf to (remove the other entries from the [Bitcoin] part):
> 
> ```
> [Bitcoin]
> bitcoin.active=1
> bitcoin.node=bitcoind
> # enable either testnet or mainnet
> #bitcoin.testnet=1
> bitcoin.mainnet=1
> ```
> 
> according to: https://github.com/rootzoll/raspiblitz/blob/v1.7.0/home.admin/assets/lnd.bitcoin.conf

openoms, 2021-11-03:
not sure what the problem here, but this is the line showing the error:
```
[ERR] LTND: unable to create partial chain control: status code: 401, response: ""
```
did you do a full restart yet?

stakoholic, 2021-11-03:
yes and i tried a second reset lnd and restoring, it ends always with the same lnd error?? maybe its caused by the new v0.14?

bitcoind is running and 100% synced

https://termbin.com/h2vz

via blitz menu i can only select the reckless v0.14, is it possible to manually update lnd to v.0.13.3, after new restore (where i have 0.12) ?

openoms, 2021-11-03:
I have not tested the LND 0.14 with channels yet so it might be possible that it needs some config changes.

It is possible to update to a specific version using these scripts: https://www.lightningnode.info/hardware-deployment/raspiblitz/lnd.updates

would recommend v0.13.3 and as you say use a new SDcard (and recovery process) to upload the rescue file which was not opened with lnd 0.14 yet.

openoms, 2021-11-03:
on v1.7.0 reset the lnd wallet from REAPIR and upload the latest rescue file which did not run with v0.14 before.

stakoholic, 2021-11-04:
> on v1.7.0 reset the lnd wallet from REPAIR and upload the latest rescue file which did not run with v0.14 before.

ok same error also with v0.13.3 beta  , lnd.conf looks like it should be

```
2021-11-04 15:26:40.529 [INF] LNWL: Opened wallet
2021-11-04 15:26:40.705 [INF] CHRE: Primary chain is set to: bitcoin
2021-11-04 15:26:40.977 [ERR] LTND: unable to create chain control: status code: 401, response: ""
2021-11-04 15:26:40.979 [INF] LTND: Shutdown complete
2021-11-04 15:27:41.169 [INF] LTND: Version: 0.13.3-beta commit=v0.13.3-beta, build=production, logging=default, debuglevel=debug
2021-11-04 15:27:41.169 [INF] LTND: Active chain: Bitcoin (network=mainnet)
```

maximlomans, 2021-11-04:
same here too .. I know what happened though tls.cert regeneration started the whole thing. Zap-Desktop doesn't have an option to edit tls certs i had to uninstall it completely twice in one week ,clear its dir out too.
Third time though lnd wasn't having any of it.The only diff was i had instlled BTCPay the day before..
..i paniced and disabled electrs btcpay and on the last try tor... 
i should'av just focused on the certs
i thought i have SBC setup and create lnd-rescue and refresh the blitz , i had no other option it shutdown automatically

i ended up with a fresh install still waiting for sync to finish and manually recover funds with chantools, 
fingers crossed

[image]

[image]

openoms, 2021-11-07:
synchronise the RPC user and password between bitcoin.conf and lnd.conf (if there is a password at all)

check the logs of bitcoin and LND from the SYSTEN menu

stakoholic, 2021-11-08:
> synchronise the RPC user and password between bitcoin.conf and lnd.conf (if there is a password at all)
> 
> check the logs of bitcoin and LND from the SYSTEM menu

ok i checked the configs, and the rpc pws was not equal, the lnd.conf has the rpc pw from the backup and the bitcoin.conf has the pw, from the fresh node i made before. 
after changing the pw in bitcoin.conf, restarted bitcoind + lnd, syncing is 100% and backup restore was successfull, everything works fine now, thx openoms ;)

openoms, 2021-11-08:
good to know that the update from the rescue file worked on the v1.7.0 image.

outstanding task here is to reproduce the problem on v1.7.1.

- [ ] make LND backup
- [ ] reflash with v1.7.1 image
- [ ] start fresh node and upload the lnd backup

maximlomans, 2021-11-19:
[image]
same here -

maximlomans, 2021-11-19:
[image]
reboot fresh image

openoms, 2021-12-20:
Confirmed still an issue on 1.7.2RC1 - passwordC is not asked
```
# setting PASSWORD A

OK - password A changed for user pi, root, admin & bitcoin
...
Calling _provision.setup.sh for basic setup tasks ..
Follow in a new terminal with: 'tail -f raspiblitz.provision-setup.log'
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(config: missing passwordC)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
```

openoms, 2021-12-20:
The next hindrance is:
```
Calling _provision.setup.sh for basic setup tasks ..
Follow in a new terminal with: 'tail -f raspiblitz.provision-setup.log'
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(lnd.initwallet.py new returned error)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off

₿ tail raspiblitz.provision-setup.log
Prepare Lightning (lnd)
############## Setup LND
Restore LND data from uploaded rescue file /home/bitcoin/temp/upload/lnd-rescue-b5fafae73fea83f9e8ebd96454188e78.tar.gz ...
*** Init LND Service & Start ***
Starting LND Service ...
Starting LND Service ... executed
WALLET --> NEW
FAIL: setup: lnd wallet SEED failed (2)
missing parameters
```

openoms, 2021-12-20:
OK, first test look good with `dev` scripts. Adding final testing.

rootzoll, 2021-12-21:
OK also on my test the setup ran thru BUT on the after setup reboot it didnt recognized that the wallet was locked and even after unlocking manually the status display didnt pick up on the LND status .. so maybe still something with the redis state to be fixed.

rootzoll, 2021-12-21:
Best guess when looking at the created `raspiblitz.conf` ..
```
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.7.2rc1'
lcdrotate='1'
lightning='lnd'
network='bitcoin'
chain='main'
hostname='testrecover'
runBehindTor=on
```
... its because the `lnd=on` is missing when setting up fresh with LNDRESCUE file.

rootzoll, 2021-12-21:
OK this looks good now. Removing the bug-lable and is ready for final testign as part of the relese candidates.

joernroeder, 2022-01-09:
ran into the same issue yesterday night, ported over the changes of the `_provision.setup.sh` script into my fresh v1.7.1 script folder and was able to restore a `lnd-rescue-*.tar.gz` seamlessly. 
Thanks, great job!

Silarous, 2022-01-16:
Any other solutions to this?  I have tried everything I could make out above and haven't had luck.  I was running v1.7.1 with LND upgraded to v0.14.1.  I did a LND backup and tried to restore in a fresh v1.7.1 Raspiblitz but LND fails to load.  Tried restoring it in 1.7.0 and I get the same problem with LND failing to load even after upgrading it to v0.14.1  I verified RPC user and pass match in bitcoin.conf and lnd.conf.  I made sure lnd=on is in the raspiblitz config.    [https://termbin.com/9njm](url)

joernroeder, 2022-01-17:
@Silarous I restored successfully with  an `lnd-rescue-*.tar.gz` file the following way:

- boot up a fresh v1.7.1 SD card
- ssh into it. stopping the menu with ctrl+c
- replacing the contents of `_provision.setup.sh` with my modified version (which contains the fixes mentioned in this thread) https://pastebin.com/Gj9hxXSC
- reboot the blitz
- install as usual: menu selections were: fresh blitz, I have an exising lnd followed by the upload of the `lnd-rescue-*` file

hope that helps

Silarous, 2022-01-17:
Unfortunately I am now getting hit with the missing password C.  I  did another fresh install of v1.7.1, signed in and immediately ctrl+c out of the menu, updated the _provision.setup.sh, rebooted the Blitz, signed in and selected to start fresh, existing LND, uploaded the rescue file, it started the install and stopped with this error.

###########################################################
# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Mon 17 Jan 04:17:11 GMT 2022
***********************************************
# *** blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
No HDMI switch found.
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No SSHRESET switch found.
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 604
drwxr-xr-x   2 root root   4096 Oct 28 11:00 .
drwxr-xr-x 118 root root   4096 Oct 28 11:00 ..
-rw-r--r--   1 root root 565189 Jan 31  2020 moduli
-rw-r--r--   1 root root   1580 Jan 31  2020 ssh_config
-rw-r--r--   1 root root   3250 Jan 31  2020 sshd_config
-rw-------   1 root root   1381 Oct 28 11:00 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Oct 28 11:00 ssh_host_dsa_key.pub
-rw-------   1 root root    505 Oct 28 11:00 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Oct 28 11:00 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Oct 28 11:00 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Oct 28 11:00 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Oct 28 11:00 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Oct 28 11:00 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Mar  5  2021 ssh_import_id
wifiIsSet=0
hddBackupLocationAvailable=0
hddRestoreConfigAvailable=0
memRestoreConfigAvailable=0
wifiRestore=0
wifiBackup=0
No UASP FIX needed (2nd-try).
isMounted: 0
Skipping UASP deactivation ... cmdlineExists(1) hddAdapterUSB() hddAdapterUSAP(0)
HDD already part of system: 0
Temp mounting (1) data drive (sda)
Temp mounting (1) result: 1
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
# df
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root       61068748   4386668  54153396   8% /
devtmpfs         1933976         0   1933976   0% /dev
tmpfs            1968600         0   1968600   0% /dev/shm
tmpfs            1968600      8632   1959968   1% /run
tmpfs               5120         0      5120   0% /run/lock
tmpfs            1968600         0   1968600   0% /sys/fs/cgroup
tmpfs              32768         4     32764   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095     30359    227737  12% /boot
tmpfs             393720         0    393720   0% /run/user/1000
/dev/sda1      960322016 432769612 478700980  48% /mnt/hdd
# lsblk -o NAME,FSTYPE,LABEL
NAME        FSTYPE LABEL
sda
└─sda1      ext4   BLOCKCHAIN
mmcblk0
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=0
[... 103 lines cut ...]
*** Prepare bitcoin ***
mkdir: cannot create directory ‘/mnt/hdd/bitcoin’: File exists
mkdir: cannot create directory ‘/mnt/hdd/bitcoin/blocks’: File exists
mkdir: cannot create directory ‘/mnt/hdd/bitcoin/chainstate’: File exists
File Exists: /mnt/hdd/bitcoin/bitcoin.conf --> 1
# setting PASSWORD B
# OK -> RPC Password B changed
# Reboot is needed
...
*** Start bitcoin (SETUP) ***
- This can take a while ..
# (0/50) checking if bitcoind is running ...
# (1/50) checking if bitcoind is running ...
# (2/50) checking if bitcoind is running ...
# (3/50) checking if bitcoind is running ...
OK bitcoin startup successful
Prepare Lightning (lnd)
############## Setup LND
FAIL: missing passwordC in (/var/cache/raspiblitz/temp/raspiblitz.setup)!
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(_provision.setup.sh fail)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off

openoms, 2022-01-17:
This should be fixed in the latest release candidate image. The download link is in the dev branch.
Bugs found there will be actively fixed so beat is to keep it up to date by running the shortcut command:
`patch`
in the command line.

joernroeder, 2022-01-17:
didn't know about the `patch` command. very helpful

Silarous, 2022-01-17:
Unfortunately still experiencing issues when trying to recover on 1.7.2rc1d.  I completed the following steps
-Selected option to delete Hdd and keep blockchain
-Flashed fresh 1.7.2rc1d image
-Selected fresh setup
-Use current blockchain and delete other data
-Selected to use LND and recover old wallet
-Selected LND Rescue and successfully transferred rescue file.
-Set passwords A & B

That was as far as it got, it then proceeded to the screen pictured below and is stuck for a hour now.  
[image]

openoms, 2022-01-17:
ok will try to reproduce this
to get more logs open a new terminal and log in with admin
check:
`tail raspiblitz.log`
 and 
`lndlogs`

maybe:
`sudo journalctl -u lnd`

Silarous, 2022-01-17:
**I  again deleted the Hdd while keeping the blockchain
Flashed fresh image
Booted, CTRL-C, and ran the patch command and gave this output**

 your active GitHubUser is: rootzoll
 your wanted GitHubUser is: rootzoll
 OK
 your active branch is: dev
 your wanted branch is: dev
 OK
 *** SYNCING RASPIBLITZ CODE WITH GITHUB ***
 This is for developing on your RaspiBlitz.
 THIS IS NOT THE REGULAR UPDATE MECHANISM
 and can lead to dirty state of your scripts.
 REPO ----> origin     https://github.com/rootzoll/raspiblitz.git (push)
 BRANCH --> dev
 ******************************************
remote: Enumerating objects: 197, done.
remote: Counting objects: 100% (172/172), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 113 (delta 90), reused 74 (delta 52), pack-reused 0
Receiving objects: 100% (113/113), 18.34 KiB | 4.58 MiB/s, done.
Resolving deltas: 100% (90/90), completed with 32 local objects.
From https://github.com/rootzoll/raspiblitz
   f65adc3a..0e9a45e6  dev        -> origin/dev
   fd425622..4878730b  v1.7       -> origin/v1.7
Updating f65adc3a..0e9a45e6
Fast-forward
 CHANGES.md                                     |  3 ++-
 FAQ.cl.md                                      | 41 ++++++++++++++++++++++++++++++++++++++---
 README.md                                      |  4 ++--
 home.admin/98repairMenu.sh                     |  6 ++++--
 home.admin/99clRepairMenu.sh                   | 12 ++++++++----
 home.admin/99lndRepairMenu.sh                  |  4 ++--
 home.admin/99updateMenu.sh                     |  2 +-
 home.admin/_background.scan.sh                 | 34 +++++++++++++++++++---------------
 home.admin/_bootstrap.sh                       | 17 +++++------------
 home.admin/_commands.sh                        |  1 +
 home.admin/_provision.migration.sh             |  1 +
 home.admin/_provision.setup.sh                 |  1 +
 home.admin/_provision.update.sh                |  1 +
 home.admin/config.scripts/blitz.datadrive.sh   | 17 +++++++++++------
 home.admin/config.scripts/blitz.migration.sh   | 21 +++++++++------------
 home.admin/config.scripts/blitz.setpassword.sh | 38 +++++---------------------------------
 home.admin/config.scripts/blitz.shutdown.sh    |  2 +-
 home.admin/config.scripts/bonus.lit.sh         |  2 +-
 home.admin/config.scripts/bonus.mempool.sh     | 72 +++++++++++++++++++++++++++++++++++++-----------------------------------
 home.admin/config.scripts/bonus.rtl.sh         |  2 +-
 home.admin/config.scripts/cl.backup.sh         |  6 +++---
 home.admin/config.scripts/cl.hsmtool.sh        | 12 +++++++++++-
 home.admin/config.scripts/lnd.backup.sh        |  6 +++---
 home.admin/config.scripts/lnd.compact.sh       | 18 ++++++++++++++++++
 home.admin/setup.scripts/controlFinalDialog.sh |  1 +
 25 files changed, 186 insertions(+), 138 deletions(-)
 COPYING from GIT-Directory to /home/admin/
 - basic admin files
 - asset directory
 - config.scripts directory
 - setup.scripts directory
 ******************************************
 Syncing Webcontent ..
 Checking if the content of BlitzPy changed ..
 checkSumBlitzPyBefore = 2f7909185be8ce3fd711f3029aedf714  -
 checkSumBlitzPyAfter  = 2f7909185be8ce3fd711f3029aedf714  -
 BlitzPy did not changed.
 ******************************************
 OK - shell scripts and assets are synced
 Reboot recommended

**I then rebooted
-Selected Fresh Setup
-Use Blockchain
-Delete HDD data
-LND option
-Restore OLD 
-LND Rescue and transferred rescue file
-Set passwords A & B**

**Again am stuck on the LND Testrun screen**

[image]

Silarous, 2022-01-17:
**It actually finally left the LND Testrun screen and output this log file**

###########################################################
 /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.2rc1
Mon 17 Jan 19:17:42 GMT 2022
***********************************************
 *** /home/admin/config.scripts/blitz.ssh.sh checkrepair
 countKeyFiles(8)
 OK: SSHD RUNNING
 INIT raspiblitz.info
baseimage=raspios_arm64
cpu=aarch64
displayClass=lcd
displayType=
setupPhase=setup
setupStep=0
fundRecovery=0
fsexpanded=1
state=starting
 INIT RaspiBlitz Cache ... wait background.scan.service to finish first scan loop
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime(6)
OK - logs are at 42 MB - within safety limit
Waiting for HDD/SSD ...
total 628
drwxr-xr-x   4 root root   4096 Jan 15 12:49 .
drwxr-xr-x 120 root root   4096 Jan 15 12:50 ..
-rw-r--r--   1 root root 577771 Mar 13  2021 moduli
-rw-r--r--   1 root root   1650 Mar 13  2021 ssh_config
drwxr-xr-x   2 root root   4096 Mar 13  2021 ssh_config.d
-rw-r--r--   1 root root   3289 Mar 13  2021 sshd_config
drwxr-xr-x   2 root root   4096 Mar 13  2021 sshd_config.d
-rw-------   1 root root   1385 Jan 15 12:49 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Jan 15 12:49 ssh_host_dsa_key.pub
-rw-------   1 root root    513 Jan 15 12:49 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Jan 15 12:49 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Jan 15 12:49 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Jan 15 12:49 ssh_host_ed25519_key.pub
-rw-------   1 root root   2602 Jan 15 12:49 ssh_host_rsa_key
-rw-r--r--   1 root root    570 Jan 15 12:49 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    342 Feb 28  2020 ssh_import_id
HDD/SSD connected: sda
No HDMI switch found.
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No WIFI RESTORE because no /var/cache/raspiblitz/hdd-inspect/wpa_supplicant.conf
No SSH SERVER CERTS RESTORE because no /var/cache/raspiblitz/hdd-inspect
No SSHRESET switch found.
No DISPLAY RESTORE because no /var/cache/raspiblitz/hdd-inspect/raspiblitz.conf
No UASP FIX needed
isMounted: 0
HDD already part of system: 0
Temp mounting (1) data drive
Temp mounting (1) result: 1
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
 df
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root      122615272   4871364 112710208   5% /
devtmpfs         1933976         0   1933976   0% /dev
tmpfs            1968600        16   1968584   1% /dev/shm
tmpfs             787440       712    786728   1% /run
tmpfs               5120         0      5120   0% /run/lock
[... 116 lines cut ...]
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
 migrationFile()
 Sourcing /mnt/hdd/raspiblitz.conf
 RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.7.2rc1'
lcdrotate='1'
lightning='lnd'
network='bitcoin'
chain='main'
hostname='LN-MasterNode'
runBehindTor='on'
 Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
 setting PASSWORD A

OK - password A changed for user pi, root, admin & bitcoin
...
Calling _provision.setup.sh for basic setup tasks ..
Follow in a new terminal with: 'tail -f raspiblitz.provision-setup.log'
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(lnd service not getting to running status)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off

openoms, 2022-01-17:
can you show your lnd.conf:
`sudo cat /home/bitcoin/.lnd/lnd.conf`

It clearly complains about this line:
```
 failed to load config: /home/bitcoin/.lnd/lnd.conf:53: unknown option: tor.streamisolation
 ```
You might have some syntax error in there.

Silarous, 2022-01-18:
# lnd configuration

[Application Options]
accept-keysend=true
listen=0.0.0.0:9735
tlsdisableautofill=1
tlsautorefresh=1
stagger-initial-reconnect=1
sync-freelist=1
ignore-historical-gossip-filters=1
debuglevel=debug
maxpendingchannels=5
minchansize=1000000
alias=************
color=#68F442
nat=false

#specify DDNS domain (port optional)
#externalhosts=net83.asuscomm.com:9735

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# REST open to all connections on Port 8080
restlisten=0.0.0.0:8080
# Domain, could use https://freedns.afraid.org
#tlsextradomain=
tlsextraip=0.0.0.0

[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

[autopilot]
autopilot.active=0
autopilot.maxchannels=5
autopilot.allocation=0.6

[bitcoind]
bitcoind.rpchost=127.0.0.1:8332
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=*************
bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332
bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333

#deactivate streamisolation for hybrid-mode
tor.streamisolation=false

#activate split connectivity
tor.skip-proxy-for-clearnet-targets=true

[tor]
tor.active=true
tor.v3=true
tor.privatekeypath=/mnt/hdd/lnd/v3_onion_private_key
tor.socks=9050
tor.control=9051

openoms, 2022-01-18:
these lines:
```
#deactivate streamisolation for hybrid-mode
tor.streamisolation=false

#activate split connectivity
tor.skip-proxy-for-clearnet-targets=true
```
need to be under the 
```
[tor]
```
section.

this is causing your LND to fail.

Silarous, 2022-01-18:
That was it!  Thank you sir!  Does the delete HDD data not delete those files too?  I deleted the HDD entirely twice and flashed the SD at least 10 times.  If that wasn't formatted that way by the install script it is odd that it survived multiple wipes?

openoms, 2022-01-18:
The lnd.conf is part of the rescue file as it is in the lnd directory and that's how your settings and alias is restored.

It is parsed for most breaking options, but not for line order.

Silarous, 2022-01-18:
That makes sense.  Thanks again.

commit f1ed46af (2021-12-20): provision.setup: register LNDRESCUE, fix #2639
closed 2022-02-19 by rootzoll