raspiblitz/raspiblitz #2647: Upgrade to 1.7.1

AnimaI, 2021-10-20:
Maybe I write too early and can not wait :-)
I have today refreshed my version 1.7.0 to 1.7.1.
I bought an identical SD card and flashed the image 1.7.1.

After a short start with the newly flashed SD card I got the message whether I want to upgrade. Then I came to the input of the password A.

Now I see in my terminal since about 15min.
```
"Upgrade/Recover/Provision
----> Setup System ..."
Run "tail -f ./raspiblitz.log" in new terminal to follow install logs.
```

When I open a new terminal, I get back to the same screen. How do I get into the user input so that I can type tail -f ./raspiblitz.log?

CTRL -c or ctrl -x did nothing and I can't get into the input.
I am logged in as admin.

AnimaI, 2021-10-20:
I receive the following message

```
admin@raspberrypi:~ $ tail -f ./raspiblitz.log
### BASIC SYSTEM SETTINGS ###
# Optimizing log files: rotate daily, keep 2 weeks & compress old days
Setting lightning alias: zeroBaseFee
Copy HDD content for user admin
### PREPARE LND                                       
*** Make backup of LND directory 
FAIL: Backup was not successful 
Files /mnt/hdd/lnd/data/graph/mainnet/channel.db and /mnt/hdd/backup_lnd/data/graph/mainnet/channel.db differ
removing backup dir to prevent false override
```

which backup dir do i need to remove?
What to do after that?

bubelov, 2021-11-08:
> The SD card with 1.7.1 started, but it remains for about 4h at the following message.

Same here, waiting for about 30 mins already. Will leave it for a few hours just in case it resolves by itself

rootzoll, 2021-11-30:
This seems to be related to this part of the `_provision_.sh` script:
https://github.com/rootzoll/raspiblitz/blob/5e074d0ccaeac8c6f1a54fc4bb12750244801bcf/home.admin/_provision_.sh#L163

looking why this backup was created was because of issue #324 - it was to keep tls certs the same on backup. so basically copy everything in the lnd directory might be overkill anyway - especially large files like the channel.db can delay and waste HDD/SSD space.

When looking thru the code the only this files are used from that backup directoy. Those could even be copied `/var/cache/raspiblitz`.

@openoms can you see any other reason why we should keep that full backup copy of the lnd directory during and after provision? I would recommend to slim that part down to just the TLS files.

closed 2022-02-17 by rootzoll