raspiblitz/raspiblitz #2682: BTCPayServer Update v1.3.0

rootzoll, 2021-10-29:
https://blog.btcpayserver.org/btcpay-server-1-3-0/

openoms, 2021-10-29:
tested working on v1.7.1 with a simple reinstall after editing the version in the script

openoms, 2021-10-30:
```
To update BTCPayServer to v1.3.1:

#download
wget https://raw.githubusercontent.com/openoms/raspiblitz/btcpayv1.2.3/home.admin/config.scripts/bonus.btcpayserver.sh -O config.scripts/bonus.btcpayserver.sh

# run the update
config.scripts/bonus.btcpayserver.sh update
 ```

https://github.com/rootzoll/raspiblitz/pull/2683

farahats9, 2022-02-15:
It would be nice if we could get the latest version 1.4.x for the new UI overhaul.
The only difference in the script I think is upgrading dotnet from v3.1 to v6.0

rootzoll, 2022-02-17:
Yes BTCPay will come on version v1.4.4.0  with the v1.7.2 Raspiblitz ... closing issue for release.

closed 2022-02-17 by rootzoll