raspiblitz/raspiblitz #2684: [Update] LND v0.14.0

openoms, 2021-10-29:
The v0.14.0rc1 is out
https://github.com/lightningnetwork/lnd/releases/tag/v0.14.0-beta.rc1

frennkie, 2021-10-30:
What do you think about the Postgres support? When should we consider switching (if at all)

https://github.com/lightningnetwork/lnd/blob/master/docs/release-notes/release-notes-0.14.0.md

### Postgres database support
This release adds support for Postgres as a database backend to lnd. Postgres has several advantages over the default bbolt backend:

- Better handling of large data sets.
- On-the-fly database compaction (auto vacuum).
- Database replication.
- Inspect data while lnd is running (bbolt opens the database exclusively).
- Usage of industry-standard tools to manage the stored data, get performance metrics, etc.
 
Furthermore, the SQL platform opens up possibilities to improve lnd's performance in the future. Bbolt's single-writer model is a severe performance bottleneck, whereas Postgres offers a variety of locking models. Additionally, structured tables reduce the need for custom serialization/deserialization code in lnd, saving developer time and limiting the potential for bugs.

Instructions for enabling Postgres can be found in docs/postgres.md.

openoms, 2021-10-30:
need to dig into it more. Certainly would not be a good idea to deploy it for beginners before it is well tested and maybe becomes the default for LND.

openoms, 2021-11-07:
v0.14.0rc2 is out, but suggest to wait with testing:

https://twitter.com/c_otto83/status/1457092049661009922
https://twitter.com/c_otto83/status/1457296938903261184

rootzoll, 2021-11-11:
rc3 is out: https://github.com/lightningnetwork/lnd/releases/tag/v0.14.0-beta.rc3

openoms, 2021-11-11:
- [ ] the `build_sdcars.sh` should call the `lnd.install.sh` to avoid code duplication and make updates easier

openoms, 2021-11-24:
Will need to wait for LND v0.14.1 to have this sorted: LND v.0.14.0 is incompatible with eclair and c-lightning! Please st...

benjamin-wilson, 2021-12-03:
v0.14.1 is out.
Has any consideration been given to tor.skip-proxy-for-clearnet-targets flag as default or a menu option? Seems like it could really boost routing performance for the network.

openoms, 2021-12-03:
>  v0.14.1 is out.
> Has any consideration been given to tor.skip-proxy-for-clearnet-targets flag as default or a menu option? Seems like it could really boost routing performance for the network. 

It could improve the response time to clearnet nodes, but won't help the liquidity issues which is the limiting factor in most cases.

Also it is a privacy leak as it exposes the public IPaddress to the clearnet peers so would not offer  it as default, neither in the menu.

benjamin-wilson, 2021-12-03:
> Also it is a privacy leak as it exposes the public IPaddress to the clearnet peers so would not offer it as default, neither in the menu.

It seems like it would fit in the configuration flow when selecting between Tor or clearnet as a 3rd option. I can't think of a reason you might want **only** clearnet and not this hybrid mode.

openoms, 2021-12-03:
Agree. The setup is Tor by default now and it should remain so, but there could be an option to activate this "hybrid-mode" with clear warnings re. privacy.

Opt in, not on-by-default.

benjamin-wilson, 2021-12-03:
Yes, I meant default over clearnet, as in clearnet option could be replaced by the hybrid option. Privacy first.

rootzoll, 2021-12-09:
I like the hybrid mode as an option. I can make sense to bundle/integrate this with the `Auto-NAT` option/setting .. because if you want to go hybrid with clean IP you would like to have that your public as well.

[image]

@openoms do you want to add this for v1.7.2 ... otherwise this would be a new issue for v1.8.0

openoms, 2021-12-09:
Agree that would make sense to bundle this. Should be some something like "LND Activate clearnet address" to make clear what the effect is.
The setting itself is just a config entry so easy to add. 
Would be nice if someone actually running a clearnet node would test it, @benjamin-wilson maybe?

openoms, 2021-12-10:
BTW there is an equivalent setting for C-lightning:
https://lightning.readthedocs.io/TOR.html#quick-start-on-linux
[image]

`always-use-proxy=false`

openoms, 2021-12-10:
@benjamin-wilson this should work on RPi: https://mullvad.net/en/help/linux-openvpn-installation/

I am using a simple openvpn config and autostart with another provider (not PIA): https://www.raspberrypi-spy.co.uk/2020/06/raspberry-pi-vpn-setup-guide/

benjamin-wilson, 2021-12-28:
Might be easier to do this on the router (already running pfSense). More flexible and don't have to redo when updating. Thorough tutorial by Lawrence Systems: [https://www.youtube.com/watch?v=ulRgecz0UsQ](https://www.youtube.com/watch?v=ulRgecz0UsQ)

closed 2022-02-17 by rootzoll