raspiblitz/raspiblitz #2685: RESET-LND doesn't create a new wallet

grnqrtr, 2021-10-30:
Selecting RESET-LND asked me to setup a new alias, but that was it. It looks like the menu option is supposed to run `/home/admin/70initLND.sh` but that script doesn't even seem to exist.

openoms, 2021-10-30:
Will try to reproduce. Just to confirm:
Are you running v1.7.1 and updated by reflashing the SDcard?

Also please try running:
`patch`
and see if the issue persists.

grnqrtr, 2021-10-30:
Sorry, yes, v1.7.1 on a new SDcard. I actually had already ran `patch` to fix a different issue.

openoms, 2021-11-01:
Yes, instead of 
`home/admin/70initLND.sh` 

it should be
```
/home/admin/config.scripts/lnd.install.sh on mainnet initwallet
```

Will test the fix, but you could try now.

grnqrtr, 2021-11-01:
This did work to create a new wallet, but what I actually wanted to do was restore a wallet from seed. It seems like it should give you that option?

openoms, 2021-11-01:
Will need to ping @rootzoll here because that seem to need user flow not present currently in the new setup process.

What you can do now is to delete the existing wallet:
**DANGER** `sudo rm -rf /mnt/hdd/lnd/data` **DANGER**
and use:
`lncli create`

grnqrtr, 2021-11-01:
As always, thanks for the help @openoms!

PR #2690 (2021-11-02): Fix RESET-LND
fixes #2685
closed 2021-11-02 by openoms