raspiblitz/raspiblitz #2686: [Security] git verify-commit everywhere

openoms, 2021-10-30:
Introducing a script https://github.com/rootzoll/raspiblitz/pull/2683/commits/d51440216436acbf82f16e4728f29b3ea0ec1fca  to verify the checked out git commits with the BTCpay update: https://github.com/rootzoll/raspiblitz/pull/2683

All scripts installing services built from the source code can have a header like:
```
PGPsigner="nicolasdorier"
PGPpubkeyLink="https://keybase.io/nicolasdorier/pgp_keys.asc"
PGPpubkeyFingerprint="AB4CFA9895ACA0DBE27F6B346618763EF09186FE"
```

and can run:
```
# pin version
sudo -u $installingUser git reset --hard $version
# PGP verify
sudo -u $installingUser /home/admin/config.scripts/blitz.git-verify.sh "${PGPsigner}" "${PGPpubkeyLink}" "${PGPpubkeyFingerprint}" || exit 1
```
       
to reduce trust in the download source.

Failing to verify the PGP signature on the checked commit breaks the installation.

nyxnor, 2021-10-30:
YES !!!

Kixunil, 2021-10-30:
Note that you can run `gpg --recv-keys FINGERPRINT` and so don't need to specify URL.

Also be mindful of downgrade attacks (an attacker could send a valid old signed version with known vulnerability so that he can then exploit it).

openoms, 2021-10-30:
> Note that you can run `gpg --recv-keys FINGERPRINT` and so don't need to specify URL.

yes, but some keys are not shared on the keyservers unfortunately. I find it best when people share it on their own website and then can be double checked in person, on a keyserver and maybe having some verified signatures too from other known entities ("circle of trust").

Kixunil, 2021-10-30:
So far all keys I use are on keyserver (Ubuntu), if there are any missing it may be nicer to ask people to share them.
One only needs to check fingerprint, no need to check whole key.

rootzoll, 2021-11-30:
PR merged for final testing

rootzoll, 2022-02-17:
Works so far good with RCs  .. closing foir release.

closed 2022-02-17 by rootzoll