raspiblitz/raspiblitz #2688: Upgrading to v1.7.1 requires RPC port update for LND testnet node

kaloudis, 2021-10-30:
Upon upgrading from ~v1.6 to v1.7.1 I had to ensure I updated the `lnd.conf` to use the bitcoind testnet RPC port. This should probably be handled in the migration stage if possible.

Let me know if you'd like any more info. Cheers.

openoms, 2021-10-30:
Thank you for reporting. I am aware that the upgrade from a testnet node is not handled well due to the change to parallel miannet - testnet - signet usage now (since 1.7.1).
Will think how can we improve this.

Can you share which config settings you needed to change manually?
@frennkie had a similar issue.
Let's aim to handle them automatically.

kaloudis, 2021-10-30:
`bitcoind.rpchost` in `lnd.conf`

Seems like it gets reset often too

openoms, 2021-10-30:
yes, it will be reset after every lnd restart by the precheck script.

Set the default `lnd.conf` for mainnet
and there will be a `tlnd.conf` for testnet when you activate the `TESTNETS` in `SETTINGS` as in https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/lnd.install.sh

The default password for tetsnet LND node is: `raspiblitz`.

cstenglein, 2023-06-07:
@kaloudis  is this still an issue with 1.9.0?

Closing this for now, please re-open if it still persists.

closed 2023-06-07 by cstenglein