raspiblitz/raspiblitz #2689: IP2TOR Subscription Failing

DocCodez, 2021-10-31:
When attempting to use any bridge the subscription fails after payment is sent? "Error: Fail on subscription storage" [image][image]

rootzoll, 2021-10-31:
On the first view this seems to be a local issue because it has problems of storing the subscription to the toml file.
Can you take a look if the following file already exists `/mnt/hdd/app-data/subscriptions/subscriptions.toml` and if yes if contains text data .. can you check if its valid toml with validators like this: https://www.toml-lint.com

Another thing to check is also the amount of free space on the HDD ... what does "HDD use" on the LCD display say?

cubanhodl, 2021-10-31:
The HDD use is at 456GB (27%) so that doesn't seem to be the issue. However, the subscriptions.toml file is not there.

openoms, 2021-11-08:
is this solved? IP2Tor subscriptions are tested ok. Did you lose any sats @cubanhodl ?

cubanhodl, 2021-11-08:
This is not solved and I did in the process of attempting to open a bridge. It would pay the invoice and then attempt to open the bridge before getting the error above. @rootzoll

rootzoll, 2021-11-11:
> However, the subscriptions.toml file is not there.

Maybe its a problem with the script not being able to write into `subscriptions` directory. @cubanhodl can you post the result of the following command:
```
sudo ls -la /mnt/hdd/app-data | grep subscriptions
```

frennkie, 2021-11-13:
@cubanhodl the Subscription was created on Oct. 31th and ran for 24 hours. So I believe that @rootzoll 's idea is right: Please check the file/directory permissions on the path where the subscriptions are saved.

cubanhodl, 2021-11-14:
This is the result of that command.
drwxr-xr-x  2 bitcoin     bitcoin     4096 Jun 23 21:36 subscriptions

@rootzoll

rootzoll, 2021-11-25:
@cubanhodl can you run a `sudo chown admin:admin /mnt/hdd/app-data/subscription` and then check again if subscriptions can be created?

rootzoll, 2021-12-07:
OK for v1.7.2 I added a command that on bootstrap the `/mnt/hdd/app-data/subscriptions` directory will always be set to owner admin. Hope this prevents such errors in the future ... closing for v1.7.2 release.

closed 2021-12-07 by rootzoll