raspiblitz/raspiblitz #2691: detect c-lightning wallet locked

rootzoll, 2021-11-01:
In the systembackgound scan thats feeding the status cache c-lightning is seen as always unlocked - this needs to be implemented for v1.8

openoms, 2021-11-01:
Currently the locked state causes the general error `ERROR_TALKING_TO_LIGHTNINGD` and exits with error code 2:
https://github.com/ElementsProject/lightning/blob/3d8293dbf40d2d940a89850ecc87b8c6e1fa0772/cli/lightning-cli.c#L26

maximlomans, 2021-11-04:
ich bekomme genau den gleichen fehler

still at 98% sync
autounlock is enabled
can i edit c-lightning/config to point towards 9735 instead of 9736 or is it supposed to be that way,  ..in a node with both it and lnd.

\# lightningd configuration for bitcoin mainnet

network=bitcoin
announce-addr=127.0.0.1:9736
log-file=cl.log
log-level=info
plugin-dir=/home/bitcoin/cl-plugins-enabled

 \# Tor settings
proxy=127.0.0.1:9050
bind-addr=127.0.0.1:9736
addr=statictor:127.0.0.1:9051/torport=9736
always-use-proxy=true
 
i guess only one of them runs at any one time anyways, so technically i could but that means i'd have to edit all port entries in .lightning/config
and add ufw 9736/tcp allow entry, there isn't one.

i couldn't say it got temp 'fixed' by enabling LND instead.
but still occasionally getting statuscheck error code 124

openoms, 2021-11-04:
@maximlomans please open a new issue for this as you question is not related here.
(you can't run LND and c-lightning on the same port,  but can edit the c-lightning port to.9735 if LND is deactivated )

openoms, 2022-02-09:
added the detection from logs, but keeping the v1.8 milestone as in the next c-lightning release will be able to support to specific exit code introduced in https://github.com/ElementsProject/lightning/pull/4908

rootzoll, 2022-06-05:
Worked on my testing machine ... closing issue for v1.8.0 release.

closed 2022-06-05 by rootzoll