raspiblitz/raspiblitz #2692: LND sync stuck after upgrade to 1.7.1

johnnyfox87, 2021-11-02:
Hi,
my lightning synchronization seems to be stuck after upgrade from 1.6.2 to 1.7.1 similar to #2675. 
[image]

I my case there seems to be a problem with the TLS certificate. This is the repeated output after `sudo journalctl -f -u lnd`:
```
Nov 02 10:13:52 raspberrypi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Nov 02 10:13:52 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 1539.
Nov 02 10:13:52 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
Nov 02 10:13:52 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Nov 02 10:13:52 raspberrypi systemd[1]: Started LND Lightning Daemon.
Nov 02 10:13:52 raspberrypi lnd[28433]: unable to load TLS credentials: tls: private key does not match public key
Nov 02 10:13:52 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 02 10:13:52 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
```

This is the complete debug: https://www.toptal.com/developers/hastebin/nerugupova.yaml

My LND wallet is empty, so I guess I can just reset it.

openoms, 2021-11-02:
Try CONNECT - RESET
to reset the tls.cert instead of starting a new wallet.

johnnyfox87, 2021-11-02:
Ok that fixed it. Thanks!

closed 2021-11-02 by johnnyfox87