raspiblitz/raspiblitz #2694: sudo: unable to initialize PAM: Too many open files

openoms, 2021-11-02:
Starting a new node with existing blockchain fails:

```
###########################################################
# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Tue 19 Oct 11:06:31 BST 2021
***********************************************
# *** blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
No HDMI switch found. 
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No SSHRESET switch found. 
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 604
drwxr-xr-x   2 root root   4096 Oct 12 00:12 .
drwxr-xr-x 118 root root   4096 Oct 12 00:12 ..
-rw-r--r--   1 root root 565189 Jan 31  2020 moduli
-rw-r--r--   1 root root   1580 Jan 31  2020 ssh_config
-rw-r--r--   1 root root   3250 Jan 31  2020 sshd_config
-rw-------   1 root root   1381 Oct 12 00:12 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Oct 12 00:12 ssh_host_dsa_key.pub
-rw-------   1 root root    513 Oct 12 00:12 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Oct 12 00:12 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Oct 12 00:12 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Oct 12 00:12 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Oct 12 00:12 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Oct 12 00:12 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Mar  5  2021 ssh_import_id
wifiIsSet=0
hddBackupLocationAvailable=0
hddRestoreConfigAvailable=0
memRestoreConfigAvailable=0
wifiRestore=0
wifiBackup=0
No UASP FIX needed (2nd-try).
isMounted: 0
Skipping UASP deactivation ... cmdlineExists(1) hddAdapterUSB(174c:0825) hddAdapterUSAP(1)
HDD already part of system: 0
Temp mounting (1) data drive (sda)
Temp mounting (1) result: 0
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
# df 
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30353628 4384168  24688772  16% /
devtmpfs         3991184       0   3991184   0% /dev
tmpfs            4025808       0   4025808   0% /dev/shm
tmpfs            4025808    8628   4017180   1% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs            4025808       0   4025808   0% /sys/fs/cgroup
tmpfs              32768       4     32764   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095   30357    227738  12% /boot
tmpfs             805160       0    805160   0% /run/user/1000
# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda                
mmcblk0            
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=0
isBTRFS=0

# SETUP INFO
isSSD=0
hddCandidate='sda'
hddBytes=1000204886016
hddGigaBytes=931
hddPartitionCandidate=''
hddAdapterUSB='174c:0825'
hddAdapterUSAP=1

# RAID
isRaid=0
raidCandidates=0

# SWAP
isSwapExternal=0

Displaying Info Message: Please Login for Setup
## WAIT LOOP: USER SETUP/UPDATE/MIGRATION
# Sourcing /mnt/hdd/raspiblitz.conf 
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup 
# RASPIBLITZ SETUP STATE
setPasswordA=1
setPasswordB=1
setPasswordC=1
hostname=presync
lightning=none
network=bitcoin
chain=main
# df 
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30353628 4385772  24687168  16% /
devtmpfs         3991184       0   3991184   0% /dev
tmpfs            4025808       0   4025808   0% /dev/shm
tmpfs            4025808   25028   4000780   1% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs            4025808       0   4025808   0% /sys/fs/cgroup
tmpfs              32768      20     32748   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095   30357    227738  12% /boot
tmpfs             805160       0    805160   0% /run/user/1000
tmpfs             805160       0    805160   0% /run/user/1001
# lsblk -o NAME,FSTYPE,LABEL 
# Update /home/admin/raspiblitz.info 
state=provision
message='Starting Provision'
baseimage=raspios_arm64
cpu=aarch64
board=rp4
ramMB=7862
ramGB=8
network=
chain=
localip='192.168.3.156'
online='1'
dnsworking=1
fsexpanded=1
displayClass=lcd
displayType=
setupStep=0
setupPhase='setup'
fundRecovery=0
hostname=
hddCandidate='sda'
hddGigaBytes=931
hddBlocksBitcoin=
hddBlocksLitecoin=
hddGotMigrationData=
# migrationFile()
# setting PASSWORD A
Calling _provision.setup.sh for basic setup tasks ..
EXIT _provision.setup.sh BECAUSE OF ERROR STATE (4)
This can also happen if _provision.setup.sh has syntax errors
# /home/admin/raspiblitz.provision-setup.log

###################################
# _provision.setup.sh
###################################
*** Prepare bitcoin ***
mkdir: cannot create directory ‘/mnt/hdd/bitcoin’: No such file or directory
chown: cannot access '/mnt/hdd/bitcoin': No such file or directory
sudo: unable to initialize PAM: Too many open files
sudo: unable to initialize PAM: Too many open files
chown: cannot access '/mnt/hdd/bitcoin/bitcoin.conf': No such file or directory
error='no data drive mounted'
File Exists: /mnt/hdd/bitcoin/bitcoin.conf --> 0
# setting PASSWORD B
# OK -> RPC Password B changed
# Reboot is needed
...
*** Start bitcoin (SETUP) ***
- This can take a while ..
FAIL: setup: failed bitcoin
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(_provision.setup.sh fail)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
```

openoms, 2021-11-02:
Regarding the blockchain testrun failing the reason is bitcoin being restarted continously: 
```
sudo journalctl -fu bitcoind
-- Logs begin at Tue 2021-11-02 12:44:01 GMT. --
Nov 02 12:50:30 raspberrypi systemd[1]: Stopped Bitcoin-Daemon.
Nov 02 12:50:30 raspberrypi systemd[1]: Starting Bitcoin-Daemon...
Nov 02 12:50:30 raspberrypi systemd[1]: Started Bitcoin-Daemon.
Nov 02 12:50:45 raspberrypi systemd[1]: bitcoind.service: Main process exited, code=exited, status=1/FAILURE                                                                                                      
Nov 02 12:50:45 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Nov 02 12:51:15 raspberrypi systemd[1]: bitcoind.service: Service RestartSec=30s expired, scheduling restart.                                                                                                     
Nov 02 12:51:15 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 9.
Nov 02 12:51:15 raspberrypi systemd[1]: Stopped Bitcoin-Daemon.
Nov 02 12:51:15 raspberrypi systemd[1]: Starting Bitcoin-Daemon...
Nov 02 12:51:15 raspberrypi systemd[1]: Started Bitcoin-Daemon.
Nov 02 12:51:29 raspberrypi systemd[1]: bitcoind.service: Main process exited, code=exited, status=1/FAILURE                                                                                                      
Nov 02 12:51:29 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
```

openoms, 2021-11-02:
`sudo tail -n 30 -f /mnt/hdd/bitcoin/debug.log`

shows the reason:

```
2021-11-02T12:53:29Z init message: Loading block index…
2021-11-02T12:53:29Z Switching active chainstate to Chainstate [ibd] @ height -1 (null)
2021-11-02T12:53:29Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2021-11-02T12:53:29Z Opened LevelDB successfully
2021-11-02T12:53:29Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2021-11-02T12:53:41Z LoadBlockIndexDB: last block file = 2788
2021-11-02T12:53:41Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=65, size=82732850, heights=707767...707831, time=2021-11-01...2021-11-02)
2021-11-02T12:53:41Z Checking all blk files are present...
2021-11-02T12:53:42Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2021-11-02T12:53:42Z Fatal LevelDB error: Corruption: 2387 missing files; e.g.: /mnt/hdd/bitcoin/chainstate/532744.ldb
2021-11-02T12:53:42Z You can use -debug=leveldb to get more complete diagnostic messages
2021-11-02T12:53:42Z Fatal LevelDB error: Corruption: 2387 missing files; e.g.: /mnt/hdd/bitcoin/chainstate/532744.ldb
2021-11-02T12:53:42Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2021-11-02T12:53:42Z Aborted block database rebuild. Exiting.
2021-11-02T12:53:42Z Shutdown: In progress...
2021-11-02T12:53:42Z scheduler thread exit
2021-11-02T12:53:42Z Shutdown: done
```

So seems like it is becuase of corruption of the data on the disk.

openoms, 2021-11-02:
when this happens could be useful to dump the bitcoin logs for the user rather than just suggesting to "report to the developers" since bitcoind itself does propose the solution (the need to reindex) .

rootzoll, 2021-11-11:
This is also addressed in this issue: #2655

rootzoll, 2021-11-11:
Could be a follow-up issue for #2655 after 1.7.2

rootzoll, 2021-12-17:
Its a bit more complicated to catch the correct error (different one can occur) and give more details to the user.
Pushing this to v1.8