raspiblitz/raspiblitz #2695: Multiple RTL nodes

thephizzz, 2021-11-02:
I originally posted this as an RTL issue, but they suggested I bring it here.
I have lnd running on a Raspiblitz with RTL and it operates correctly.
RTL is running as a service on the Raspi.

I've added another lnd node running on a W10 PC on the same network.
After editing the RTL-Config.json file to add the new node and stopping and starting the RTL service, the second node information _is removed_ from the RTL-Config.json file and the second node is not recognized.
This happens as soon as I restart the RTL service with: sudo systemctl start RTL
I've confirmed several times and made sure that I was working on the correct copy of RTL-Config.json.
I'm assuming there is an error in the config file but have tried all options that I know of.
(Of note is that my second lnd node does not have a loop.macaroon file so I omitted that line in the Authentication section.)
I have attached my multi RTL-Config.json file for review and have added: restlisten=192.xxx.x.xxx:8080 to the second lnd.conf file.
Thank you for any insights.

RTL-Config.json
{
"port": "3000",
"defaultNodeIndex": 1,
"SSO": {
"rtlSSO": 0,
"rtlCookiePath": "",
"logoutRedirectLink": ""
},
"nodes": [
{
"index": 1,
"lnNode": "raspivoice",
"lnImplementation": "LND",
"Authentication": {
"macaroonPath": "/home/rtl/.lnd/data/chain/bitcoin/mainnet/",
"configPath": "/home/rtl/.lnd/lnd.conf",
"swapMacaroonPath": "/home/rtl/.loop/mainnet/",
"boltzMacaroonPath": "/home/rtl/.boltz-lnd/macaroons/"
},
"Settings": {
"userPersona": "OPERATOR",
"themeMode": "DAY",
"themeColor": "PINK",
"channelBackupPath": "/home/rtl/RTL-SCB-backup-raspivoice",
"logLevel": "ERROR",
"lnServerUrl": "https://localhost:8080",
"swapServerUrl": "https://localhost:8081",
"boltzServerUrl": "https://localhost:9003",
"fiatConversion": false
}
},
{
"index": 2,
"lnNode": "lnd XPS",
"lnImplementation": "LND",
"Authentication": {
"macaroonPath": "/mnt/hdd/xps",
"boltzMacaroonPath": "/mnt/hdd/xps"
},
"Settings": {
"channelBackupPath": "/home/rtl/RTL-SCB-backup-lndXPS",
"bitcoindConfigPath": "",
"lnServerUrl": "https://192.168.1.166:8080",
"boltzServerUrl": "https://192.168.1.166:9003",
"userPersona": "OPERATOR",
"themeMode": "DAY",
"themeColor": "TEAL",
"fiatConversion": false,
"logLevel": "ERROR"
}
}
],
"multiPassHashed": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
}

thephizzz, 2021-11-02:
I've stopped the service and started the rtl with node rtl.
now the app comes up. But times out in attempting to access the second node.
But the RTL-Config.json file remained intact.

Then I exited. re-enabled the service and when I start the service the RTL-Config.json file is truncated again, removing the second node information.

openoms, 2021-11-02:
This won't be trivial to do.
The RTL config is reconfigured on every restart 
by 
`/home/admin/config.scripts/bonus.rtl.sh prestart` 
as you can see in the `rtl.service`
You can comment out or remove the `Prestart=` line in the `rtl.service` to preserver your custom changes on restart.

thephizzz, 2021-11-02:
Thanks – this is my RTL.service. Are you referring to the ExecStartPre line?

 

# Systemd unit for RTL

 

[Unit]

Description=RTL Webinterface

Wants=lnd.service

After=lnd.service

 

[Service]

Environment="RTL_CONFIG_PATH=/home/rtl/RTL/"

ExecStartPre=-/home/admin/config.scripts/bonus.rtl.sh prestart lnd mainnet

ExecStart=/usr/bin/node /home/rtl/RTL/rtl

User=rtl

Restart=always

TimeoutSec=120

RestartSec=30

StandardOutput=null

StandardError=journal

 

# Hardening measures

PrivateTmp=true

ProtectSystem=full

NoNewPrivileges=true

PrivateDevices=true

 

[Install]

WantedBy=multi-user.target

openoms, 2021-11-07:
>Thanks – this is my RTL.service. Are you referring to the ExecStartPre line?

yes, edited above

cstenglein, 2023-06-07:
Seems resolved => Closed, please re-open otherwise

closed 2023-06-07 by cstenglein