raspiblitz/raspiblitz #2696: [Security][Research] doas as replacement for sudo

openoms, 2021-11-03:
Listening to https://anchor.fm/citadeldispatch/episodes/Citadel-Dispatch-e42---security-focused-bitcoin-nodes-with-nixbitcoinorg--n1ckler--and-seardsalmon-e19mcg1
with the nixbitcoin.org team. 

Using the minimalistic `doas` command instead of `sudo` gets a detailed mention.

Does is not readily available on my system (but should be on Debian Bullseye - https://wiki.debian.org/Doas) so will just start this here as a placeholder for further research.

rootzoll, 2021-11-11:
Could be a follow-up issue for #2655 after 1.7.2 ... marking for 1.8

nyxnor, 2021-11-30:
Simple instructions by the creator of doas [here](https://flak.tedunangst.com/post/doas-mastery)
Debian Doas [source code](https://salsa.debian.org/debian/doas)
Debian [doas.conf](https://manpages.debian.org/bullseye/doas/doas.conf.5.en.html)

It is better to refer to debian docs about doas because it is not equal as the OpenBSD doas, becase of [this](https://salsa.debian.org/debian/doas#about-the-port):

> This is not an official port/project from OpenBSD!
As much as possible I've attempted to stick to doas as tedu desired
it. As things stand it's essentially just code lifted from OpenBSD with
PAM or shadow based authentication glommed on to it.
Compatibility functions in libopenbsd come from openbsd directly
(strtonum.c, reallocarray.c, strlcpy.c, strlcat.c),
from openssh (readpassphrase.c) or from sudo (closefrom.c).
The PAM and shadow authentication code does not come from the OpenBSD project.

> Perist/Timestamp/Timeout
The persist feature is disabled by default and can be enabled with the configure
flag --with-timestamp.
This feature is new and potentially dangerous, in the original doas, a kernel API
is used to set and clear timeouts. This API is openbsd specific and no similar API
is available on other operating systems.
As a workaround, the persist feature is implemented using timestamp files
similar to sudo.
See the comment block in timestamp.c for an in-depth description on how
timestamps are created and checked to be as safe as possible.

I am using doas on home computer and on my Onion project, the `privilege_command` is a variable that can be `sudo` or `doas` to call any commands, so we can migrate easily or fallback.

`sudo apt install -y doas` > funny we need sudo, else need to login as root

The `/etc/doas.conf` needs to be created:

`printf "permit nopass keepenv admin\n" | sudo tee -a /etc/doas.conf`

it is versatile to permit calling just certain commands or certain arguments:

`permit nopass keepenv userexample cmd apt args update`

We can check if user is permitted to do an action or not checking the config file with

`doas -C /etc/doas.conf -u admin sh`

This

The basic usage and options are almost the same as in sudo

`doas whoami` will execute doas as root
`doas -u admin whoami` will execute doas as admin
`doas -u admin -s` will call the admin shell

nyxnor, 2021-12-02:
doas did not work as I expected, it sees I have permission
```
$ doas -C /etc/doas.conf usermod -aG debian-tor nyxnor
permit nopass
$ doas usermod -aG debian-tor nyxnor
doas: usermod: command not found
$ sudo usermod -aG debian-tor nyxnor
# not output, it works, can run $ echo $? to check also
```
So it does not detect usermod...
```
$ sudo usermod --help
Usage: usermod [options] LOGIN

Options:
  -b, --badnames                allow bad names
  -c, --comment COMMENT         new value of the GECOS field
  -d, --home HOME_DIR           new home directory for the user account
  -e, --expiredate EXPIRE_DATE  set account expiration date to EXPIRE_DATE
  -f, --inactive INACTIVE       set password inactive after expiration
  [  ... output ommited for clarity ]

$ doas usermod --help
doas: usermod: command not found
```

nyxnor, 2021-12-02:
How nix-bitcoin uses doas: if security config for doas is enabled, use it, else fallback to sudo
https://github.com/fort-nix/nix-bitcoin/blob/747019a9e9a9d9b100c6b104bf54f5da344926c3/modules/nix-bitcoin.nix#L37
we can harden blitz by setting this also.

nyxnor, 2021-12-03:
Actually, usermod just works when I specify its entire path:
```sh
$ doas /usr/sbin/usermod 
doas (nyxnor@deb) password: 
Usage: usermod [options] LOGIN

Options:
  -b, --badnames                allow bad names
  -c, --comment COMMENT         new value of the GECOS field
  -d, --home HOME_DIR           new home directory for the user account
  -e, --expiredate EXPIRE_DATE  set account expiration date to EXPIRE_DATE
```
This means that it has something to do with the path.