raspiblitz/raspiblitz #2697: NBXplorer stuck

openoms, 2021-11-03:
from @JurGi7:
I have the same problem, after updating to 1.7.1 I often see the message "Waiting for Chain Sync". Furthermore, I see on the BTCPayserver interface for two weeks "Your Node are synching". In the meantime, however, it works and seems to be synchronous. What can I do?

@openoms:
 check if
`sudo systemctl status nbxplorer`
and
`sudo journalctl -n10 -u nbxplorer`
has any errors.
Restart the services with:
`sudo systemctl restart nbxplorer`
`sudo systemctl restart btcpayserver`

from @JurGi7:
Hi openoms,
thank you for your answer.
Here are the outputs on the commands. nbxplorer and btcpay restarted.
Jurgi

first:
```
nbxplorer.service - NBXplorer daemon
Loaded: loaded (/etc/systemd/system/nbxplorer.service; enabled; vendor preset: enabled)
Active: active (running) since Tue 2021-11-02 17:20:49 CET; 17h ago
Main PID: 3485 (dotnet)
Tasks: 38 (limit: 4532)
CGroup: /system.slice/nbxplorer.service
├─3485 /home/btcpay/dotnet/dotnet run --no-launch-profile --no-build -c Release -p NBXplorer/NBXplorer.csproj --
└─3988 /home/btcpay/NBXplorer/NBXplorer/bin/Release/netcoreapp3.1/NBXplorer

Nov 03 10:49:21 raspberrypi dotnet[3485]: info: Configuration: BTC: Height: 708006
Nov 03 10:49:21 raspberrypi dotnet[3485]: info: Explorer: BTC: Handshaked node
Nov 03 10:49:21 raspberrypi dotnet[3485]: info: Explorer: BTC: Current Index Progress not found, start syncing from the header's chain tip (At height: 708006)
Nov 03 10:49:21 raspberrypi dotnet[3485]: info: Explorer: BTC: Starting scan at block 708006
Nov 03 10:49:21 raspberrypi dotnet[3485]: fail: Explorer: BTC: Connection unexpectedly failed: Unexpected exception while connecting to socket
Nov 03 10:49:21 raspberrypi dotnet[3485]: info: Events: BTC: Node state changed: NotStarted => NBXplorerSynching
Nov 03 10:49:21 raspberrypi dotnet[3485]: info: Events: BTC: Node state changed: NBXplorerSynching => NotStarted
Nov 03 10:49:21 raspberrypi dotnet[3485]: info: Configuration: BTC: Testing RPC connection to http://localhost:8332/
Nov 03 10:49:21 raspberrypi dotnet[3485]: info: Configuration: BTC: RPC connection successful
Nov 03 10:49:21 raspberrypi dotnet[3485]: info: Configuration: BTC: Full node version detected: 220000
```
second:
```
    Logs begin at Wed 2021-11-03 06:56:06 CET, end at Wed 2021-11-03 10:51:10 CET. --
    Nov 03 10:51:10 raspberrypi dotnet[3485]: info: Explorer: BTC: Current Index Progress not found, start syncing from the header's chain tip (At height: 708006)
    Nov 03 10:51:10 raspberrypi dotnet[3485]: info: Explorer: BTC: Starting scan at block 708006
    Nov 03 10:51:10 raspberrypi dotnet[3485]: info: Events: BTC: Node state changed: NotStarted => NBXplorerSynching
    Nov 03 10:51:10 raspberrypi dotnet[3485]: info: Events: BTC: Node state changed: NBXplorerSynching => Ready
    Nov 03 10:51:10 raspberrypi dotnet[3485]: fail: Explorer: BTC: Connection unexpectedly failed: Unexpected exception while connecting to socket
    Nov 03 10:51:10 raspberrypi dotnet[3485]: info: Events: BTC: Node state changed: Ready => NotStarted
    Nov 03 10:51:10 raspberrypi dotnet[3485]: info: Configuration: BTC: Testing RPC connection to http://localhost:8332/
    Nov 03 10:51:10 raspberrypi dotnet[3485]: info: Configuration: BTC: RPC connection successful
    Nov 03 10:51:10 raspberrypi dotnet[3485]: info: Configuration: BTC: Full node version detected: 220000
    Nov 03 10:51:10 raspberrypi dotnet[3485]: info: Configuration: BTC: Loading chain from cache...
```

openoms, 2021-11-03:
@JurGi7 as you can see NBXplorer does have an error.
Can you check bitcoind if it has any problems (SYSTEM - bitcoinLOGS). All other services are fine?

This is usually doe to a blockchain data corruption on the disk which would need a full resync, but can try reisntlling NBXplorer with the latest version: https://github.com/rootzoll/raspiblitz/issues/2682#issuecomment-955180777

JurGi7, 2021-11-03:
here is the bitcoinLOG:
2021-11-03T10:07:49Z UpdateTip: new best=00000000000000000006e0ac3aa86d49c42659c058eccbaf001daf514db08e4d height=708009 version=0x20e00004 log2_work=93.153722 tx=683637611 date='2021-11-03T10:07:22Z' progress=1.000000 cache=105.2MiB(800233txo)
2021-11-03T10:16:42Z New outbound peer connected: version: 70016, blocks=708009, peer=47425 (outbound-full-relay)
2021-11-03T10:21:35Z Socks5() connect to 2n47ftslifgm34cydhpbd2wpkur45yrfauxz2yvbszzd5ylzus3swvqd.onion:8333 failed: host unreachable
2021-11-03T10:23:56Z New outbound peer connected: version: 70016, blocks=708009, peer=47785 (block-relay-only)
2021-11-03T10:27:47Z Socks5() connect to pocnnefpau77tcqfj7ryvq7suillyy24dt2mxfoltddftmvrbohn5nyd.onion:8333 failed: host unreachable
2021-11-03T10:32:17Z Socks5() connect to b7y36uafqdczbca2agka22dc4cv34frrkgcp7zazkjizbbvw7tnfhzqd.onion:8333 failed: host unreachable
2021-11-03T10:42:33Z Socks5() connect to dfw375avjdbfsph5edsp5b6oy72v7xjlkk3vjzl3k5r5zvyxbabls6id.onion:8333 failed: host unreachable
2021-11-03T10:43:47Z Potential stale tip detected, will try using extra outbound peer (last tip update: 2158 seconds ago)
2021-11-03T10:43:55Z New outbound peer connected: version: 70016, blocks=708009, peer=48780 (outbound-full-relay)
2021-11-03T10:45:06Z Socks5() connect to wce65y6tvyu6zi3b2ry2v6gxcd77dp6si2sf7lqh4ovn7gmhyjbr6oid.onion:8333 failed: host unreachable
2021-11-03T10:45:24Z New outbound peer connected: version: 70016, blocks=708009, peer=48853 (block-relay-only)
2021-11-03T10:48:07Z New outbound peer connected: version: 70016, blocks=708009, peer=48987 (block-relay-only)
2021-11-03T10:48:14Z UpdateTip: new best=00000000000000000008c168f86ebcfa510d042e7927f124d49516f390cc9803 height=708010 version=0x20200004 log2_work=93.153734 tx=683641114 date='2021-11-03T10:47:31Z' progress=1.000000 cache=107.1MiB(815656txo)
2021-11-03T10:48:14Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2021-11-03T10:56:47Z New outbound peer connected: version: 70016, blocks=708010, peer=49401 (block-relay-only)
2021-11-03T11:04:31Z UpdateTip: new best=000000000000000000074895835a92558509dae72338c78d9a30c3da28f32e4a height=708011 version=0x20800004 log2_work=93.153746 tx=683644378 date='2021-11-03T11:03:59Z' progress=1.000000 cache=108.3MiB(825366txo)
2021-11-03T11:10:13Z New outbound peer connected: version: 70016, blocks=708011, peer=50064 (block-relay-only)
2021-11-03T11:12:36Z New outbound peer connected: version: 70016, blocks=708011, peer=50186 (block-relay-only)
2021-11-03T11:21:13Z Socks5() connect to 754f4whncohm27cihfky6zkuop2ej7izaijenjqnvopfrkte32zll6qd.onion:8333 failed: host unreachable
2021-11-03T11:21:19Z New outbound peer connected: version: 70016, blocks=708011, peer=50633 (outbound-full-relay)
2021-11-03T11:21:30Z New outbound peer connected: version: 70016, blocks=708011, peer=50643 (block-relay-only)
2021-11-03T11:22:00Z UpdateTip: new best=000000000000000000034973bb84f085800a054d06d410839fa1bc1f87a2c215 height=708012 version=0x20c00004 log2_work=93.153759 tx=683647772 date='2021-11-03T11:21:21Z' progress=1.000000 cache=109.4MiB(833749txo)
2021-11-03T11:22:29Z Socks5() connect to ohnjxfrilm3zp5ud6hgtzy5ajuk4qpkamepbj3cdlmtrdbj7zcyeoiqd.onion:8333 failed: host unreachable
2021-11-03T11:34:44Z New outbound peer connected: version: 70016, blocks=708012, peer=51287 (block-relay-only)
2021-11-03T11:43:53Z Socks5() connect to sssvc7wqg4mgfcjeooi3svqelbax3h55ycznu4qoht5wfqenotuhnyad.onion:8333 failed: host unreachable
2021-11-03T11:45:05Z Socks5() connect to utwbcu322ylagdylakpf2pvopcr45etlb2migwnvfmvur4vvy2mxfiqd.onion:8333 failed: host unreachable
2021-11-03T11:46:54Z Socks5() connect to bgoofclrijvgxwa3seqgizwg46a6l77dzsf7bwe2fp4ehpawqaib5vid.onion:8333 failed: host unreachable
2021-11-03T11:47:23Z Socks5() connect to noxhg75d6poiikzf4b2hgziom6he5im64h7tqxyfsqbnmssjbf3tvhyd.onion:8333 failed: host unreachable
2021-11-03T11:49:54Z Socks5() connect to s4kb4ntkjh3kopcqofwaprrgbwpfd4wttsjihvrnf5itcdblbrxcwqyd.onion:8333 failed: host unreachable
2021-11-03T11:53:43Z UpdateTip: new best=0000000000000000000baf825393a20a5356874f85e1854fb034377986ac4d9d height=708013 version=0x2000a004 log2_work=93.153771 tx=683650906 date='2021-11-03T11:53:13Z' progress=1.000000 cache=110.7MiB(844247txo)

how can i reinstall NBXplorer?
Thx

openoms, 2021-11-03:
Reinstall BTCPayServer (consider the new version linked above), it contains nbxplorer.

JurGi7, 2021-11-04:
BTCPayserver update to v1.3.1.0 made. but still get "node are synching" messages on the web interface. Otherwise it looks good, maybe it will take some time?

Blockhash is right: 

> UpdateTip: new best=00000000000000000009c8fcdded7291d50c72bbd3795fdc59f050ab864f1eb4 height=708160 version=0x2000a004 log2_work=93.155560 tx=683927667 date='2021-11-04T10:20:50Z' progress=1.000000 cache=105.1MiB(798461txo
[image]
)
[image]
[image]

openoms, 2021-11-07:
Is this still there after a restart?

laurentdebricon, 2021-11-07:
hello, i am also confirming this bug.

```
admin@raspberrypi:~ $ sudo journalctl -n10 -u nbxplorer
-- Logs begin at Sun 2021-11-07 18:33:10 GMT, end at Sun 2021-11-07 19:47:32 GMT. --
Nov 07 19:47:31 raspberrypi dotnet[3050]: info: Explorer:       BTC: Current Index Progress not found, start syncing from the header's chain tip (At height: 708675)
Nov 07 19:47:31 raspberrypi dotnet[3050]: info: Explorer:       BTC: Starting scan at block 708675
Nov 07 19:47:31 raspberrypi dotnet[3050]: info: Events:         BTC: Node state changed: NotStarted => NBXplorerSynching
Nov 07 19:47:31 raspberrypi dotnet[3050]: info: Events:         BTC: Node state changed: NBXplorerSynching => Ready
Nov 07 19:47:31 raspberrypi dotnet[3050]: fail: Explorer:       BTC: Connection unexpectedly failed: Unexpected exception while connecting to socket
Nov 07 19:47:31 raspberrypi dotnet[3050]: info: Events:         BTC: Node state changed: Ready => NotStarted
Nov 07 19:47:31 raspberrypi dotnet[3050]: info: Configuration:  BTC: Testing RPC connection to http://localhost:8332/
Nov 07 19:47:31 raspberrypi dotnet[3050]: info: Configuration:  BTC: RPC connection successful
Nov 07 19:47:31 raspberrypi dotnet[3050]: info: Configuration:  BTC: Full node version detected: 220000
Nov 07 19:47:32 raspberrypi dotnet[3050]: info: Configuration:  BTC: Loading chain from cache...

```

this line is not looking good.
```
Nov 07 19:47:31 raspberrypi dotnet[3050]: fail: Explorer: BTC: Connection unexpectedly failed: Unexpected exception while connecting to socket
```

this is a fresh install 1.7.1 from 1.7.0 then this update install script for latest btcpayserver
https://github.com/rootzoll/raspiblitz/issues/2682#issuecomment-955180777

then reboot, and still this message.
i read somewhere with google you should whitelist 127.0.0.1  ... https://githubmemory.com/repo/dgarage/NBXplorer/issues/311
don't know where yet ....

JurGi7, 2021-11-08:
the error is still there. "chain sync error" on the infoscreen as well the btcpayserver sync messages

openoms, 2021-11-08:
ok, try adding to the `bitcoin.conf` (menu - SYSTEM  - BITCOINCONF):
`whitelist=download@127.0.0.1` 
and restart bitcoind and nbxplorer (or the whole node).

This line gets added for electrs already, but could add it by default.

openoms, 2021-11-08:
ok, please run the commands in https://github.com/rootzoll/raspiblitz/issues/2682#issuecomment-955180777 again

openoms, 2021-11-08:
quick fix:
```
    # whitelist localhost in bitcoind
    if ! sudo grep -Eq "^whitelist=127.0.0.1" /mnt/hdd/bitcoin/bitcoin.conf;then
      echo "whitelist=127.0.0.1" | sudo tee -a /mnt/hdd/bitcoin/bitcoin.conf
      echo "# Restarting bitcoind"
      sudo systemctl restart bitcoind
    fi
   
    sudo systemctl restart nbxplorer
```

JurGi7, 2021-11-11:
btcpay update to 1.3.3 made several days ago. whitelist=127.0.0.1 already existed in the conf. still the same sync messages on the btcpay web interface. there are other suggested solutions? do you want to connect to my node?

openoms, 2021-11-12:
@JurGi7 can you share your bitcoin.conf (scrub the password)?

JurGi7, 2021-11-12:
# bitcoind configuration
# mainnet/testnet
testnet=0
# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=0
# Connection settings
rpcuser=raspibolt
rpcpassword=XXXXXXXXXXX
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
# Raspberry Pi optimizations
dbcache=1024
maxorphantx=10
maxmempool=300
maxconnections=40
maxuploadtarget=5000
datadir=/mnt/hdd/bitcoin
main.rpcport=8332
rpcallowip=127.0.0.1
main.rpcbind=127.0.0.1:8332
test.rpcport=18332
test.rpcbind=127.0.0.1:18332
upnp=0
blockfilterindex=0
main.wallet=wallet.dat
deprecatedrpc=addresses
onlynet=onion
proxy=127.0.0.1:9050
main.bind=127.0.0.1
test.bind=127.0.0.1
dnsseed=0
dns=0
whitelist=download@127.0.0.1

openoms, 2021-11-12:
switch
`whitelist=download@127.0.0.1`
to
`whitelist=127.0.0.1`

JurGi7, 2021-11-12:
I really overlooked that, shame on me... see what happens. for now thank you

JurGi7, 2021-11-15:
looks good, thank you. donation is out.

openoms, 2021-11-16:
TODO:
test whitebind instead of whitelits

JurGi7, 2021-11-16:
bitcoind does not start after 10 minutes
[image]

openoms, 2021-11-17:
@JurGi7 this is probably unrelated to this issue. Can you look at you logs with `debug` and bitcoin logs in SYSTEM an open a new issue if it is not related to nbxplorer.

JurGi7, 2021-11-17:
already suspected thx

openoms, 2022-04-12:
closing this after the v1.7.2 release

closed 2022-04-12 by openoms

robertvo, 2024-09-07:
[image]
[image]

After internet outage for several hours, now NBXplorer is stuck :(
Seems to be still a problem

rootzoll, 2024-09-10:
@robertvo this seems NBXplorer releated - not raspiblitz .. best to report that over here: https://github.com/dgarage/NBXplorer