raspiblitz/raspiblitz #2699: [Security] Linux namespaces

openoms, 2021-11-05:
Would like to discuss here which services would benefit the most from using namespaces.
Getting started:
https://linuxhint.com/use-linux-network-namespace/

Isolating services running hot wallets (lightning instances, joinmarket, bitcoind) from webGUIs could be the first aim.

rootzoll, 2021-11-11:
Adding to milestone 1.8.1 as a reminder to team discuss.

nyxnor, 2021-11-29:
- https://www.youtube.com/watch?v=iN2RnYaFn-0
- https://www.youtube.com/watch?v=_WgUwUf1d34

List namespaces:
```
ip netns
```
Lets create the netns called `red` and the other called `green`:
Add:
```
ip netns add red
ip netns add green
```
Delete:
```
ip netns del red
```
Execute command inside another namespace:
```
ip netns exec red ip link
## or enter the namespace shell
ip netns exec red sh
ip link
exit
``` 
Create a tunnel:
```
## add and connect eth0-r to veth-r
ip link add eth0-r type veth peer name veth-r
## places eth0-r to the red netns
ip link set eth0-r netns red
```
Set device up
```
ip link set veth-r up
ip netns exec red ip link set dev lo up
ip netns exec red ip link set dev eth0-r up
ip netns exec red ip address add 10.0.0.1/24 dev eth0-r
```
See the route:
```
ip netns exec red ip route
```