raspiblitz/raspiblitz #2702: CONTRIBUTING tips

nyxnor, 2021-11-06:
This is to add later on to CONTRIBUTING.md

- Use [shellcheck](https://github.com/koalaman/shellcheck) -  a static analysis tool for shell scripts -  it has a debiand packages with the same name and also can be added as a plugin to Vim, VSCode(ium) etc
- Lean from the [pure-bash-bible](https://github.com/dylanaraps/pure-bash-bible) - A collection of pure bash alternatives to external processes - avoid using heavy commands when there is already support using shell builtins
- Bash is not needed most of the cases, prefer [pure-sh-bible](https://github.com/dylanaraps/pure-sh-bible) for Posix compliancy
- Basic git commands to checkout upstream branch
- Make a script model, variables at the beggining, using case statement instead of if-then-else

openoms, 2021-11-07:
* shellcheck can be integrated to IDE-s like VSCode: https://marketplace.visualstudio.com/items?itemName=timonwong.shellcheck
* in most cases adhering to the coding style of the project is important (unless the style changes are also improvements)
* testing is crucial - here on a deployed RaspiBlitz on an RPi4

nyxnor, 2021-11-08:
* enable `trim trailing whitespace` on your editor, this helps to review only important code changes, not deleting whitespace.
* git help:
```sh
## fork the repo on github by going to the link:
https://github.com/rootzoll/raspiblitz/fork
## clone your forked repo
git clone https://github.com/<YOUR_USERNAME>/raspiblitz.git
## add upstream project
git remove add upstream https://github.com/rootzoll/raspiblitz.git
## create your new branch based on upstream/dev
git checkout -b <NEW_BRANCH> upstream/dev
## edit files
## add files to the commit or remove them from the git
git add <FILE_EDITED>
git rm <FILE_DETELED>
## commit them, short title and detailed description
git commit -m "Title with short description" -m "Detailed description of the changes"
## push to your own repo
git push -u origin <NEW_BRANCH>
##  open pull request on github and don't forget to compare against rootzoll/dev.
```

On other note, Openoms, I recommend VSCodium, which is VSCode without MS branding/telemetry/licensing, it has a little less features that does not affect shellscripts and it can install VSCode extensions, including shellcheck.

openoms, 2021-11-11:
* aim for PR-s to be atomic - solving one issue (or closely related) at a time. Example: when fixing a script don't include typos fixea for other scipts
*use `git config core.fileMode false` to ignore local permission settings

openoms, 2022-05-13:
Finding the Bisq guide clear and concise: https://github.com/bisq-network/bisq/blob/master/CONTRIBUTING.md could look into adapting it to the Raspiblitz.

The .editorconfig would especially be useful. Looking to use something like: https://github.com/Reckless-Satoshi/robosats/blob/main/.editorconfig