raspiblitz/raspiblitz #2703: Settings in RTL/ThunderHub not saved on reboot

Surenic, 2021-11-08:
Hey there,

I recently set up my own node with umbrel and migrated to raspiblitz soon after.

Is it normal that on reboot, settings on RTL like color theme, Fiat Display etc. is reset to default?

Also I can not open the info link on raspiblitz‘s default „web interface“. I am worried that these issues have something to do with migratimg from umbrel

nyxnor, 2021-11-08:
> Is it normal that on reboot, settings on RTL like color theme, Fiat Display etc. is reset to default?

Other person will respond to you about this

> Also I can not open the info link on raspiblitz‘s default „web interface“. I am worried that these issues have something to do with migratimg from umbrel

That is ok, it is a future feature that is coming soon

rootzoll, 2021-11-11:
> Is it normal that on reboot, settings on RTL like color theme, Fiat Display etc. is reset to default?

The migration from umbrel to RaspiBlitz just ports the bitcoin blockchain & lnd wallet/channel data ... to port all other additional apps data is too complex of a task so far.

> Also I can not open the info link on raspiblitz‘s default „web interface“. I am worried that these issues have something to do with migratimg from umbrel

A default web-interface fpr RaspiBlitz is planned with version 1.8

Surenic, 2021-11-11:
Thank you very much for your answer! But I guess you misunderstood, I'll try in german 😉

Angenommen ich stelle bei RTL jetzt die Theme in blau. Aus Gründen muss ich den Raspiblitz nun neustarten. Wenn ich RTL wieder öffne, ist alles auf default. Das ist jedes Mal so. Völlig unabhängig von meiner Migration von Umbrel. Ist das normal oder stimmt hier was nicht?

rootzoll, 2021-11-11:
Ah OK now I get you ;) Maybe thats a question to @saubyk .. how does RTL persist the change of the theme (for example to blue) and what are we missing on the RaspiBlitz that this perefrence setting gets losts after a reboot?

openoms, 2021-11-11:
The issue is caused by the
`config.scripts/bonus.rtl.sh prestart` script. It resets the setting at every RTL restart.

For now you can manually remove (or comment) this line from the systemd service:
`ExecStartPre=-/home/admin/config.scripts/bonus.rtl.sh prestart lnd main`

with:
`sudo nano  /etc/systemd/system/RTLservice`
then run:
`sudo systemctl daemon-reload`
`sudo systemctl restart RTL`

The settings will be preserved after that.

rootzoll, 2021-11-25:
Ah OK .. need to improve the prestart script for next release - my fault.

rootzoll, 2021-12-07:
I checked for thunderhub so far ... there is no prestart that overwrites the config on restart/reboot. The settings seems to be stored in the local storage of the web browser. So I will just concentrate on RTL on this issue.

rootzoll, 2021-12-08:
First fix comming with v1.7.2 is keeping configs reboot now BUT basically the settings file is stored on the sd card - so on every update or recovery with a fresh sd card those settings get still lost. Also changeing this with v1.7.2 that will store RTL config in `/mnt/hdd/app-data/rtl`.

rootzoll, 2021-12-08:
OK PR merged into `dev` - ready for final testing - especially that it will keep the config on updating with a new sd card.
If tests work out - delete branch `2703keepchanges`.

rootzoll, 2021-12-09:
`2703keepchanges` deleted - still keep this issue open for final testing.

commit f3773f2e (2021-12-08): fix #2703 keeping to RTL config changes thru reboot & sd card update (#2774)

* #2703 dont delete RTL config
* #2703 move RTL config to HDD
* fix config exists detection
closed 2022-02-17 by rootzoll