raspiblitz/raspiblitz #2704: After reflashing 1.7.1 - 90% of my channels disconnected and every rebalance shows error insufficient_balance

agd-media, 2021-11-08:
WIth version 1.7.0 everything was fine. Now after the update when I start the Raspiblitz most of the channels  are online and active. After a few hours 90% of my channels are offline. I have to restart the hardware (which takes at least double as longer than with version 1.7.0) to bring back the channels into an active state. Bos reconnect doesn't work here.
Rebalances are not successful. Always ends with error: "Insufficient_balance", even though I have enough balance in the channels.

rootzoll, 2021-11-11:
@nyxnor @openoms was there any problem with a tor update in v1.7.1? Any similar reports on the support chats?
Something to be addressed for the v1.7.2 released here?

nyxnor, 2021-11-18:
no similar reports to investigate

rootzoll, 2021-11-25:
OK .. no similar reports since two weeks. Closing issue - but anybody that will run into the same please report still so that we can reopen.

closed 2021-11-25 by rootzoll