raspiblitz/raspiblitz #2705: Quick question....

DrBlackross, 2021-11-08:
if i can get my vagrant up to 1.7.1 and stable, i should be able to export the backup tar.gz's from my RPI4 setup to the new vagrant setup? 
correct?

openoms, 2021-11-08:
yes, you should be able to, but not tested of course.

DrBlackross, 2021-11-10:
cool, close to making updating to 1.7.1 stable from the 1.60 image...

https://app.vagrantup.com/rootzoll/boxes/raspiblitz
(the hardest part)
is rebuilding the menu options from the 1.6 to carry through to the 1.7.1

DrBlackross, 2021-11-11:
ALSO, a suggestion on the LND-backup*-tar.gz can/is there a way to put the LND version in the backup filename? that way it makes it easier to roll the system forward to a certain point on their database db, instead of a hit or miss? lol

(i keep forgetting which lnd version i setup the LTC node on)

rootzoll, 2021-11-25:
It can make sense to put a version info file into the gz-file ... so on import it can check if data is from a "higher" version.

rootzoll, 2024-09-12:
The vagrant part is still experimental - but adding the version to the backups still makes sense. So closing this issue but I opened n new issue to work in the backup version: #4733

closed 2024-09-12 by rootzoll