raspiblitz/raspiblitz #2709: Use `scurl` instead of `wget` and plain `curl`.

nyxnor, 2021-11-09:
https://www.whonix.org/wiki/Secure_Downloads
The link refers to problems with wget and how to improve curl. This does not work with self signed certificates or any http website, including onion services.
```sh
curl --tlsv1.3 --proto =https --location  --remote-name
```

Ocurrences https://github.com/rootzoll/raspiblitz/search?q=wget

rootzoll, 2021-11-11:
Sounds worth a deeper look ... but kicking it up to v1.8.1 for now because of prio.

nyxnor, 2021-11-18:
Funny that keybase.io uses TLS 1.2, which breaks the above. (was trying to import rootz keys)
`curl --tlsv1.2 --proto =https https://keybase.io/rootzoll/pgp_keys.asc` works
from `curl --help all`
```
 -1, --tlsv1         Use TLSv1.0 or greater
     --tlsv1.0       Use TLSv1.0 or greater
     --tlsv1.1       Use TLSv1.1 or greater
     --tlsv1.2       Use TLSv1.2 or greater
     --tlsv1.3       Use TLSv1.3 or greater
```

`--tlsv1.2` will be the default, if encountering with any trouble, even blitz does not support 1.1 https://github.com/rootzoll/raspiblitz/blob/a0fd288e553678d0816d682e968a2eea0d165a9b/home.admin/assets/nginx/snippets/ssl-params.conf#L7
but maybe use 1.1?