raspiblitz/raspiblitz #2710: How to exit the status screen to use the shell?

got3nks, 2021-11-09:
I misconfigured the firewall on my Raspiblitz and now can't log in through SSH.

I plugged in the HDMI port to a monitor but whenever I plug in a USB keyboard the input goes to the LCD screen (which is showing the status of the node) so any command is ignored (I tried CTRL+C without luck).

How do I exit from the status screen to be able to execute commands?

nyxnor, 2021-11-09:
> I misconfigured the firewall on my Raspiblitz and now can't log in through SSH.

Flash the sd card, it is not possible to advise if we don't know what you did with it.

got3nks, 2021-11-09:
I found a workaround: after shutting down the Raspberry I unplugged the SSD drive and then powered on again. Since Raspiblitz fails to boot and start its processes, I was able to login as root and use the shell through the LCD screen output. 

I think a sigint handler should be added in the process used to show the status on the LCD monitor to prevent this issue.

rootzoll, 2021-11-11:
TODO: When [ssh.reset](https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#how-can-i-repair-my-ssh-login) is used also make sure to open firewall port.

got3nks, 2021-11-11:
> TODO: When [ssh.reset](https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#how-can-i-repair-my-ssh-login) is used also make sure to open firewall port.

Although that's a good idea, it would still require to remove the SD card from the Raspberry. Wouldn't be possible to use something like this in the 00infoLCD.sh script?

```
quit() {
  echo "SIGINT or SIGTERM received, exiting..."
  kill -9 $$
}

trap quit INT
trap quit TERM
```

rootzoll, 2021-12-08:
@got3nks tried to trap those signal in the `00infoLCD.sh` but was still not able to exit that loop with connected USB keyboard. 

If you searching for a way "to shutdown" the device there is the idea to detect if the LAN cable is getting removed and to use that as a signal to shutdown. Would that be a future solution?

TODO in this ticket:
- [ ] check ssh port open when ssh.reset file is detected
- [ ] check if the trap mechanism help to break loop during setup/sync (already in branch)

got3nks, 2021-12-14:
The point is there is no way to interact with the OS when an LCD is connected. If we can break from the loop and be able to use a keyboard would be much better than having to remove the SD card from the Raspberry to fix the filesystem.

rootzoll, 2022-02-17:
Exiting the status screen with CTRL+c is now orking again. Closing issue for release.

closed 2022-02-17 by rootzoll