raspiblitz/raspiblitz #2713: Sync from scratch stared at and jammed at: Lightning Progress : 21.45 % 0 peers

philiprhoades, 2021-11-10:
People,

Today I just got started with Lightning in a Box - the setup was painless and I only chose LN instead of LN + BN - but the syncing has been jammed on (from the beginning) at:

Lightning Progress  :  21.45 % 0 peers 

for about 6 hours now - why did it start at that value instead of 0.0% and in any case I should be seeing some increase in % ?

The hdd is only growing by about 430b every 10mins - that can't be right can it?

Thanks,
Phil.

rootzoll, 2021-11-25:
The lightning progress is based in how much blockchain data is already synced .. if in the begining just a few blocks are synced the percentage may quickly jump to something like 21.45% ... this might be confusing. Maybe it just makes sense to not display lightninig progress until blockchain is synced and just show number of peers until then.

Did the blockchain sync pulled thru afterall?

philiprhoades, 2021-11-27:
> Did the blockchain sync pulled thru afterall?

No, in the end I decided to update from v1.7.0 and flash v1.7.1 and atter that everything started working as expected - it took about 3 days and 10 hours to fully sync, start finding peers etc.  I have now created 6 channels but there have been no transactions (~10 days) so far . .

cstenglein, 2023-06-07:
@philiprhoades  is this still an issue with 1.9.0?

Closing this for now, please re-open if it still persists.