raspiblitz/raspiblitz #2714: Block Explorer missing after 1.7.1 Update (migration)

velgard1, 2021-11-12:
If it really is it is only a minor issue, but I wanted to report it anyway.
After an issue with BTRFS I waited for the new release and used a migration file to setup 1.7.1 without BTRFS.
After syncing I checked all services I had used (e.g. Mempool Explorer) and wondered that I was not able to connect to the Block Explorer.
I then realized that the service was not mentioned within the menu and therefore it was not installed - installation worked and I could use it afterwards as before.

rootzoll, 2021-11-25:
Thanks for the report ... will check if recovering is still functioning for blockexplorer on next v1.7.2 release.

rootzoll, 2022-01-17:
OK did a test with the v1.7.2rc1d and BlockExplorer gets installed correctly on recover/update.

velgard1, 2022-01-20:
Thanks, I hope I don't have to migrate next time, but it's good to know that it should work.