raspiblitz/raspiblitz #2715: Lightning Terminal web page no connection

LupusPluvia, 2021-11-12:
Since a transaction of 100k satochi, I have not come back to the site.

New installation was done nothing happened

LupusPluvia, 2021-11-12:
# menu loop received exit code 1 --> exit to terminal

               -==@@@====@===--       --===@====@@@==-         
            -@@=====-----=-===@@=====@@=====-----=====@@-       -==@-
            -@@------==---------@@@@@=--------==------@@-  --=@@@@=
             @@=------======-----@@@-----======------=@@=@@===@@=
             =@@=---------=======@@@=======-----===@@@==-  =@@-
          -=@@==@@=----------=@@@@@@@@@=----==@@@==--   -=@@-  
        -@@@=----=@@===--====@@@@@@@@@@@@@@@@=--      -=@@@@=  
       =@@=--------@@@@@@@@@@@@@@@@@@@@@=--         -@@=---=@@-
     -@@=-------=@@@=====@@@@@@@@@@=--            =@@=-------@@@
    =@@=-------=@@====@@@@@@@@==-              -=@@@@=--------=@@-
   =@@---------@@==@@@@@@==-                 -=@@@=@@@---------=@@
  -@@=--------=@@@@@@=-                    -@@@@@@@@@@=---------=@@
  @@=-------@@@@@@@@@=-                  =@@@===@@@@=@@@@--------@@-
 -@@=------@@====@@@=@@@=-            -=@@@======@@====@@@-------=@@
 -@@------@@@====@@@====@@@=-         =@@@======@@@=====@@=------=@@
 -@@------=@@====@@@@@=====@@@=-        -=@@@=@@@@@@===@@@=------=@@
 -@@-------@@@=@@@@@@@@@@@@@@@=-           -=@@@@@@@@=@@@=-------=@@
  @@=-------=@@@@@@@@@@@@@@@=                 -=@@@=@@@@=--------@@=
  -@@--------=@@======@@@@-                    -=@@@@@@=--------=@@
   =@@--------@@@===@@@=-                 --=@@@@@@=@@@--------=@@-
    =@@--------@@@@@@=-               -==@@@@@=====@@@--------=@@-
     =@@=-------@@@=             -==@@@@@@@=====@@@@=--------=@@-
      -@@=----=@@-          -==@@@@@@@@@@@@@@@@@==---------=@@=
        =@@==@@-       -==@@@@@=========@@@@@=-----------=@@@- 
         -@@=-    --=@@@==-=@@@@@@@@@@@@@=-------------=@@=-   
       -@@=  --=@@@==----------=======-------------==@@@=      
     -@@=-==@==-=@@@===------------------------==@@@@=         
   =@@@@==-        -==@@@@@======----======@@@@@=--            
 =@@=--                 --===@@@@@@@@@@@===--                  

***********************************
* RaspiBlitz Commandline
* Here be dragons .. have fun :)
***********************************
Bitcoin command line options: bitcoin-cli help
LND command line options: lncli -h
Blitz command line options: blitzhelp
Back to menus use command: raspiblitz

admin@raspberrypi:~ $ sudo system status litd
sudo: system: command not found
admin@raspberrypi:~ $ sudo journalctl -fu litd
-- Logs begin at Fri 2021-11-12 20:20:08 GMT. --
Nov 12 21:28:54 raspberrypi litd[32196]: 2021-11-12 21:28:54.823 [DBG] LNDC: Wait for invoices to finish
Nov 12 21:28:54 raspberrypi litd[32196]: 2021-11-12 21:28:54.823 [DBG] LNDC: Lnd services finished
Nov 12 21:28:54 raspberrypi litd[32196]: 2021-11-12 21:28:54.823 [DBG] GRPC: Channel Connectivity change to SHUTDOWN
Nov 12 21:28:54 raspberrypi litd[32196]: 2021-11-12 21:28:54.823 [DBG] GRPC: Subchannel Connectivity change to SHUTDOWN
Nov 12 21:28:54 raspberrypi litd[32196]: 2021-11-12 21:28:54.824 [INF] SGNL: Received shutdown request.
Nov 12 21:28:54 raspberrypi litd[32196]: 2021-11-12 21:28:54.824 [INF] SGNL: Shutting down...
Nov 12 21:28:54 raspberrypi litd[32196]: 2021-11-12 21:28:54.824 [INF] SGNL: Gracefully shutting down.
Nov 12 21:28:54 raspberrypi litd[32196]: unable to start account manager: unable to resume account 03b2ee8846f55f81393bdf75defe3530da038734ee97886360989c76e722f0c9d9: rpc error: code = Unknown desc = current minimum account expiry allowed is height 709572
Nov 12 21:28:54 raspberrypi systemd[1]: litd.service: Main process exited, code=exited, status=1/FAILURE
Nov 12 21:28:54 raspberrypi systemd[1]: litd.service: Failed with result 'exit-code'.
Nov 12 21:29:55 raspberrypi systemd[1]: litd.service: Service RestartSec=1min expired, scheduling restart.
Nov 12 21:29:55 raspberrypi systemd[1]: litd.service: Scheduled restart job, restart counter is at 56.
Nov 12 21:29:55 raspberrypi systemd[1]: Stopped litd Service.
Nov 12 21:29:55 raspberrypi systemd[1]: Started litd Service.
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.165 [INF] LITD: Dialing lnd gRPC server at 127.0.0.1:10009
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.167 [DBG] GRPC: parsed scheme: ""
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.167 [DBG] GRPC: scheme "" not registered, fallback to default scheme
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.167 [DBG] GRPC: ccResolverWrapper: sending update to cc: {[{127.0.0.1:10009  <nil> 0 <nil>}] <nil> <nil>}
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.168 [DBG] GRPC: ClientConn switching balancer to "pick_first"
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.168 [DBG] GRPC: Channel switches to new LB policy "pick_first"
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.168 [DBG] GRPC: Subchannel Connectivity change to CONNECTING
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.173 [DBG] GRPC: Subchannel picks a new address "127.0.0.1:10009" to connect
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.168 [DBG] GRPC: pickfirstBalancer: HandleSubConnStateChange: 0x4000532130, {CONNECTING <nil>}
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.174 [DBG] GRPC: Channel Connectivity change to CONNECTING
Nov 12 21:29:55 raspberrypi litd[4997]: 2021-11-12 21:29:55.220 [DBG] GRPC: Subchannel Connectivity change to READY
[... 112 lines cut ...]
Nov 12 21:30:01 raspberrypi litd[4997]: 2021-11-12 21:30:01.099 [DBG] LOOP: Wait for goroutines to finish
[... cut ...]
Nov 12 21:30:01 raspberrypi litd[4997]: 2021-11-12 21:30:01.103 [DBG] GRPC: transport: loopyWriter.run returning. connection error: desc = "transport is closing"
Nov 12 21:30:01 raspberrypi litd[4997]: unable to start account manager: unable to resume account 03b2ee8846f55f81393bdf75defe3530da038734ee97886360989c76e722f0c9d9: rpc error: code = Unknown desc = current minimum account expiry allowed is height 709572
Nov 12 21:30:01 raspberrypi systemd[1]: litd.service: Main process exited, code=exited, status=1/FAILURE
Nov 12 21:30:01 raspberrypi systemd[1]: litd.service: Failed with result 'exit-code'.

LupusPluvia, 2021-11-13:
After a few hours it works again, but now it is displayed on the page
[image]

openoms, 2021-11-15:
did you try a full restart yet?

LupusPluvia, 2021-11-15:
they mean a new SD card to surfaces.  No, I haven't yet, but I can try it out tonight.

LupusPluvia, 2021-11-16:
SD card reassembled.  Is against the same thing the side cannot be reached

LupusPluvia, 2021-11-16:
Now after a few hours I come back to the page but the error message is the same

LupusPluvia, 2021-12-14:
I also described the problem with leitning terminal.  She then knows that the installation is not correct.  Cannot see pool.log data

LupusPluvia, 2021-12-14:
https://github.com/lightninglabs/lightning-terminal/issues/293#issue-1077467136

openoms, 2021-12-14:
if you edit the script you can install the latest version of LiT:

use:
```
nano config.scripts/bonus.lit.sh
```

change the version to `v0.6.1-alpha`
save with ctrl+o, enter, ctrl+x

then reinstall:
```
config.scripts/bonus.lit.sh off
config.scripts/bonus.lit.sh on
```

openoms, 2021-12-14:
No it should available as soon as the install is finished.

You an see if there are any errors with:
```
sudo journalctl -fu litd
```

openoms, 2021-12-14:
This seems to be problem, but not sure what to make of it:
```
unable to start account manager: unable to resume account 03b2ee8846f55f81393bdf75defe3530da038734ee97886360989c76e722f0c9d9: rpc error: code = Unknown desc = current minimum account expiry allowed is height 714313
```

LupusPluvia, 2021-12-15:
If I had not made a transaction, I would not have had the problem since then.  Because I wanted to try it out

openoms, 2021-12-15:
Can you please provide your debug logs?
Use this command and share the link here:
```
debug | torsocks nc termbin.com 9999
```

LupusPluvia, 2021-12-15:
That’s what came out of it
admin@raspberrypi:~ $ debug | torsocks nc termbin.com 9999
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
Unit blitzapi.service could not be found.
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
https://termbin.com/tdzn
admin@raspberrypi:~ $

LupusPluvia, 2021-12-16:
Es geht alles wider vor ab würde den Fall schließen Danke

closed 2021-12-16 by LupusPluvia