raspiblitz/raspiblitz #2717: LCD won't show status screen (stuck at 100% on blue screen)

dantrauner, 2021-11-15:
Has anyone else encountered this before? I updated to 1.7.1 recently and seem to be synced with the blockchain, but the LCD won't switch away from the "in-progress" screen to the usual black status screen.

[image]

openoms, 2021-11-15:
login with ssh as it shows and restart from.the menu

dantrauner, 2021-11-15:
Definitely tried that first — doesn’t seem to work. What script bootstraps the LCD? I figured maybe there’s a problem with that and running it manually/debugging there would help.

dantrauner, 2021-11-16:
I just reflashed the SSD in the end and the issue is obviously now gone, but I guess this is some kind of edge case bug. I'll try to debug more thoroughly if I ever see it again.

closed 2021-11-16 by dantrauner