raspiblitz/raspiblitz #2718: screen updating in a loop

Darien46, 2021-11-15:
my node is synced, the screen has been stuck for 2 days. It says screen updating in a loop. not sure how to properly address this issue.

openoms, 2021-11-15:
is it your desktop monitor or LCD?
press CTRL+C or try a restart through ssh

Darien46, 2021-11-15:
It is happening on the lcd screen. I have restarted it a couple of times
now. It always goes back to the same spot during the reboot and says screen
updating in loop. On my monitor when I hash in, everything is synced and
running perfectly fine

Darien46, 2021-11-16:
> This is the lcd screen.
> Second photo is monitor after control C. Not sure how to proceed

Monitor screen after control C. How do I proceed. When I put in admin
password, it goes back to screen you see, which says updating in loop

Darien46, 2021-11-16:
the control c did not allow me to restart through hash. it asked for adin password and continued on same screen indication looping

rootzoll, 2021-11-25:
If this issue still persist please try to make a photo of the LCD screen so that I can get an idea at which point of the display process this is happening.

Darien46, 2021-12-01:
Pi lcd screen. Stuck 2 weeks

Darien46, 2021-12-01:
My screen when I hash in from my phone that tells me it's updating in a
loop.

Darien46, 2021-12-06:
Hi,
Will you be able to assist me with this?

rootzoll, 2021-12-09:
Again .. please send a photo of your screen. Or if you cannot send a photo please quote the exact text you see on your screen. Without any more detailed info this is hard to debug from remote.

Darien46, 2021-12-09:
Sorry,
I am attaching photos to this email. I'll try to go into the github and add
the photos. I apologize, I'm not a tech wizard. I attached the photos to
this email thread. I appreciate your time.

closed 2022-07-04 by openoms