raspiblitz/raspiblitz #2722: upgrading problem to 1.7.1

xpsx, 2021-11-16:
[image]

 │ Upgrade/Recover/Provision            │  
                                                │ ---> Setup System .                  │  
                                                │                                      │  
                                                │ Run 'tail -f ./raspiblitz.log' in    │  
                                                │ new terminal to follow install logs. 

admin@raspberrypi:~ $ tail -f ./raspiblitz.log
###################################
### DISPLAY SETTINGS ###
raspiblitz.info() raspiblitz.conf(lcd)
WARN values in raspiblitz info and/or conf file seem broken
### BASIC SYSTEM SETTINGS ###
# Optimizing log files: rotate daily, keep 2 weeks & compress old days 
Setting lightning alias: Mia
Copy HDD content for user admin
### PREPARE LND
*** Make backup of LND directory

DrBlackross, 2021-11-16:
stuck too

https://github.com/rootzoll/raspiblitz/issues/2581#issuecomment-970688008

xpsx, 2021-11-20:
I have waited for all night long, I found the upgrading was successful on the next morning.
PS: before upgrading, I couldn't backup LND, was stuck after the file channels.db
so I skipped the backup

closed 2021-11-20 by xpsx

rootzoll, 2021-11-25:
@xpsx just to verify ... there the `*** Make backup of LND directory` is the last line in the `raspiblitz.log`? 

If yes can you give me the result of the following command:
```
sudo diff -arq /mnt/hdd/lnd /mnt/hdd/backup_lnd | grep -c "differ"
```

xpsx, 2021-12-20:
> @xpsx just to verify ... there the `*** Make backup of LND directory` is the last line in the `raspiblitz.log`?
> 
> If yes can you give me the result of the following command:
> 
> ```
> sudo diff -arq /mnt/hdd/lnd /mnt/hdd/backup_lnd | grep -c "differ"
> ```

excuse me that I'm responding only now, I don't know if it is still an issue, anyway the output is: `6`