raspiblitz/raspiblitz #2727: v1.7.1 Restoring from channel backup (_provision.setup.sh fail)

Csi18nAlistairMann, 2021-11-18:
Experimenting with SCB recovery to a node with existing blockchain data. This process fails trying to import the SCB. This is different to issue #2639 which failed on Password C.

- Two nodes - Old still working, New has an unused installation. This process to work through migrating from Old to New using the seed & scb method
- SD card is flashed with raspiblitz-v1.7.1-2021-10-12.img
- Choose FRESHSETUP (via Other options)
- USE BLOCKCHAIN to question "We found BITCOIN blockchain data..."
- DELETE DATA to question "OK we will keep the blockchain data..."
- Name of New node to match Old
- LND to question "Choose your Lightning Client..."
- OLD to question "LND Data & Wallet"
- SEED+SCB to question "Data you have to recover from?"
- paste in seed words
- NO EXTRA PASSWORD to question "Are your seed words protected..."
- scp of latest available channel backup file per instructions
- ENTER 
- ENTER to "Ok - File found ..."
- Passwords A, B, C on New set to match those on Old
- <provisioning intensifies>
Fails with the following output:

```
###########################################################
# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Tue 12 Oct 00:12:57 BST 2021
***********************************************
# *** blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
No HDMI switch found. 
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No SSHRESET switch found. 
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 604
drwxr-xr-x   2 root root   4096 Oct 12 00:12 .
drwxr-xr-x 118 root root   4096 Oct 12 00:12 ..
-rw-r--r--   1 root root 565189 Jan 31  2020 moduli
-rw-r--r--   1 root root   1580 Jan 31  2020 ssh_config
-rw-r--r--   1 root root   3250 Jan 31  2020 sshd_config
-rw-------   1 root root   1385 Oct 12 00:12 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Oct 12 00:12 ssh_host_dsa_key.pub
-rw-------   1 root root    505 Oct 12 00:12 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Oct 12 00:12 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Oct 12 00:12 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Oct 12 00:12 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Oct 12 00:12 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Oct 12 00:12 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Mar  5  2021 ssh_import_id
wifiIsSet=0
hddBackupLocationAvailable=0
hddRestoreConfigAvailable=0
memRestoreConfigAvailable=0
wifiRestore=0
wifiBackup=0
No UASP FIX needed (2nd-try).
isMounted: 0
Checking for UASP deactivation ...
HDD already part of system: 0
Temp mounting (1) data drive (sda)
Temp mounting (1) result: 1
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
# df 
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root       30445284   4377540  24783308  16% /
devtmpfs         1933976         0   1933976   0% /dev
tmpfs            1968600         0   1968600   0% /dev/shm
tmpfs            1968600     16824   1951776   1% /run
tmpfs               5120         0      5120   0% /run/lock
tmpfs            1968600         0   1968600   0% /sys/fs/cgroup
tmpfs              32768         4     32764   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095     30357    227738  12% /boot
tmpfs             393720         0    393720   0% /run/user/1000
/dev/sda1      960379920 421334228 490191268  47% /mnt/hdd
# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda                
└─sda1      ext4   BLOCKCHAIN
mmcblk0            
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=1
hddRaspiVersion='1.7.1'
isSSD=0
datadisk='sda'
datapartition='sda1'
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddBytes=1000203091968
hddGigaBytes=931
hddUsedInfo='402G (47%)'
hddDataFreeKB=490191268
hddAdapterUSB='174c:1153'
hddAdapterUSAP=0

# RAID
isRaid=0
raidCandidates=0

# SWAP
isSwapExternal=0

COPY und Activating old SSH host keys
# *** blitz.ssh.sh restore
# backup dir: /mnt/hdd/ssh
# OK - ssh keys restore done
Displaying Info Message: Please Login for Recovery
## WAIT LOOP: USER SETUP/UPDATE/MIGRATION
# PROVISION PROCESS with setupPhase(setup)
Temp mounting (2) data drive (sda)
Temp mounting (2) result: 1
Refreshing links between directories/drives ..
# Sourcing /mnt/hdd/raspiblitz.conf 
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.7.1'
lcdrotate=1
lightning=lnd
network=bitcoin
chain=main
hostname='goosnargh'
runBehindTor=on
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup 
# RASPIBLITZ SETUP STATE
network=bitcoin
hostname=goosnargh
lightning=lnd
network=bitcoin
chain=main
setPasswordA=1
setPasswordB=1
setPasswordC=1
seedWords='ABOUT GLAD SOURCE METHOD MUSIC VOYAGE WISH OUTPUT COUSIN CLEAN EXAMPLE SLIM GOLD ARCH RESIST BOTTOM TATTOO WEALTH HARD FATIGUE PITCH PLACE SOLVE THUNDER'
seedPassword=''
staticchannelbackup='/home/bitcoin/temp/upload/channel-20211114-192036.backup'
# df 
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root       30445284   4283376  24877472  15% /
devtmpfs         1933976         0   1933976   0% /dev
tmpfs            1968600         0   1968600   0% /dev/shm
tmpfs            1968600     25024   1943576   2% /run
tmpfs               5120         0      5120   0% /run/lock
tmpfs            1968600         0   1968600   0% /sys/fs/cgroup
tmpfs              32768        20     32748   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095     30357    227738  12% /boot
tmpfs             393720         0    393720   0% /run/user/1000
tmpfs             393720         0    393720   0% /run/user/1001
/dev/sda1      960379920 421331572 490193924  47% /mnt/hdd
# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda                
└─sda1      ext4   BLOCKCHAIN
mmcblk0            
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
# Update /home/admin/raspiblitz.info 
state=provision
message='Starting Provision'
baseimage=raspios_arm64
cpu=aarch64
board=rp4
ramMB=3844
ramGB=4
network=bitcoin
chain=main
localip='192.168.1.112'
online='1'
dnsworking=1
fsexpanded=1
displayClass=lcd
displayType=
setupStep=0
setupPhase='setup'
fundRecovery=0
hostname=
hddCandidate='sda'
hddGigaBytes=931
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddGotMigrationData=
# migrationFile()
# setting PASSWORD A

OK - password A changed for user pi, root, admin & bitcoin
...
Calling _provision.setup.sh for basic setup tasks ..
EXIT _provision.setup.sh BECAUSE OF ERROR STATE (10)
This can also happen if _provision.setup.sh has syntax errors
# /home/admin/raspiblitz.provision-setup.log

###################################
# _provision.setup.sh
###################################
*** Prepare bitcoin ***
mkdir: cannot create directory ‘/mnt/hdd/bitcoin’: File exists
mkdir: cannot create directory ‘/mnt/hdd/bitcoin/blocks’: File exists
mkdir: cannot create directory ‘/mnt/hdd/bitcoin/chainstate’: File exists
File Exists: /mnt/hdd/bitcoin/bitcoin.conf --> 1
# setting PASSWORD B
# OK -> RPC Password B changed
# Reboot is needed
...
*** Start bitcoin (SETUP) ***
- This can take a while ..
# (0/50) checking if bitcoind is running ... 
# (1/50) checking if bitcoind is running ... 
# (2/50) checking if bitcoind is running ... 
# (3/50) checking if bitcoind is running ... 
# (4/50) checking if bitcoind is running ... 
OK bitcoin startup successful 
Prepare Lightning (lnd)
############## Setup LND
Creating new LND config ...
*** Init LND Service & Start ***
Starting LND Service ...
Preparing static channel backup file /home/bitcoin/temp/upload/channel-20211114-192036.backup ...
FAIL: setup: lnd import SCB failed
unknown parameter
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(_provision.setup.sh fail)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
admin@raspberrypi:~ $ 
```
On IRC advice, repeated the reinstall using raspiblitz-v1.7.0-2021-04-25.img. This took some minutes to complete bootstrap before logging in could complete. Repeatedly, and despite reboots, it got stuck on
```
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
```
Second time around, /mnt/hdd was manually cleared using
```
cd /mnt/hdd
sudo rm -rf ssh/ app-storage/ backup_lnd/ tor-lnd/ tor/ app-data/ raspiblitz.conf lnd/ temp/
```
This time the reinstall works including accepting SEED+SCB and asks for a further hour before being sure funds have been recovered. After being left overnight it's clear installing to v1.7.0 then recovering SEED+SCB has nevertheless resulted in all channels being closed.

P1xelfehler, 2021-12-13:
In line 348 of lnd.backup.sh there is "exit 0" missing. I think that's the error here.

rootzoll, 2022-01-16:
@openoms this was what you were fixing recently or is that another scenario we should look in again?

openoms, 2022-01-16:
This should be fixed with the new REPAIR-LND menu and tested ok.

closed 2022-02-20 by rootzoll