raspiblitz/raspiblitz #2728: LND wont start after upgrading to v0.14.0

wchancao, 2021-11-19:
Hello, i upgraded my Raspiblitz v1.7.1 to lnd v0.14.0 and after the reboot, lnd wont start, what can i do? Reflashing sd card to 1.7.1 will help? These is the result of debug

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.1
chainnetwork: bitcoin / main
 07:44:40 up  9:03,  2 users,  load average: 0.61, 0.92, 1.06

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin-Daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-11-18 22:42:43 -04; 9h ago
  Process: 2457 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 2459 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
 Main PID: 2509 (bitcoind)
    Tasks: 18 (limit: 4532)
   CGroup: /system.slice/bitcoind.service
           └─2509 /usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Fri 2021-11-19 03:47:18 -04, end at Fri 2021-11-19 07:44:40 -04. --
-- No entries --

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-11-19T10:36:15Z Socks5() connect to is3rbhbfg7vlavq5wijlzwrbrns3psb7cbxk4cn3xyg7r5gruxwslcqd.onion:8333 failed: host unreachable
2021-11-19T10:37:18Z Socks5() connect to is3rbhbfg7vlavq5wijlzwrbrns3psb7cbxk4cn3xyg7r5gruxwslcqd.onion:8333 failed: host unreachable
2021-11-19T10:38:37Z UpdateTip: new best=00000000000000000008fddd0eb72e2e93f00e34d10826a7f92ef4d20ad50012 height=710413 version=0x27ffe004 log2_work=93.183152 tx=688110727 date='2021-11-19T10:38:15Z' progress=1.000000 cache=41.4MiB(308822txo)
2021-11-19T10:39:44Z New outbound peer connected: version: 70016, blocks=710413, peer=501 (outbound-full-relay)
2021-11-19T10:41:21Z New outbound peer connected: version: 70016, blocks=710413, peer=503 (outbound-full-relay)
2021-11-19T10:44:40Z New outbound peer connected: version: 70016, blocks=710413, peer=508 (block-relay-only)
2021-11-19T10:54:15Z New outbound peer connected: version: 70016, blocks=710413, peer=520 (outbound-full-relay)
2021-11-19T10:56:08Z New outbound peer connected: version: 70016, blocks=710413, peer=525 (block-relay-only)
2021-11-19T10:58:56Z UpdateTip: new best=00000000000000000003aadf3be749ef98ef21b60d5f148051f5ea7a1c78ad1b height=710414 version=0x37ffe000 log2_work=93.183164 tx=688111901 date='2021-11-19T10:58:42Z' progress=1.000000 cache=46.2MiB(347889txo)
2021-11-19T11:03:07Z New outbound peer connected: version: 70016, blocks=710414, peer=538 (block-relay-only)
2021-11-19T11:12:58Z New outbound peer connected: version: 70016, blocks=710414, peer=554 (block-relay-only)
2021-11-19T11:15:51Z New outbound peer connected: version: 70016, blocks=710414, peer=557 (outbound-full-relay)
2021-11-19T11:16:43Z Socks5() connect to wzjkjhn2mypnmc6u2h2cleactuf2l6ju4mfnawsag26qnd7qgvjq5zyd.onion:8333 failed: host unreachable
2021-11-19T11:18:31Z UpdateTip: new best=000000000000000000038caf3a32f4f2bd3c2681acf7e6fc12ff32e5d33e19e1 height=710415 version=0x20800000 log2_work=93.183177 tx=688113498 date='2021-11-19T11:18:27Z' progress=1.000000 cache=47.8MiB(360594txo)
2021-11-19T11:19:27Z Socks5() connect to kue5ldyzchppq6bfnidueuip7pbyyslx4fyt6gmdlrvawvklmaptm6id.onion:8333 failed: host unreachable
2021-11-19T11:19:48Z Socks5() connect to jlimg73wee7etjzmsc6ax4nbpvs3qxwjviqjrjteg7psgxjiszjer7id.onion:8333 failed: host unreachable
2021-11-19T11:22:29Z UpdateTip: new best=000000000000000000022b107bc552bba39c29bf59bbc5aaf0f2b8d373a138e4 height=710416 version=0x20c00000 log2_work=93.183189 tx=688113977 date='2021-11-19T11:21:55Z' progress=1.000000 cache=48.0MiB(362168txo)
2021-11-19T11:27:18Z Socks5() connect to koapok5rmqy25wwtjrr2zvhgava6kp72izu2qtkb6mma3k542hbgcead.onion:8333 failed: host unreachable
2021-11-19T11:30:23Z New outbound peer connected: version: 70016, blocks=710416, peer=575 (block-relay-only)
2021-11-19T11:43:59Z New outbound peer connected: version: 70016, blocks=710416, peer=588 (block-relay-only)

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-11-19 07:44:38 -04; 1s ago
  Process: 2910 ExecStartPre=/home/admin/config.scripts/lnd.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
 Main PID: 3006 (lnd)
    Tasks: 10 (limit: 4532)
   CGroup: /system.slice/lnd.service
           └─3006 /usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.conf

Nov 19 07:44:38 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Nov 19 07:44:38 raspberrypi systemd[1]: Started LND Lightning Daemon.
[... 227 lines cut ...]
tempCelsius='64.2'
uptime=32601
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainPeers=22
blockchainHeight=710416
initialSync=0
syncProgress=100.00 
startcountLightning=320
lndActive=1
lndRPCReady=1
walletOpened=0
walletReady=0
walletLocked=0
lndErrorShort='Unknown Error - see logs'
lndErrorFull='[lncli] rpc error: code = Unknown desc = waiting to start, RPC services not available'
# OK: log 'ERROR: [lncli] rpc error: code = Unknown desc = waiting to start, RPC services not available' written to /home/admin/systemd.lightning.log
lndRPCReady=1
clActive=0
CLwalletLocked=0
blitzTUIActive=0
blitzTUIRestarts=0
vagrant=0
scriptRuntime=1

redmac68, 2021-11-19:
Just in case it helps - a similar issue is raised for LND 0.14.0 here https://github.com/lightningnetwork/lnd/issues/6002

wchancao, 2021-11-19:
Yes, commenting out or deleting 'protocol.no-anchors=true' from lnd.conf solved it. Thank you!!

closed 2021-11-24 by openoms