raspiblitz/raspiblitz #2729: c-lightning Backup

SirSir21, 2021-11-19:
Hello everyone. I have a problem with my c-lightning wallet or my channels. When I load the backup file, the channels are displayed on RTL, as soon as the wallet is 100% synchronized, they disappear again. I've done it three times with the result described above. The backup was created under v10.2cr2. I'm currently using v0.10.2. Could that be the problem, why do channels go away when they're 100% synced? Ps. I use Raspiblitz v1. 7.1

openoms, 2021-11-24:
The channels were likely force closed as you restored the backup (possibly an outdated state).
Will need to rescan to restore the onchain balance:

https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.cl.md#rescan-the-chain-after-restoring-a-used-c-lightning-wallet

openoms, 2021-11-30:
Duplicate of https://github.com/ElementsProject/lightning/issues/4923 and also discussed on the c-lightning Discord: https://discord.com/channels/899980449231814676/899989729183940629/914655501076287550

closed 2021-11-30 by rootzoll