raspiblitz/raspiblitz #2730: Blockchain progress stuck at zero percent

jstrauss501, 2021-11-21:
it seems like the blockchain sync process stucks. I have reboot many times already. can someone please help me solve the issues.

=======
[image]
============================================
cat raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Sun 21 Nov 03:25:57 GMT 2021
***********************************************
# *** blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
No HDMI switch found. 
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No SSHRESET switch found. 
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 604
drwxr-xr-x   2 root root   4096 Oct 28 11:00 .
drwxr-xr-x 118 root root   4096 Nov 20 01:12 ..
-rw-r--r--   1 root root 565189 Jan 31  2020 moduli
-rw-r--r--   1 root root   1580 Jan 31  2020 ssh_config
-rw-r--r--   1 root root   3250 Jan 31  2020 sshd_config
-rw-------   1 root root   1385 Oct 28 11:00 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Oct 28 11:00 ssh_host_dsa_key.pub
-rw-------   1 root root    513 Oct 28 11:00 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Oct 28 11:00 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Oct 28 11:00 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Oct 28 11:00 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Oct 28 11:00 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Oct 28 11:00 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Mar  5  2021 ssh_import_id
wifiIsSet=0
hddBackupLocationAvailable=1
hddRestoreConfigAvailable=0
memRestoreConfigAvailable=0
wifiRestore=0
wifiBackup=0
No UASP FIX needed (2nd-try).
isMounted: 1
Skipping UASP deactivation ... cmdlineExists(1) hddAdapterUSB() hddAdapterUSAP(0)
HDD already part of system: 1
load configfile data
running LND users credentials update
###### SYNCING MACAROONS, RPC Password AND TLS Certificate ######
# make sure LND app-data directories exist

# copy macaroons to central app-data directory and ensure unix ownerships and permissions
# make sure admin has a symlink at ~/.lnd to /mnt/hdd/app-data/lnd/
# make sure LND conf is readable and symlinked
# make sure TLS certificate is readable and symlinked
Checking if additional backup device is configured .. ()
No additional backup device was configured.
CLEANING TEMP DRIVE/FOLDER
OK: Temp cleaned
DONE BOOTSTRAP

================================================
cat raspiblitz.provision-setup.log

###################################
# _provision.setup.sh
###################################
*** Prepare bitcoin ***
mkdir: cannot create directory ‘/mnt/hdd/bitcoin’: File exists
File Exists: /mnt/hdd/bitcoin/bitcoin.conf --> 1
# setting PASSWORD B
# OK -> RPC Password B changed
[... 841 lines cut ...]
1637512542 STARTED
1637512602 STARTED
1637512663 STARTED
1637512723 STARTED
1637512784 STARTED
1637512845 STARTED
1637512905 STARTED
1637512966 STARTED
1637513026 STARTED
1637513087 STARTED
1637513147 STARTED
1637513208 STARTED
1637513268 STARTED
1637513329 STARTED
1637513389 STARTED
1637513450 STARTED
1637513510 STARTED
1637513571 STARTED
1637513631 STARTED
1637513692 STARTED
1637513692 ERROR: [lncli] could not load global options: unable to decode macaroon: empty macaroon data
1637513752 STARTED
1637513813 STARTED
1637513873 STARTED
1637513934 STARTED

rootzoll, 2021-11-25:
I dont think in your case the problem is with the lightning part ... something seems to be off with bitcoind that makes lnd restart a lot of times. If this is a fresh setup I would recommend starting fresh again. Turn everything off, flash a new sd card and during then dont choose `recover` do with fresh setup and dont delete the HDD/SSD when asked (dont reuse blockchain). 

If you run into the same problem, please write again.

rootzoll, 2021-11-30:
Closing this issue .. please reopen if problem still exists.

closed 2021-11-30 by rootzoll