raspiblitz/raspiblitz #2733: Network Connection Drops

anon21m, 2021-11-27:
I've been running a new node on v1.7.1 for a few days now. My blockchain synced OK for the first few days, but after I installed apps yesterday my node has been seemingly randomly dropping the connection to my LAN (and WAN too, I think). This makes it so that I can't SSH into the node. I don't even see the node on my list of connected devices from my router.

In order for me to get my node working again, I must unplug the power cable and plug it back in to force a reboot (I feel terrible about this with SSD read/write operations).

I have tried looking through the raspiblitz debug logs and the syslog to try and find an answer as to why the connection drops, but so far have been unable to figure anything out.

Linked is my debug log from my currently working node: https://termbin.com/0mol

A couple of observations:
- My blockchain is currently indexing after installing Mempool Explorer and BTC-RPC Explorer. This seems to lose progress after the network connection goes down. Said another way, I don't see much progress in the percentage indexed after I reboot the machine.
- The rasberry pi is running really hot (~82C). I'm using the case that came with the Rasberry Pi 4 starter kit that I bought, but I bought a case that should cool the CPU better coming on Monday.

Please let me know if you have any other ideas on things to check. Thanks.

bitsam, 2021-11-29:
Does the screen say "waiting for dhcp". I have reported a similar issue under the following link https://github.com/rootzoll/raspiblitz/issues/2640#issue-1028652829.

rootzoll, 2021-11-30:
> ﻿The rasberry pi is running really hot (~82C). 

Can you check what process is heatingh the CPU? Check command "htop" on command line.

anon21m, 2021-12-01:
Mostly it's just bitcoind that is using the most CPU. A few days ago I installed the rasberry pi in case that manages heat a lot better and that has _drastically_ improved the performance of the node. It's hovering right around 50C now. Much better temps.

I almost want to certainly say this issue was caused by CPU overheating during the blockchain indexing after installing mempool and btc rpc explorer. It's probably safe to close this ticket.

closed 2021-12-14 by rootzoll