raspiblitz/raspiblitz #2737: lnd.initwalllet.py does not work woth LND v0.14.1 in VM

openoms, 2021-11-28:
Updated LND to v0.14.1 on my VM setup and lnd.initwalllet.py does not work:
```
# Adding aliases
alias lncli="sudo -u bitcoin /usr/local/bin/lncli -n=mainnet --rpcserver localhost:10009"
...
Traceback (most recent call last):
  File "/home/admin/config.scripts/lnd.initwallet.py", line 8, in <module>
    from lndlibs import walletunlocker_pb2 as lnrpc
  File "/home/admin/config.scripts/lndlibs/walletunlocker_pb2.py", line 14, in <module>
    from . import rpc_pb2 as rpc__pb2
  File "/home/admin/config.scripts/lndlibs/rpc_pb2.py", line 22, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
seedwords=''
seedwords6x4=''

# The installed LND version is: lnd version 0.14.1-beta commit=v0.14.1-beta
```

Tried to resinstall the protobuffs:

```
cd
sudo python3 -m pip install grpcio grpcio-tools googleapis-common-protos pathlib2
rm -rf googleapis 
git clone https://github.com/googleapis/googleapis.git
rm -rf protobuffs
mkdir protobuffs
curl -o ./rpc.proto -s https://raw.githubusercontent.com/lightningnetwork/lnd/master/lnrpc/rpc.proto
curl -o ./walletunlocker.proto -s https://raw.githubusercontent.com/lightningnetwork/lnd/master/lnrpc/walletunlocker.proto
python3 -m grpc_tools.protoc --proto_path=googleapis:. --python_out=./protobuffs --grpc_python_out=./protobuffs ./rpc.proto ./walletunlocker.proto
cp ./*.proto ./protobuffs
```

But the same error persists. Investigating.

openoms, 2021-11-28:
Fixed by running:
```
sudo -H python3 -m pip install grpcio==1.38.1
sudo -H python3 -m pip install googleapis-common-protos==1.53.0
```

Should these (with the protobuff install) be part of the lnd install script to run when there is a major version update?

rootzoll, 2021-11-29:
Also those versions could change over time ... I think we need to take care of updating manual over time.

openoms, 2021-11-30:
now getting in the VM when activating the parallel services:
```
# !!! LND testnet wallet creation failed
# RPCError GenSeedRequest
# press ENTER to continue
```

openoms, 2021-11-30:
ok fixed with https://github.com/rootzoll/raspiblitz/pull/2744

closed 2021-11-30 by openoms