raspiblitz/raspiblitz #2738: BTRFS recovery notes

openoms, 2021-11-29:
Managed to convert a corrupted RAID1 USB Drive to single disk allowing partial data recovery on my Debian PC following https://www.axllent.org/docs/btrfs-raid1-recovery/

```
sudo mkdir /mnt/usb

sudo mount -o rw,degraded /dev/sdb /mnt/usb

sudo btrfs balance start -dconvert=single -mconvert=single /mnt/usb -f
Done, had to relocate 22 out of 22 chunks

sudo btrfs device delete missing /mnt/usb
```

The errors showing were:
```
$ sudo btrfs check --force  /dev/sdb
Opening filesystem to check...
WARNING: filesystem mounted, continuing because of --force
warning, device 1 is missing
Checking filesystem on /dev/sdb
UUID: f8d04abe-712c-4b9b-ac94-d2b99426b223
[1/7] checking root items
[2/7] checking extents
[3/7] checking free space cache
failed to load free space cache for block group 93447913472
failed to load free space cache for block group 95595397120
failed to load free space cache for block group 96669138944
failed to load free space cache for block group 97742880768
failed to load free space cache for block group 102104956928
failed to load free space cache for block group 103245807616
failed to load free space cache for block group 106701914112
failed to load free space cache for block group 109050724352
failed to load free space cache for block group 114788532224
failed to load free space cache for block group 117070233600
failed to load free space cache for block group 120023023616
failed to load free space cache for block group 123076476928
failed to load free space cache for block group 126431920128
failed to load free space cache for block group 130055798784
failed to load free space cache for block group 131532193792
failed to load free space cache for block group 272158818304
[4/7] checking fs roots
[5/7] checking only csums items (without verifying data)
[6/7] checking root refs
[7/7] checking quota groups skipped (not enabled on this FS)
found 5607559168 bytes used, no error found
total csum bytes: 5400100
total tree bytes: 72876032
total fs tree bytes: 29687808
total extent tree bytes: 30703616
btree space waste bytes: 16236980
file data blocks allocated: 13031481344
 referenced 5153636352

sudo smartctl -a -d sat /dev/sdb
smartctl 7.2 2020-12-30 r5155 [x86_64-linux-5.10.0-9-amd64] (local build)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     SanDisk based SSDs
Device Model:     SanDisk pSSD
Serial Number:    003d6c65d
LU WWN Device Id: 5 001b44 0d6035dc6
Firmware Version: 6EB 1030
User Capacity:    128,043,712,512 bytes [128 GB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Form Factor:      1.8 inches
TRIM Command:     Available, deterministic
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2 T13/2015-D revision 3
SATA Version is:  SATA 3.0, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Mon Nov 29 07:08:08 2021 GMT
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: FAILED!
Drive failure expected in less than 24 hours. SAVE ALL DATA.
See vendor-specific Attribute list for failed Attributes.

General SMART Values:
Offline data collection status:  (0x00)	Offline data collection activity
					was never started.
					Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0)	The previous self-test routine completed
					without error or no self-test has ever 
					been run.
Total time to complete Offline 
data collection: 		(  120) seconds.
Offline data collection
capabilities: 			 (0x51) SMART execute Offline immediate.
					No Auto Offline data collection support.
					Suspend Offline collection upon new
					command.
					No Offline surface scan supported.
					Self-test supported.
					No Conveyance Self-test supported.
					Selective Self-test supported.
SMART capabilities:            (0x0003)	Saves SMART data before entering
					power-saving mode.
					Supports SMART auto save timer.
Error logging capability:        (0x01)	Error logging supported.
					General Purpose Logging supported.
Short self-test routine 
recommended polling time: 	 (   2) minutes.
Extended self-test routine
recommended polling time: 	 (  41) minutes.

SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  5 Reallocated_Sector_Ct   0x0002   100   100   000    Old_age   Always       -       4
  9 Power_On_Hours          0x0002   100   100   000    Old_age   Always       -       0
 12 Power_Cycle_Count       0x0002   100   100   000    Old_age   Always       -       0
165 Total_Write/Erase_Count 0x0002   100   100   000    Old_age   Always       -       3221911988
171 Program_Fail_Count      0x0002   100   100   000    Old_age   Always       -       0
172 Erase_Fail_Count        0x0002   100   100   000    Old_age   Always       -       4
173 Avg_Write/Erase_Count   0x0002   100   100   000    Old_age   Always       -       70
174 Unexpect_Power_Loss_Ct  0x0002   100   100   000    Old_age   Always       -       0
187 Reported_Uncorrect      0x0002   100   100   000    Old_age   Always       -       0
194 Temperature_Celsius     0x0022   092   008   000    Old_age   Always       -       8 (0 2 0 192 0)
230 Perc_Write/Erase_Count  0x0002   100   100   000    Old_age   Always       -       233
232 Perc_Avail_Resrvd_Space 0x0003   000   100   005    Pre-fail  Always   FAILING_NOW 0
234 Perc_Write/Erase_Ct_BC  0x0002   100   100   000    Old_age   Always       -       10000
241 Total_LBAs_Written      0x0002   100   100   000    Old_age   Always       -       0
242 Total_LBAs_Read         0x0002   100   100   000    Old_age   Always       -       0

SMART Error Log not supported

SMART Self-test Log not supported

SMART Selective self-test log data structure revision number 0
Note: revision number not 1 implies that no selective self-test has ever been run
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.
```

rootzoll, 2021-11-30:
Marked for v1.8.1 to take a look again how we can make it easier to run repairs/changes in a BTRFS setup.

rootzoll, 2024-11-14:
Closing this issue because we consider to remove BTRFS - see: https://github.com/raspiblitz/raspiblitz/issues/2045#issuecomment-2471845122

closed 2024-11-14 by rootzoll