raspiblitz/raspiblitz #2742: Installation error: FAIL: setup: lnd no macaroons

P1xelfehler, 2021-11-29:
Hello,

I started a fresh install with deleting all data on the SSD in the setup process of Raspiblitz (I already had the blockchain on there which was detected). I chose LND and restored my wallet. Entered the passwords. After that, it failed. What can I do? 

Thank you in advance for your help.

Here the complete log:

###########################################################
# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Thu 28 Oct 11:00:49 BST 2021
***********************************************
# *** blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
No HDMI switch found. 
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No SSHRESET switch found. 
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 604
drwxr-xr-x   2 root root   4096 Oct 28 11:00 .
drwxr-xr-x 118 root root   4096 Oct 28 11:00 ..
-rw-r--r--   1 root root 565189 Jan 31  2020 moduli
-rw-r--r--   1 root root   1580 Jan 31  2020 ssh_config
-rw-r--r--   1 root root   3250 Jan 31  2020 sshd_config
-rw-------   1 root root   1381 Oct 28 11:00 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Oct 28 11:00 ssh_host_dsa_key.pub
-rw-------   1 root root    513 Oct 28 11:00 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Oct 28 11:00 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Oct 28 11:00 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Oct 28 11:00 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Oct 28 11:00 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Oct 28 11:00 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Mar  5  2021 ssh_import_id
wifiIsSet=0
hddBackupLocationAvailable=0
hddRestoreConfigAvailable=0
memRestoreConfigAvailable=0
wifiRestore=0
wifiBackup=0
No UASP FIX needed (2nd-try).
isMounted: 0
Checking for UASP deactivation ...
HDD already part of system: 0
Temp mounting (1) data drive (sda)
Temp mounting (1) result: 1
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
# df 
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root       30353628   4380960  24691980  16% /
devtmpfs         3991184         0   3991184   0% /dev
tmpfs            4025808         0   4025808   0% /dev/shm
tmpfs            4025808      8644   4017164   1% /run
tmpfs               5120         0      5120   0% /run/lock
tmpfs            4025808         0   4025808   0% /sys/fs/cgroup
tmpfs              32768         4     32764   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095     31392    226704  13% /boot
tmpfs             805160         0    805160   0% /run/user/1000
/dev/sda1      960303812 423362940 488086364  47% /mnt/hdd
# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda                
└─sda1      ext4   
mmcblk0            
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=1
hddRaspiVersion='1.7.1'
isSSD=0
datadisk='sda'
datapartition='sda1'
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddBytes=1000204851712
hddGigaBytes=931
hddUsedInfo='404G (47%)'
hddDataFreeKB=488086364
hddAdapterUSB='174c:0823'
hddAdapterUSAP=0

# RAID
isRaid=0
raidCandidates=0

# SWAP
isSwapExternal=0

COPY und Activating old SSH host keys
# *** blitz.ssh.sh restore
# backup dir: /mnt/hdd/ssh
# OK - ssh keys restore done
Displaying Info Message: Please Login for Recovery
## WAIT LOOP: USER SETUP/UPDATE/MIGRATION
# PROVISION PROCESS with setupPhase(setup)
Temp mounting (2) data drive (sda)
Temp mounting (2) result: 1
Refreshing links between directories/drives ..
# Sourcing /mnt/hdd/raspiblitz.conf 
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.7.1'
lcdrotate=1
lightning=lnd
network=bitcoin
chain=main
hostname='P1xlNode'
runBehindTor=on
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup 
# RASPIBLITZ SETUP STATE
hostname=P1xlNode
lightning=lnd
network=bitcoin
chain=main
setPasswordA=1
setPasswordB=1
setPasswordC=1
seedWords='absorb exit fatigue mirror rubber ticket link arm analyst style blur mystery elbow little pitch permit hand trumpet song bulb romance vacuum hidden fat'
seedPassword=''
# df 
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30353628 4389156  24683784  16% /
devtmpfs         3991184       0   3991184   0% /dev
tmpfs            4025808       0   4025808   0% /dev/shm
tmpfs            4025808   25040   4000768   1% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs            4025808       0   4025808   0% /sys/fs/cgroup
tmpfs              32768      16     32752   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095   31392    226704  13% /boot
tmpfs             805160       0    805160   0% /run/user/1000
tmpfs             805160       0    805160   0% /run/user/1001
/dev/sda1      960379920   77892 911447604   1% /mnt/hdd
# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda                
└─sda1      ext4   BLOCKCHAIN
mmcblk0            
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
# Update /home/admin/raspiblitz.info 
state=provision
message='Starting Provision'
baseimage=raspios_arm64
cpu=aarch64
board=rp4
ramMB=7862
ramGB=8
network=bitcoin
chain=main
localip='192.168.178.161'
online='1'
dnsworking=1
fsexpanded=1
displayClass=lcd
displayType=
setupStep=0
setupPhase='setup'
fundRecovery=0
hostname=
hddCandidate='sda'
hddGigaBytes=931
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddGotMigrationData=
# migrationFile()
# setting PASSWORD A

OK - password A changed for user pi, root, admin & bitcoin
...
Calling _provision.setup.sh for basic setup tasks ..
EXIT _provision.setup.sh BECAUSE OF ERROR STATE (14)
This can also happen if _provision.setup.sh has syntax errors
# /home/admin/raspiblitz.provision-setup.log

###################################
# _provision.setup.sh
###################################
*** Prepare bitcoin ***
mkdir: cannot create directory ‘/mnt/hdd/bitcoin’: File exists
File Exists: /mnt/hdd/bitcoin/bitcoin.conf --> 1
# setting PASSWORD B
# OK -> RPC Password B changed
# Reboot is needed
...
*** Start bitcoin (SETUP) ***
- This can take a while ..
# (0/50) checking if bitcoind is running ... 
# (1/50) checking if bitcoind is running ... 
OK bitcoin startup successful 
Prepare Lightning (lnd)
############## Setup LND
Creating new LND config ...
*** Init LND Service & Start ***
Starting LND Service ...
WALLET --> SEED
*** Copy LND Macaroons to user admin ***
FAIL: setup: lnd no macaroons
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(_provision.setup.sh fail)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off

P1xelfehler, 2021-11-30:
I also tried with a completely new SSD - same. Do I miss something here? Using the latest Raspiblitz image.

openoms, 2021-11-30:
Stuck on the same place in a VM, seem to be a grpc error:
```
sudo /home/admin/config.scripts/lnd.initwallet.py new mainnet password
Traceback (most recent call last):
  File "/home/admin/config.scripts/lnd.initwallet.py", line 7, in <module>
    import grpc
ModuleNotFoundError: No module named 'grpc'
```

openoms, 2021-11-30:
Fixed by running:
`sudo -H python3 -m pip install grpcio==1.38.1`

openoms, 2021-11-30:
should be part of the build_sdcard.sh, not sure how it gets uninstalled yet.

rootzoll, 2021-11-30:
Merged PR that should fix this. Put on final testing (with new sd card build).

rootzoll, 2022-02-17:
new sd card builds worked fine - closing issue for release.

PR #2744 (2021-11-30): Improve setup reliability and VM compatibility
fixes: #2742

As was rebased from v1.7 contains the documentation updates not present in dev yet.
closed 2022-02-17 by rootzoll