raspiblitz/raspiblitz #2743: RTL for C-Lightning Installation

LightRider5, 2021-11-29:
I tried to install RTL for C-Lightning. The Installation process got stuck with:
`"http fetch GET 200 https:/... some typescript stuff"`

I solved the problem with: `npm install --logLevel warn`

As shown here: https://stackoverflow.com/questions/45800022/how-to-avoid-npm-fetch-get-200-logs-when-invoking-install-in-npm-5/45805478

rootzoll, 2021-11-30:
@openoms or @LightRider5 which `npm install` is this? Was not able to find a `npm install` in the cl-install scripts.

openoms, 2021-11-30:
Likely the RTL installation, but could not confirm this yet.

rootzoll, 2021-12-08:
I also sometime see RTL taking a loooong time on that install message (lookingh like it got stuck) but at one point it continues. I will try if the `npm install` runs OK with the `-logLevel warn` but basically I only thing this will change the log outputs and may not prevent that "hanging" in general - wich happens a bit sporadic, maybe based on the response on some npm servers.

rootzoll, 2021-12-08:
Also with the `-logLevel warn` it hangs quite a bit at that `reify:typescript: http fetch GET 200 https://registry.npmjs.org/typescript/-/typescript-4.2.4.tgz` step ... but ran thru after a long waiting time.

So for the v1.7.2 I say ... lets roll with it so far - closing issue. But if anybody has the same error (RTL really stopping at this point) please repoert - if possible with more details on the error.

closed 2021-12-08 by rootzoll