raspiblitz/raspiblitz #2753: ItchySats - peer-to-peer CFDs on Bitcoin for Raspiblitz?

itchymax, 2021-12-01:
Hi guys,

ItchyMax here from https://github.com/itchysats/itchysats/

We are building a tool which enables users to trade CFDs (BTC/USD) on Bitcoin - no trust and no accounts needed - just you and your keys. 

We are currently preparing a plugin for Umbrel and wondered if users of this distro would be interested in this? If so, we are happy to work on this and create a service. 

p.s. if you want to give it a try in a save environment, we have a blog post about how to run ItchySats on testnet: 
https://itchysats.medium.com/p2p-bitcoin-cfds-give-it-a-try-4db2d5804328

You can also find us on twitter: https://twitter.com/itchysats)

rootzoll, 2021-12-13:
Could be interesting to integrate - but not on my personal prio list ... would need someone to take the lead on this. Marking as "help wanted". Happy to assist on any open technical questions.

itchymax, 2022-05-19:
Hi @rootzoll ,

I finally had time to implement it. An early draft but working branch can be found here: https://github.com/itchysats/raspiblitz/tree/itchysats

Before I open a PR here, I was wondering, is there a preference to build from source or can we download the release binary from github. I saw both being done by different apps so I wanted to ask. 

The linked branch above has both implemented, first fetching the source, and then I overwrote it with downloading the release binary from github.

Building ItchySats from source takes a long time on a RPi4 and I don't know about the performance on a RPi3.

rootzoll, 2022-05-19:
Hi - first let me point you out to our new template for additional apps ... it contains much information on how to add your app to raspiblitz - maybe check your bonus script for more details here:
https://github.com/rootzoll/raspiblitz/blob/dev/home.admin/config.scripts/bonus.template.sh

Also make sure to base your PR to the `dev` branch once you open it.

>  I was wondering, is there a preference to build from source or can we download the release binary from github.

Basically both is fine. When using binaries keep in mind that RaspiBlitz also opens up to other CPU architectures. Building from source code is might be more flexible. Dont worry about RPi3 - they are not recommended anymore to use with RaspiBlitz.

itchymax, 2022-05-19:
Hi, thanks for sharing the new template. I'll see to update what I have so far accordingly. 

Does anything speak against an interactive installation menu where the user is being asked if he/she wants to build from source or download the release?

rootzoll, 2022-05-30:
Added to v1.8.0 release

rootzoll, 2022-06-01:
Because the re-install of itchysats during an update/recover I fixed the install in `_provision_.sh` with `bonus.itchysats.sh on --download` even if the user before installed from code because there cannot be an interactive dialog blocking the recover/update process.

rootzoll, 2022-06-01:
I needed to fix the version to 0.4.20 because the github API has a ratelimit and once it blocks the IP the install failed.

rootzoll, 2022-06-01:
OK is installing and recovering fine - closing issue for v1.8.0 release

closed 2022-06-01 by rootzoll

itchymax, 2022-06-02:
Thanks for the fixes @rootzoll . One minor bug sneaked in which made the `update` function useless as it would have re downloaded the smae version. I've provided a fix here: https://github.com/rootzoll/raspiblitz/pull/3154

PR #3139 (2022-05-30): Add Itchysats service
Hi guys, 

Thanks for this amazing platform, I've been using Raspiblitz for quite a while and really appreciate your work. Also, the templates were super useful and it's easy to work with it. I'd wish more open source projects would be like that :) 

I've opted for a little wizard when enabling ItchySats which allows the user to select if he/she wants to build from source or download the binary. Building the binary takes really a long time (~1h), and if not taken care and your computer goes to sleep or drops the ssh connection then the build will be cancelled, so I think, [... cut]