raspiblitz/raspiblitz #2756: failed to mount drive on Virtualbox VM

nyxnor, 2021-12-01:
Build sdcard is okay, failed to mount drive on Virtualbox
```
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=0
isBTRFS=0

# SETUP INFO
isSSD=0
hddCandidate='sda'
hddBytes=536870912000
hddGigaBytes=500
hddPartitionCandidate='sda4'
hddDataPartitionBytes=289970061312
hddPartitionGigaBytes=270
hddFormat='ext4'
hddError='data mount failed'
hddError='storage mount failed'
hddAdapterUSB=''
hddAdapterUSAP=0

# RAID
isRaid=0
raidCandidates=0

# SWAP
isSwapExternal=0

Displaying Info Message: Please Login for Setup
## WAIT LOOP: USER SETUP/UPDATE/MIGRATION
# PROVISION PROCESS with setupPhase(setup)
Temp mounting (2) data drive (sda)
Temp mounting (2) result: 0
Refreshing links between directories/drives ..
# Sourcing /mnt/hdd/raspiblitz.conf 
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.7.1'
lcdrotate=1
lightning=none
network=bitcoin
chain=main
hostname='test'
runBehindTor=on
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup 
# RASPIBLITZ SETUP STATE
setPasswordA=1
setPasswordB=1
setPasswordC=1
hostname=test
lightning=none
network=bitcoin
chain=main
# df 
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             1996184       0   1996184   0% /dev
tmpfs             402592     548    402044   1% /run
/dev/sda1      226603524 4781400 210238432   3% /
tmpfs            2012960       0   2012960   0% /dev/shm
tmpfs               5120       0      5120   0% /run/lock
tmpfs              32768      20     32748   1% /var/cache/raspiblitz
/dev/sda2         942764   81452    796084  10% /boot
/dev/sda4      277618756  707256 262736424   1% /home
tmpfs             402592       0    402592   0% /run/user/1002
# lsblk -o NAME,FSTYPE,LABEL 
NAME   FSTYPE LABEL
sda           
├─sda1 ext4   
├─sda2 ext4   
├─sda3 swap   
└─sda4 ext4   
sr0           
# Update /home/admin/raspiblitz.info 
state=provision
message='Starting Provision'
baseimage=Debian
cpu=x86_64
board=
ramMB=3931
ramGB=4
network=bitcoin
chain=main
localip='10.0.2.15'
online='1'
dnsworking=1
fsexpanded=0
displayClass=headless
displayType=
setupStep=0
setupPhase='setup'
fundRecovery=0
hostname=
hddCandidate='sda'
hddGigaBytes=500
hddBlocksBitcoin=
hddBlocksLitecoin=
hddGotMigrationData=
# migrationFile()
# setting PASSWORD A

OK - password A changed for user pi, root, admin & bitcoin
...
Calling _provision.setup.sh for basic setup tasks ..
EXIT _provision.setup.sh BECAUSE OF ERROR STATE (4)
This can also happen if _provision.setup.sh has syntax errors
# /home/admin/raspiblitz.provision-setup.log

###################################
# _provision.setup.sh
###################################
*** Prepare bitcoin ***
error='no data drive mounted'
```

_Originally posted by @nyxnor in https://github.com/rootzoll/raspiblitz/issues/2711#issuecomment-983047128_

nyxnor, 2021-12-01:
two drives mounted at the same mount point...
[image]

how can I test? need to modify the `blitz.datadrive.sh`.
```
$ cat raspiblitz.info 
state='error'
message='_provision.setup.sh fail'
baseimage=Debian
cpu=x86_64
board=
ramMB=3931
ramGB=4
network=bitcoin
chain=main
localip='10.0.2.15'
online='1'
dnsworking=1
fsexpanded=0
displayClass=headless
displayType=
setupStep=0
setupPhase='setup'
fundRecovery=0
hostname=
hddCandidate='sda'
hddGigaBytes=500
hddBlocksBitcoin=
hddBlocksLitecoin=
hddGotMigrationData=
```

nyxnor, 2021-12-01:
[Openoms](https://github.com/rootzoll/raspiblitz/pull/2711#issuecomment-983262070)
> @nyxnor mounting drives is not relevant to the Tor config, please move to a new issue.
Mount a second disk in the VM rather than multiple partitions.

[Openoms](https://github.com/rootzoll/raspiblitz/issues/1500#issuecomment-983448680)
> @nyxnor mounting drives is not relevant to the Tor config, please move to a new issue.
Mount a second disk in the VM rather than multiple partitions.

@openoms just to include

openoms, 2021-12-01:
testing the Debian VM in VirtualBox
need to:
enable ssh server:
`sudo apt install openssh-server`
and set the Bridged adapter:
[image]

openoms, 2021-12-01:
after build:
```
$ lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0   500G  0 disk 
├─sda1   8:1    0 220.6G  0 part /
├─sda2   8:2    0   954M  0 part /boot
├─sda3   8:3    0   8.4G  0 part [SWAP]
└─sda4   8:4    0 270.1G  0 part /home
sr0     11:0    1  1024M  0 rom  
```

sr0 is the CDROM btw (just ignore)

Now shut down:
`sudo shutdown now`

and mount a new disk:
[image]
it appears as `sda`:
```
$ lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0    20G  0 disk 
sdb      8:16   0   500G  0 disk 
├─sdb1   8:17   0 220.6G  0 part /
├─sdb2   8:18   0   954M  0 part /boot
├─sdb3   8:19   0   8.4G  0 part [SWAP]
└─sdb4   8:20   0 270.1G  0 part /home
sr0     11:0    1  1024M  0 rom  
```
and is picked up correctly:
```
Running Format: (ext4) (sda)
# DATA DRIVE - FORMATTING to EXT4 layout (old)
# Unmounting all partitions of this device
# Wiping all partitions (sfdisk/wipefs)
# Creating the one big partition
Information: You may need to update /etc/fstab.
```

openoms, 2022-04-04:
This works now, see the notes at https://github.com/rootzoll/raspiblitz/tree/dev/alternative.platforms#virtual-machine

closed 2022-04-04 by openoms