raspiblitz/raspiblitz #2758: Windows-to-raspi blockchain COPY over SCP is capped at 4.2Mb/s

henkisdabro, 2021-12-02:
It seems the copy speed is capped at 4Mb/s (max 4.2Mb/s) using the GIT bash terminal. My LAN is Gigabit and my Raspi hardware is RPi4 4Gb, Samsung 1TB Plus SSD and UGreen USB3 enclosure (the recommended SSD and enclosure). Is there a reason the copy is so slow and what should I do to speed it up? Is the USB bus speed restricting this?

Grov-r, 2021-12-02:
Hi
I'm copying right now over windows7 shell (pscp in command line) im averaging 8000kb.s on gigabit wired.
Rpi4/4g and samsung T5 1gb ssd on USB3 port.
I dont know what the limiting factor is, but it's a bit faster over here, but could be 4, 5x faster.

henkisdabro, 2021-12-02:
thanks for your answer. I have an update after googling and finding issues around SCP slowness and windows restrictions. I hence installed WSL2 for Windows 10 with the full Ubuntu Linux Kernel and re-flashed raspiblitz SD card and SDD. Now I ran the setup via the Ubuntu WSL (Windows Terminal) and this time ran the `rsync` command and now it is copying with ~73Mb/s! Perhaps this can be added in the setup docs as a good solution for Windows 10?

closed 2021-12-02 by henkisdabro