raspiblitz/raspiblitz #2765: Screen doesn't show Info Status Screen; shows "Lightning Wallet Locked"

unlockingscript, 2021-12-06:
Hello community,

I'm new to github, so I don't know if this is the right way to describe my issue...
I've been building my own raspiblitz in the last few days/weeks. Everything works fine so far. I set up the raspiblitz using this very good instruction (thank you for that!). So far I can see my menu in the console and I also can change settings etc.

There is just one issue: 
Screen Console: So I ssh in and use my password A. It starts syncing the blocks that have been added since my raspi was offline. Then it shows the main menu.
Screen raspi: While I am doing the things mentioned above, the screen on my raspiblitz stays in the "Lightning Wallet Locked"-screen. It's the note that says:

Lightning Wallet Locked
"ssh admin@*IP-address*
Use your password A

But I already passed this step. On the console I am already in the menu and the Lightning Wallet should be unlocked. It does not show the Status Screen which is in the first picture of this instruction. Of course I can see the screen when I go to "INFO" in the menu but it's not showing that on the lcd screen on the raspi.

What it shows is: The upper part/head line "Raspiblitz v1.7.1 - bitcoin mainnet" as well as the four buttons on the left which I can also use due to a installation via the settings.

What I have aready tried: So I already searched the internet and the FAQs in this instruction. I also tried flashing the SD again because I thought that a mistake happened during that step. I also tried switching from lcd to lcd in order to uninstall and install the drivers again. But this lead to nothing.

Hopefully someone has a clue how this issue can be addressed.

openoms, 2021-12-06:
This might be a bug, but try to unlcok your lightning wallet from the command line to make that screen go away.
In the terminal (get there with pressing CTRL+C) type:
`lncli unlock`
and use your password C.

cstenglein, 2023-06-07:
@unlockingscript is this still relevant? Closing for now, if it's still relevant, please re-open

bita-bitcoin, 2025-03-20:
Having this issue as well.