raspiblitz/raspiblitz #2767: [Research] Cockpit - Web admin with terminal access

openoms, 2021-12-06:
https://cockpit-project.org/

Very active repo with recent developments: https://github.com/cockpit-project/cockpit

Readily available on Debian (including the RaspiOS):
`sudo apt install cockpit`

Will be available on the port 9090 over HTTPS with a self-sign certificate.
Authenticated with the linux username and password egg:
`admin`
`passswordA`

To access over the LAN allow though the firewall:
`sudo ufw allow 9090`

[image]
[image]

For remote Tor access:
`config.scripts/internet.hiddenservice.sh cockpit 443 9090`

Somewhat limited in the Tor browser due to the JavaScript blocking, but the terminal login works.

[image]

On a non-RPi Linux base system it can even be used to manage VM (just like virt-manager): https://github.com/cockpit-project/cockpit-machines

And has a graphical ZFS interface: https://github.com/optimans/cockpit-zfs-manager (which seem to be EoL being picked up by the 45Drives corp https://www.45drives.com/  https://github.com/45Drives/cockpit-zfs-manager)

openoms, 2021-12-09:
@cstenglein @rootzoll what do you think about having a terminal tab integrated to the webUI?

Same security practices should apply:
* HTTPS only (even when served over Tor)
* compare the fingerprint of the self-signed certificate on the LCD or from the menu (as already done.for webUIs)
* terminal login simply protected by the ssh password (passwordA) - as in Cockpit

Cockpit does have plugins - https://cockpit-project.org/blog/cockpit-starter-kit.html, but probably better to take the terminal part and integrate to the RaspiBlitz webUI.

A terminal in the browser window can have all the functionality not implemented in the web-interface yet and would be very handy for debugging.

openoms, 2021-12-09:
Cockpit TLS cert info (from the manpage):
```
To specify the TLS certificate the web service should use, simply drop
a file with the extension .cert in the /etc/cockpit/ws-certs.d
directory. If there are multiple files in this directory, then the
highest priority one is chosen after sorting.

To check which certificate cockpit-ws will use, run the following
command
```
```
$ sudo /usr/lib/cockpit/cockpit-certificate-ensure --check
/etc/cockpit/ws-certs.d/0-self-signed.cert
```

Display the SHA-1 fingerprint:
```
openssl x509 -sha1 -in /etc/cockpit/ws-certs.d/0-self-signed.cert -fingerprint -noout | cut -d"=" -f2
```
Display the SHA-256 fingerprint:
```
openssl x509 -sha256 -in /etc/cockpit/ws-certs.d/0-self-signed.cert -fingerprint -noout | cut -d"=" -f2
```

cstenglein, 2021-12-11:
Hi, sorry for the late answer. Just writing my thoughts and concerns, don't wanna make it sound negative 😮

Fist of all, interesting idea. I'm currently not entirely convinced about the usecases though:

> A terminal in the browser window can have all the functionality not implemented in the web-interface yet 

I'm not sure what the benefit for the user would be. Can't the user just ssh into the raspiblitz then?

A cool idea would be to access non-implemented functionality on-the-go over the webui => terminal but I'm not sure how the cockpit handles the terminal output. I guess it's not mobile friendly and that would introduce another issue.

> and would be very handy for debugging.

I would say it depends. Currently, the webUI would be more targeted towards non-technical users and errors should be displayed over the webUI as well (don't know if that *really* works in practice if there is some config file that needs change etc.)

To get better debugging output, would implementing cstenglein/raspiblitz-web#49 be enough or should there some additional solutions in the webUI?

openoms, 2021-12-11:
Thanks for your input @cstenglein !
Agree that having a terminal in the browser is mainly just a convenience factor.
I was thinking of the troubles with getting a terminal open on Windows and also on mobile, but as you say text and dialog menus are not suitable for small screens.

It actually does come with some security tradeoff also because it relies on the SSL encryption rather than ssh itself (let alone the possible browser extensions in play).

So agree, Cockpit might come useful as an option to manage the server underlying RaspiBlitz (like my case managing VMs in it), but for the RaspiBlitz WebGUI the aim should be to be able to expose the log / cli outputs.

openoms, 2021-12-12:
I find myself using the virtual machine manager extension the most: https://github.com/cockpit-project/cockpit-machines
[image]

Components are directly linked like the Terminal:
https://localhost:9090/system/terminal

cstenglein, 2022-12-17:
After having the WebUI out for some time now, I now think this would be a good idea.

Reasoning is that there are errors which you can't recover from a webUI perspective (e.g. peers.dat file from bitcoin corrupted etc.) where a terminal might make sense.

rootzoll, 2022-12-18:
As far as I remember all solutions we found for a WebUI-Terminal just had clearnet communication between browser & server. So instead of having a fake terminal in the WebUI I would recommend a special section first that basically explains how to open a real terminal on a desktop or mobile phone. I thin kthe v1.10 milestone would a great spot for that.