raspiblitz/raspiblitz #2769: LND and C-Lightning fail when setting up compiled Raspiblitz

Epic-Sandwich, 2021-12-07:
I compile raspiblitz on my raspberry Pi using the wget command. I reboot and the raspitblitz setup begins. I enter my passwords A B and C. Then the setup asks me if I want to do LND or C-Lightning. This is where the failure occurs.

If I choose LND, the setup fails when trying to copy macaroons to admin. It says:
FAIL: setup: lnd no macaroons
state(error) message(_provision.setup.sh fail)

If I choose C-Lightning, the setup fails and says: 
Generate new CL wallet ...
FAIL: setup: no cl seedwords
state(error) message(_provision.setup.sh fail)

Looking at the log files doesn't reveal anything obvious to me. How can I fix these issues?

openoms, 2021-12-07:
Please share the full commands you built the SDcard with so we know whihc branch you used.
Current development (most up-to-date) is in the `dev` branch from which you can build with:

```
# download the build script
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/build_sdcard.sh
# run
sudo bash build_sdcard.sh false false rootzoll dev headless
```

Epic-Sandwich, 2021-12-08:
Sure. After installing and updating to the latest Raspberry Pi OS, I ran this:

`wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.7/build_sdcard.sh && sudo bash build_sdcard.sh`

openoms, 2021-12-08:
the fixes are in the `dev` branch until v1.7.2 is released, can you clean your SDcard and try again with:

```
# download the build script
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/build_sdcard.sh
# run
sudo bash build_sdcard.sh false false rootzoll dev
```

Epic-Sandwich, 2021-12-09:
Thank you, I will clean my SD card and try again using the commands you provided. Any problems, I'll pop back in here.

Epic-Sandwich, 2021-12-09:
OK I just finished trying the dev branch compiled version. I chose LND option, entered passwords A B and C. Then it gave me this error:

*** Copy LND Macaroons to user admin ***
FAIL: Setup: lnd no macaroons
ERROR - please report to development team
state(error) message(_provision.setup.sh.fail)

How do I fix this macaroon problem?

openoms, 2021-12-09:
can you check the LND logs?
in menu - SYSTEM

or
tgere should be a shortcut bow:
`lndlogs`

Epic-Sandwich, 2021-12-19:
I reformatted my SD card to run the raspiblitz setup again and duplicate the error in order to see the logs. However, now when I try to run:

`sudo bash build_sdcard.sh false false rootzoll dev`

It gives me this error and wont compile:

```
For details on optional parameters - see build script source code:
1) NO-INTERACTION --> 'false'
2) FATPACK --> 'false'
3) GITHUB-USERNAME --> 'rootzoll'
4) GITHUB-BRANCH --> 'dev'
5) DISPLAY-CLASS --> 'lcd'
6) TWEAK-BOOTDRIVE --> 'true'
7) WIFI --> 'US'
X) CPU-ARCHITECTURE --> 'armv7l (armhf)'
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
!!! FAIL: Base Image cannot be detected or is not supported.
```

What am I doing wrong?

openoms, 2021-12-19:
@Epic-Sandwich you are using the 32bit Raspbian image which is not supported any 
`uname -m`
should show `aarch64`, not `armv7l`

Find the link to the 64bit image on the top of the script.

Epic-Sandwich, 2021-12-20:
OK I just reflashed with the 64 bit version of raspberry pi OS. I successfully compiled raspiblitz and began the setup process. It is now at the point where it is syncing the blockchain. I've never gotten this far before, so that's good!

I noticed this 1.7.2rc version only asked me for password A (no B or C) and didn't give me the 12 words to write down either, it just asked for password A, then rebooted, and started syncing the blockchain. Is that normal? Will it give me the 12 words later?

Epic-Sandwich, 2021-12-21:
I did previously write down seed words during a failed installation attempt. After the blockchain finishes downloading, I'll use the LND menu and see if the seed is saved. If not, then I'll format the SSD and do a fresh blockchain with all new seed words.

openoms, 2022-01-02:
closing as the OP is solved

closed 2022-01-02 by openoms