raspiblitz/raspiblitz #2771: Test: Latest WebUI & Backend preview

rootzoll, 2021-12-07:


rootzoll, 2021-12-07:
check: https://github.com/rootzoll/raspiblitz/pull/2770

cstenglein, 2021-12-12:
One caveat when testing: Wallet needs to be unlocked for the webUI to work.

Don't think I get cstenglein/raspiblitz-web#55 done today / tomorrow.

If there is an automatic logout, it's most likely the `list-all-tx` call that returns an error code like 500. Better error handling comes later in cstenglein/raspiblitz-web#15

rootzoll, 2021-12-23:
@fusion44 OK I adapeted the api isntall script in my working branch `2771webui` for new new config values. I can see the api running, but still get this non-blocking error in the logs .. anything I missed from the config side?

```
Dec 22 23:07:01 testlnd sudo[92539]: Traceback (most recent call last):
Dec 22 23:07:01 testlnd sudo[92539]:   File "./app/repositories/system.py", line 93, in _handle_gather_hardware_info
Dec 22 23:07:01 testlnd sudo[92539]:     info = await get_hardware_info()
Dec 22 23:07:01 testlnd sudo[92539]:   File "./app/repositories/system.py", line 50, in get_hardware_info
Dec 22 23:07:01 testlnd sudo[92539]:     return await get_hardware_info_impl()
Dec 22 23:07:01 testlnd sudo[92539]:   File "./app/repositories/hardware_impl/raspiblitz.py", line 56, in get_hardware_info_impl
Dec 22 23:07:01 testlnd sudo[92539]:     "public_ip": await _redis_get("publicIP"),
Dec 22 23:07:01 testlnd sudo[92539]:   File "./app/repositories/hardware_impl/raspiblitz.py", line 11, in _redis_get
Dec 22 23:07:01 testlnd sudo[92539]:     return v.decode("utf-8")
Dec 22 23:07:01 testlnd sudo[92539]: AttributeError: 'NoneType' object has no attribute 'decode'
```

fusion44, 2021-12-23:
@rootzoll looks like an error in my code. I'm fetching a value from Redis which doesn't exist. I'm on it.

rootzoll, 2022-01-25:
OK I think the both look good for a preview in v1.7.2 :)

closed 2022-02-20 by rootzoll